dewenni/ESP_Buderus_KM271 #2: option to use additional 1-wire sensors

markusn1, 2022-12-07:
Thanks for your work sven, i am using your software on one of daniels boards, great stuff!
Previously i was simply cutting power to the whole boiler, and monitored temperatures via DS18B20 sensors. I would really like to retire that setup, but i would like to keep those additional temperature sensors (especially on flow and return, since this enables me to see the difference (Spreizung)). Would it be possible to support DS18B20 sensors on one of the free GPIO ports with your software?

dewenni, 2022-12-07:
Hi Markus,
for sure this would be possible. The main problem is, that the great board from Daniel has no connectors for the free GPIO besides the 4 switches.

I´am also thinking about some further extensions and I also ordered some DS18B20 to measure flow and return.

An other idea is, to connect an additional TFT Screen.

Maybe I will do a remake of Daniels broad with additional connectors. So please be patient.

the78mole, 2022-12-08:
Not completely true :-P
If you replace one of the 1M resistors and take away a 33K and a 100nF Cs, you just need to fiddle in some Pull-Up from 3V3 and you have your 1-wire signal you could take from J2 Pin 18, 20, 22 or 24.

[image]

Another option would be to misuse the serial connector for programming. IO0 is only scanned during reset for the boot mode and it should not be a problem to pull it up through R27 with e.g. 1K instead of 10K for 1-wire. 3V3 and GND is also available on J4

[image]

the78mole, 2022-12-08:
@dewenni I will do some redesign (upgrade) soon since my stock is close to empty. I'd like to your suggestion into account. What is the interface of the display you prefer? SPI, I2C,...?

dewenni, 2022-12-08:
@the78mole 
If you would ask me for every wish I have, then I would say:

**Display:**
nearly all Display except the small OLED are using SPI interface, that unfortunately needs more Pins.
But to be honest, such a display would be only a nice gimmick.

**I2C**
to have the additional two pins for I2C would open the possibilities to a wide range of external periphery.

**Oil-Counter**
at the moment I have connected my reed contact from the oil meter to GPIO 26 instead of the User SW2.
There is already the connection to GPIO and also GND if you replace the switch with a 2-Pin connector.

**additional Temperature Sensors**
Possibility to connect two optional DS18B20 3-Wire sensors wold be fine.

** additional and optional Input/Output**
If there are still some GPIO free, it would be nice to have this also open to use.
I have already something in mind where I would need an additional output signal to control a relays or MOSFET.

summary wishlist for Christmas:
- **I2C:** GPIO21 - SDA | GPIO22 - SCL  | VCC | GND
- **(V)SPI:** GPIO23 - MOSI | GPIO19 - MISO | GPIO18 - SCLK | GPIO5 - CS | GPIO17? - DC | VCC | GND
- **Oil-Counter:** GPIO26 (instead SW2) | GND
- **DS18B20 (2x):**  GPIOXX | GPIOXX | VCC | GND
- **optional Input:** GPIO27 (instead SW1) |  GND
- **optional Output:** GPIOXX | GND

That would need a lot of redesign an maybe that would also mean, to remove the 4 LED´s and the analog Inputs.
So I don't know If you are willing to do such a big change.

And maybe this will also cause some problems with the power supply of the board.
As I remember, there were already problems in the past, because of the limited power supply from the Logamatic.
If we add a lot more components, especially the display, there might be the need of an external power supply?

the78mole, 2022-12-09:
Yes, some people had issues with the power supply I do not fully understand. Even if I activate all LEDs and creating heavy WiFi traffic, I can not see any glitch. It seems to depend on the Buderus configuration. Maybe some people have original add-ons or sensors or whatever connected that create a load close to it's maximum.
I believe, most people don't have supply issues and the features, I'll integrate next are optional. If it does not work with the supply, you can simply remove it.

markus-hi, 2023-11-14:
@dewenni 
Today i have received my new V0.0.7 Version of @the78mole the Board. Are there any support for the additional interfaces already? OW etc.?

dewenni, 2023-11-14:
No, sorry.
It is on my to do list.
Maybe I will find some time in the winter months.

markus-hi, 2023-11-14:
> No, sorry. It is on my to do list. Maybe I will find some time in the winter months.

Okay thanks 👍 Maybe i give it a try to myself if i find some time 😉

Franck78, 2024-01-27:
**Yes, some people had issues with the power supply** 

I made my version of this board with ESP32-S3. I spent some time figuring there is no power problem.

With the software found in this repo, the thing won't start without the USB plugged. Ok. 
The problem is this line.

https://github.com/dewenni/ESP_Buderus_KM271/blob/906393bf5c9a8624de0224cba7fd04bd82bd6f19/src/main.cpp#L50

When not plugged, pins are 'floating' and serial keeps detecting something. As soon as usb is plugged (with jumpers power on buderus), it's OK. 

You 'think' it is not starting but it is. Just waiting nothing.

dewenni, 2024-02-03:
@markusn1: OneWire Support added in v3.3.0

closed 2024-02-03 by dewenni

dewenni, 2024-02-04:
@Franck78 thanks for the input regarding `while(!Serial) {} // Wait`
The Communication with the logamatic is based on Serial2. Serial is only used for debug outputs on the Serial Monitor.
The `while(!Serial) {}` is in my point of view a basic implementation if you use Serial Port and I haven't given it any further thought yet.

I have not made any negative observations here either, but today I received feedback that the board did not start after flashing. Only when it was installed. Maybe this is related to this point.

What is your suggestion on that? Should I leave it out completely? Or an alternative check?

Franck78, 2024-02-05:
Hello,
From Arduino Doc

```
if(Serial)
Description

Indicates if the specified Serial port is ready.

On the boards with native USB, if (Serial) (or if(SerialUSB) on the Due) indicates whether or not the USB CDC serial connection is open. For all other boards, and the non-USB CDC ports, this will always return true.

```
It only tells if the serial is present and with that info, you decide to use it or not. Waiting for it it useless in this case (debug output).

ESP32S3 (my implementation) maybe different on the ESP32 you have. Without the PC plugged in, not USB CDC established and 100% no start   ;)

Send a specific binary to your contact.

I added a particular light show at startup. With that you cannot miss a restart (or no start).

It's basic in setup() because I know ports LED so won't be OK 'as this' on your implementation:

```
#define _on     (0)
#define off     (1)
void Blink(void) {

  digitalWrite (LED1, off);  //  clear
  digitalWrite (LED2, off); 
  digitalWrite (LED3, off);  
  digitalWrite (LED4, off);  
  delay(400);                     //allow the cpu to switch to other tasks
 
  digitalWrite (LED4, _on);  //sequentially turn on each LED
  delay(200); 
  digitalWrite (LED3, _on);
  delay(200); 
  digitalWrite (LED2, _on);
  delay(200); 
  digitalWrite (LED1, _on);
  delay(200); 

  digitalWrite (LED4, off);  // then off again
  digitalWrite (LED3, off);
  digitalWrite (LED2, off);
  digitalWrite (LED1, off);
}

```