dewenni/ESP_Buderus_KM271 #3: Consumption as value

qutschwalze, 2022-12-23:
Hi! 
Thanks for the great firmware. I use it with the78mole board.
Is it possible to integrate the calculation of the oil consumption direct into the firmware and populate it via mqtt? 

I use this calculation: 
(Einschaltdauer_in_h * Verbrauch_KG_h * kg_in_l_umrechnungsfaktor)*100)/100
Verbrauch_KG_h = 1.9
kg_in_l_umrechnungsfaktor = 1.197

I think the same calcualtion is applicable for gas with other values :) 

Thanks

Jens

the78mole, 2022-12-23:
Hi dwenni,

I also implemented a consumption counter using the running hours of the buderus controller:

```
sensor
 - platform: template
    name: "Oil Energy Equivalent"
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    entity_category: diagnostic
    accuracy_decimals: 3
    lambda: !lambda |-
      if(id(runtime1_minutes).state) {
        return id(runtime1_minutes).state * (21.0 / 0.95) / 60.0;
      } 
      return NAN;
    filters:
      - debounce: 1s
```

dewenni, 2022-12-23:
Hi Jens, Hi Daniel,

up to now, that was nothing I was interested, because I have this separate oil meter installed.
But I can think of it for sure.

At the moment I also did not compute the overall runtime inside the ESP. There are only the 3 different values:
Brenner_Laufzeit_Minuten
Brenner_Laufzeit_Minuten256
Brenner_Laufzeit_Minuten65535
```
overall minutes = Brenner_Laufzeit_Minuten + (256*Brenner_Laufzeit_Minuten256) + (65535*Brenner_Laufzeit_Minuten65535)
```
This is also something I can provide as separate mqtt topic.
And if I have this overall runtime inside the code, I could also think of some calculated consumption.

The question is, what kind of config value to provide?
Option 1 could be a definition of "oil consumption per hour" (Litre/hour)
Option 2 could be the approach from Daniel with the Heater Type in (kW) and the efficiency factor.
Option 3 could be to support both :-)

Lets see, I am also working on the next update with some new stuff.

dewenni, 2022-12-27:
@qutschwalze with release [v1.3.3](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v1.3.3) or later, I have already implemented the calculated "overall runtime" in minutes directly in the program.
You will get a new mqtt message: ```"Brenner_Laufzeit_Summe"``` / ```"burner_runtime_overall"``` based on:
```Brenner_Laufzeit_Minuten + (256*Brenner_Laufzeit_Minuten256) + (65536*Brenner_Laufzeit_Minuten65536)```

Next step would be to add a calculation for the consumption.

dewenni, 2022-12-28:
Hi @qutschwalze ,

I would suggest to implement this calculation:

**Config Part:**
```
/*--------------------------------------------------------------------------------
Optional: calculation of oil consumption based on burner runtime
--------------------------------------------------------------------------------*/
#define USE_CALCULATED_CONSUMPTION      // use calculated oil consumption
                                        // formula: Consumption_Litre = Runtime_Minutes / 60 * CFG_CONSUMPTION_KG_H / CFG_OIL_DENSITY_KG_L

#define CFG_CONSUMPTION_KG_H 2.0        // oil consumption in "kilograms per hour" - see documentation of your heater
#define CFG_OIL_DENSITY_KG_L 0.85       // densitiy of oil in "kilograms per Litle"
```

**Cyclic Part:**
```
#ifdef USE_CALCULATED_CONSUMPTION
double_t oil_consumption = runtime_sum / 60 * CFG_CONSUMPTION_KG_H / CFG_OIL_DENSITY_KG_L;
"mqttPublish(oil_consumption);"
#endif
```

Would that fit to your needs?
That would be a "endless consumption" value. You have to build your own Offset, for example after refill your oil tanks.
Otherwise I have to add additional functions to set an offset value and store it to EEPROM.
Would be also possible if this is necessary, but If you don't need it, I wouldn't implement this.

dewenni, 2023-01-16:
@qutschwalze 
with the new release I also implemented the calculated consumption value.  
You can enable this in the config.h  as described above.

The german mqtt topic is: `Oelverbrauch_Gesamt_berechnet`

It is also available in the new webUI under "Brenner"

[image]

It would be nice if you can give me a feedback if this fits your needs.

qutschwalze, 2023-01-17:
Wow! The new firmware is very cool! It also seems that it runs stable in my environment. 
Can you check if your calculation matches your oilmeter?

dewenni, 2023-01-17:
It is very strange... 
Yesterday I have activated the calculated oil consumption with `#ifdef USE_CALCULATED_CONSUMPTION` and then I had several reboots after 5-10 minutes. Then I have disabled it and it runs stable in the last 24h.
I can not imagine what´s the reason behind this.

Up to now I don't use the FreeRTOS in the Project and everything runs in the normal loop() in one Core.

I will have a look at FreeRTOS in the next few days and see if I can implement that in this project.
This could be an advantage for the parallel tasks and maybe it makes the software more stable.

Whereby it's basically been stable for me for months.

qutschwalze, 2023-01-21:
Hi Sven, at the moment it runs very stable with the consumption calculation. No reboots, no ping losses. 
My old calculation program and your calculation matches almost. I think the differences are because I use 4 decimal places with the old version.

dewenni, 2023-01-21:
thanks for the feedback. Then I will close the issue

closed 2023-01-21 by dewenni