dewenni/ESP_Buderus_KM271 #5: unknown messages from Type 0x01h (Timer?)

dewenni, 2022-12-23:
At the moment I get still this unknown messages (Typ: 0x01) after a restart of the Programm:

<0:DataType> <1:Offset> <2..7:payload>

```
01_d2_78_0a_05_37_4b_00
01_d9_28_0a_02_ff_ff_ff
01_e0_fb_00_65_65_65_65
```

I think this are also some timer configurations.  
My implementation ends with 0x01cb for "HC2_Timer14"

The "01e0:1" is known for a Time offset. In that case the offset is index3=00.  
The question is what index2=fb means?
```
01_e0_fb_00_65_65_65_65
```

If you have any ideas what's this messages are about, please let me know.

Franck78, 2024-02-05:
About 'timer' or time, something that has annoyed me since day one with the KMxxxx and a BFU is the absence of temporarily derogation.
EG, you come home earlier, turn it on, and you want to forget it. It returns to cold at the end of the 'hot' period.
It does not do this. It stays 'hot' forever.

So a super interesting message from the KM would be 
"According to the schedule of CHx, it is transition time."  With indication of circuit, auto/manual, next state.

Upon reception of this, the action is simple: Send a "return to automatic" message.

Can this be a message of this type from the KM ?

Franck78, 2024-02-05:
btw, I have the BFU/F, the F is for receiving time from DCF77 service.
Maybe another class of message when time is updated?

dewenni, 2024-11-25:
Update:

```
01_d2_78_0a_05_37_4b_00
01_d9_28_0a_02_ff_ff_ff
```
are related to solar FM244