dewenni/ESP_Buderus_KM271 #10: Unstable after 1.1.0

qutschwalze, 2023-01-04:
Hi!
I tried to update from 0.0.1 to 1.3.4 but it seems that the wifi is very unstable and the board stops responding. I have a lot of ping losses and changing values no longer works. 
After flashing back to 0.0.1 everything is fine. I tried to find the version that works and the last version that has no lags is 1.1.0.

dewenni, 2023-01-04:
@qutschwalze Thanks for that feedback.
I heard from an other user that he had also problems to update to the latest version using OTA.
After he has flashed the latest version with a USB-Serial Adapter, everything worked fine.
So maybe you can also try to flash the latest version with the Adapter.

Otherwise I am a bit clueless about the reason why this is not working in your case.
For sure I have also this version and the board from Daniel in use.

Maybe you have a different configuration?
What are your settings in config.h?

qutschwalze, 2023-01-04:
Maybe my feature request is the key to solving this. I don't how the system reacts when the topics are not reachable.

qutschwalze, 2023-01-06:
I tried everything but I didn't get any special message. I also created all mqtt-topics, but it wasn't the problem. :) 
Let's for the v2 and see if the problems are still there. 
Thanks for the good work.

closed 2023-01-20 by dewenni