dewenni/ESP_Buderus_KM271 #14: Controlling the Circulation pump

Eule01x, 2023-01-27:
Hi, seems like a really nice project!

I have KM271-like adapter and RPI with fhem connected to it. 
Wiring it to an ESP32 should be easy going. Would love to try that.

I don't know, if you already saw the fhem implementation for the KM271?
There's an option to control the circulation pump for warm water. As far as I could read from your docs, this is not yet implemented.

https://fhem.de/commandref_modular.html#KM271
```
ww_zirkulation [count]
count pumping phases for hot water circulation per hour
count must be between 0 and 7 with special meaning for

    0: no circulation at all
    7: circulation is always on
```

dewenni, 2023-01-27:
Hi @Eule01x 
I know the FHEM module and this was also the base for most of the implementation.  
For sure I can also implement this command. To be honest, I don't know why I haven't done this before.
I will see if I can implement and test this tomorrow.
Then I can create a new release with some other extensions and bugfixes that I have done last days.

dewenni, 2023-01-28:
I have added the command and have created a new release [v2.2.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v2.2.0)

Feel free to give me some feedback if it fits your needs.

closed 2023-01-28 by dewenni

Eule01x, 2023-02-14:
Runs like a charm, as far as I can say!
Thank you for implementing :)