dewenni/ESP_Buderus_KM271 #19: Control logamatic 2107 modes

Toastbrot20, 2023-03-25:
Hello Sven, 
First of all, thank you very much for your cool project. I've been using Daniel's adapter with your software since December and logging via MQTT and iobroker works great. Now I wanted to try to optimise the heating a bit and try out the settings from your WebGUI. The settings are also stored in the Logamatic and displayed in the IOBroker, but the Logamatic does not switch to the other mode. Do you have any idea what could be the reason for this? 
I tested this with "Sommer_Ab" and the HK2 operating modes. When I press the "Summer/Winter" button on the Logamatic itself, the previously set values also take over. Somehow it looks as if the Logamatic is saving the values, but is not yet actively switching. 

[image]

[image]

Bet regards, 
Matthias

dewenni, 2023-03-25:
Hi @Toastbrot20 Matthias,

first of all, there was a little typo in the webUI for the summer/winter Threshold.
9:Winter / 31:Summer is wrong.  It must be 9:summer / 31:winter

But beside this, the slider in webUI should work correct. I use this also in my smart home.
So please test some different values and compare it at the page "allgemeine Werte > Sommer_ab"
Or check it with an mqtt client.

Regarding the operation modes you are absolutely right :-)
Unfortunately there was still a bug in the callback of the webUI function. The callback always returned the same value.
But I have fixed it already and pushed it back to github.

Please take the last commit and test if it is now working correctly.

Thanks
Sven

Toastbrot20, 2023-03-25:
Hi Sven, 
yes, I have already seen the typo, but that only affects the GUI. The slider also works as intended and the values are also taken over under "allgemeine Werte > Sommer_ab". However, my Logamatic does not switch to this mode. In order for the Logamatic to adopt the winter or summer mode, I have to press the So/Wi button once on the Logamatic itself. 
I have the same behaviour when I switch on the summer mode via mqtt. The "registers" are all adopted, but my Logamatic does not switch over. (or change the state)
best regards, 
Matthias

dewenni, 2023-03-25:
hmm, strange.
On my Logamatic R2107 this setting works as it should. I use it sometimes to force the summer mode.
Maybe there is a dependency to some other settings or hardware switches at the logamatic?!

But in the callback for the operation modes for heating circuit 1+2 and hot-water was definitely a bug that now has been fixed. This could not work before.

Toastbrot20, 2023-03-25:
I would also like to force the summer mode when, for example, all my radiator valves are closed.
hmm, very strange. 
I'll run a few more tests and check my settings.

dewenni, 2023-03-25:
If the value for "sommer_ab" is written correctly to the logamatic and the value is also read back correctly, then I don't see any errors in the software.

Are all the other hardware switches on logamatic set to automatic?

Toastbrot20, 2023-03-25:
Yes, I'm with you, the software is working properly. I was just hoping that there might be a simple solution to my problem. (write some dummy values, or a specific sequence ....). 
It also seems that only the "Sommer_Ab" is affected. The rest works as expected. 

Yes, I have all settings and switches on automatic and only the HK2 with external mixer is active. And of course the hot water preparation but without priority.

Toastbrot20, 2023-03-25:
It seems the HK2 is the problem. I changed the "case KM271_SENDCMD_SUMMER:" datatyp from 0x07 to 0x08  (km271.cpp:1707) and the logamatic switches into summer mode. 
[image]

dewenni, 2023-03-26:
ok, thats very interesting! Thanks for testing! 👍
Up to now I was thinking that it is a general value. But yes, the datatyp indicates that it is related to the heating circuit.
(btw. the same is to be expected for "frost_ab")

But what about the read back config value?
Do both settings (data type 7 and 8) refer to the same configuration value that is read back?
Or is there also a separate config value that I have to add for the "sommer_ab" value for hc2?
If yes, we have to find it out.

dewenni, 2023-03-26:
Update: I have already found the right read back config address for the "HK2_Sommer_ab" value.
So I will extend this to separate commands and different read back values.

I thin the same is necessary for "Frost_ab"

dewenni, 2023-03-26:
I have pushed a new version where ```summer mode threshold``` and ```frost_mode_threshold``` is divided into separate values for heating circuit 1 and heating circuit 2. And I have also updated the webUI part for that.

@Toastbrot20: Could you please check if everything works as it should?

If yes, I will create a new release with that changes.

Toastbrot20, 2023-03-26:
Awesome! Your really fast. I tested the summer/winter mode for HK2 and it works as expected. I don't have a plan how I can test the "Frost_ab" feature at the moment, but I think this should also work. 
Thank you very much.....

closed 2023-03-26 by dewenni