dewenni/ESP_Buderus_KM271 #23: Translation mistakes

JieffeEl, 2023-04-22:
Great new release 3, thanks for the good work !

I just tried it on a spare ESP32 (before flashing it on the boiler) and noticed a few untranslated items.

In language.h : 
Line 66: Sommer instead of Summer
Line 68: Space missing between Design and Temperature
Line 100: Benutzer instead of User
Line 108: Sprache instead of Language
Line 181: Days instead of Hours
Line 564: "Warmwasser bleibt kalt" instead of "Warm Water remains cold"

On the webpage, in the "Control" bookmark, "hc1_operation_mode" is in German despite English chosen as language

dewenni, 2023-04-22:
Thanks for that. 👍
I will fix it soon.

closed 2023-04-22 by dewenni

JieffeEl, 2023-04-22:
In the "Control Tab", the HC1, HC2 and WW_operation_mode drop-boxes are still in German. I think the bug might be in webUI.cpp but I'm familiar enough with your code to pinpoint the error.

And I missed one error in language.h, line 70 : Setpoint for Hot-Water, sorry

dewenni, 2023-04-22:
yes, you are right. It forgot it the first commit.
Now it should be fixed. I also changed the code of the select controls a bit. (working with index instead of string compare)
Hope it is still working correct.

JieffeEl, 2023-04-22:
Just uploaded the new version on the Buderus and everything works fine ! Job well done !