dewenni/ESP_Buderus_KM271 #36: Activating MQTT will result in Accesspoint Mode

qutschwalze, 2023-06-16:
Hi!
I updated my board from Daniel from 2.2 to 3.2.1. Everything works except MQTT. I used the same credentials that I used for 2.2 und I can see in my ioBroker installation that the boards connects via MQTT but then it reboots and stays in Accesspoint Mode. 

mqtt.0 | 2023-06-16 11:15:58.011 | info | Client [ESP-Buderus-KM271] connection closed: timeout
-- | -- | -- | --
mqtt.0 | 2023-06-16 11:15:19.306 | info | Client [ESP-Buderus-KM271] publishOnSubscribe send all known states
mqtt.0 | 2023-06-16 11:15:19.285 | info | Client [ESP-Buderus-KM271] publishOnSubscribe send all known states
mqtt.0 | 2023-06-16 11:15:19.106 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung_2/setvalue/#" with regex /^mqtt\.0\.esp_heizung_2\.setvalue\..*/
mqtt.0 | 2023-06-16 11:15:19.106 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung_2/setvalue/#" with regex /^esp_heizung_2\.setvalue\..*/
mqtt.0 | 2023-06-16 11:15:19.085 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung_2/cmd/#" with regex /^mqtt\.0\.esp_heizung_2\.cmd\..*/
mqtt.0 | 2023-06-16 11:15:19.085 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung_2/cmd/#" with regex /^esp_heizung_2\.cmd\..*/
mqtt.0 | 2023-06-16 11:15:19.024 | info | Client [ESP-Buderus-KM271]

I compiled the firmware by myself und updated it again via OTA with your version. 

Thanks!

qutschwalze, 2023-06-16:
The Problem does not exist in 3.2.0

JieffeEl, 2023-06-16:
Daniel changed the MQTT library between 3.2.0 and 3.2.1, went from pubsubclient to async-mqtt-client to solve a reconnection problem. 3.2.1 works fine with Mosquitto, must be an incompatibility with ioBroker.

dewenni, 2023-06-17:
I´am also using a Mosquitto Broker on a separate Docker Container and have no problems.

A quick test with an ioBroker Setup and the mqtt client in ioBroker works also for me.
Maybe it has something to do with the mqtt broker from the ioBroker?

I will try to test this also...

[image]

dewenni, 2023-06-17:
so, I have tested with the mqtt broker from ioBroker and this works also fine for me

```
mqtt.0 | 2023-06-17 10:11:05.350 | info | Client [ESP-Buderus-KM271] publishOnSubscribe send all known states
-- | -- | -- | --
mqtt.0 | 2023-06-17 10:11:05.345 | info | Client [ESP-Buderus-KM271] publishOnSubscribe send all known states
mqtt.0 | 2023-06-17 10:11:05.149 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung/setvalue/#" with regex /^mqtt\.0\.esp_heizung\.setvalue(\..*)?/
mqtt.0 | 2023-06-17 10:11:05.149 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung/setvalue/#" with regex /^esp_heizung\.setvalue(\..*)?/
mqtt.0 | 2023-06-17 10:11:05.145 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung/cmd/#" with regex /^mqtt\.0\.esp_heizung\.cmd(\..*)?/
mqtt.0 | 2023-06-17 10:11:05.145 | info | Client [ESP-Buderus-KM271] subscribes on "esp_heizung/cmd/#" with regex /^esp_heizung\.cmd(\..*)?/
mqtt.0 | 2023-06-17 10:11:05.096 | info | Client [ESP-Buderus-KM271] connected with secret 1686989465093_5403
```

[image]  

for now, I don´t know how to help you...
At least it seems not to be a general problem. It must be something special with your setup.

dewenni, 2023-06-23:
Could you please describe what you exactly do?
Then I could try to reproduce it.

dewenni, 2023-07-15:
ok, so it has something to do with a restart or shutdown of the ioBroker.
I will try to do some tests in that direction within the next days and see if I can find out anything.

qutschwalze, 2023-07-15:
I can also export my MQTT-Objects with the "setvalues" and my MQTT configuration if you need it.

dewenni, 2023-07-19:
@qutschwalze please have a look here https://github.com/dewenni/ESP_Buderus_KM271/discussions/38#discussioncomment-6491537

qutschwalze, 2023-07-21:
Hi! It works but it does not prevent the problem when ioBroker is down or rebooting. In this case the board resets itself and stays in AP/Setup mode, then I have to reboot the ESP.

dewenni, 2023-07-22:
Have you disabled both marked settings?  
In my Test-Setup also a shutdown and resort works. But If you enable the setting "Eigene States beim verbinden publizieren" then all commands are send more or less at the same time when the connection established.
But the ESP can only handle one command at the same time. After every command the "Log-Mode" of the Loagmatic is activated again and all values are read again.
So I is in general not a good idea to send more than one command at the same time.

Otherwise please attach your ioBroker Settings and export of your objects.

qutschwalze, 2023-07-24:
Soo, I updated the ESP to 3.2.3 and rebooted ioBroker, reinstalled MQTT instance and added your changes. 
For now it seems that fixes the problem!
Many thanks!!

dewenni, 2023-07-24:
Thanks for the feedback and good to hear that it's finally working.

closed 2023-07-24 by dewenni