dewenni/ESP_Buderus_KM271 #37: HK2_Heizsystem zeigt Heizkörper obwohl an der Logamatic auf Fussboden gestellt

number91982, 2023-07-05:
HK2_Heizsystem zeigt bei mir Heizkörper an obwohl der Heizkreis auf Fussboden eingestellt ist.

[image]

Weiterhin wollte ich mal nachfragen ob es möglich wäre über deine Firmware die Hysteresis (Standardwert wohl 5°C) zu ändern. Dies geht laut meinem Heizungsbauer wohl nur über die Software ECOcare von Buderus, welche jedoch nicht mehr verfügbar ist. 

Vielen Dank für deine Tolle Arbeit!

dewenni, 2023-07-05:
Hallo,

danke für den Hinweis. Ich denke ich habe die Ursache auch schon gefunden.

in der `km271.cpp` muss folgendes angepasst werden:
[image]

Hier muss `data[2+1]` auf `data[2+2]` geändert werden.

Könntest du das mal korrigieren und testen?
Ic habe selbst keinen Heizkreis 2 um das zu testen.

dewenni, 2023-07-05:
Achso, bezüglich der Hysteresis - was genau meinst du damit?
Ich habe eigentlich alle Kommandos umgesetzt die irgendwo dokumentiert waren.
Da ist jetzt nichts mehr offen. Also alles was mir bekannt ist habe ich auch umgesetzt.

number91982, 2023-07-05:
Vielen Dank für deine schnelle Antwort!

Bzgl. Hysteresis, damit ist folgendes gemeint:
Standardmäßig schaltet die Logamtic sobald die Warmwasser-IST-Temperatur unter die Warmwasser-SOLL-Temperatur fällt den Kessel an. Laut meinem Heizungsbauer und auch einigen Internetforen ist es wohl mit der Software ECOCare 1.0 von Buderus möglich gewesen üb er das KM271 Modul die Schwelle zwischen 1 Kelvin und 10 Kelvin (Standard 5) anzupassen. Ich befürchte Dokumentiert ist hierzu nichts. Falls es dennoch möglich wäre das irgendwie nachzubauen wäre das ein absoluter Traum!

Ob mit EcoCare evtl. auch noch andere Einstellungen möglich waren (wie z.B. Legionellenschutzschaltung) kann ich leider nicht sagen. Jedenfalls sind diese Einstellungen über das ServiceMenü direkt an der Logamatic nicht möglich.

Ich habe die km271.cpp geändert und das ganze mal hochgeladen. Fussbodenheizung wird jetzt richtig erkannt! Perfekt!
[image]

closed 2023-07-10 by dewenni