dewenni/ESP_Buderus_KM271 #42: Compilation error: `error: 'result' was not declared in this scope`

christian-heusel, 2023-08-20:
There currently is an error when building this project for the esp32:

```
Compiling .pio/build/esp32/lib560/AsyncMqttClient/AsyncMqttClient/Packets/Out/PubAck.cpp.o
Compiling .pio/build/esp32/lib560/AsyncMqttClient/AsyncMqttClient/Packets/Out/Publish.cpp.o
Compiling .pio/build/esp32/lib560/AsyncMqttClient/AsyncMqttClient/Packets/Out/Subscribe.cpp.o
.pio/libdeps/esp32/ESPUI/src/ESPUI.cpp: In function 'String heapInfo(const __FlashStringHelper*)':
.pio/libdeps/esp32/ESPUI/src/ESPUI.cpp:58:5: error: 'result' was not declared in this scope
     result += ESP.getFreeHeap();
     ^~~~~~
```

I think this problem was already reported in https://github.com/s00500/ESPUI/issues/243 and is fixed by https://github.com/s00500/ESPUI/commit/f8fbcf887af6cf1fede688a1ea15a63307565a48, so it should be solvable by a dependency bump 👍🏻

dewenni, 2023-08-20:
So it should be fixed with 2.2.3 right?
But why is 2.2.3 not used if I set platform.io to
```
lib_deps =
  s00500/ESPUI @ ^2.2.1
```

2.2.3 should be the latest release.
also this is not working.

```
lib_deps =
  s00500/ESPUI @ 2.2.3
```

christian-heusel, 2023-08-20:
💡 the library author has to run `pio pkg publish` as per https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing, will file an Issue and report back here 👍🏻

christian-heusel, 2023-08-20:
So after a lightning fast response from the ESPUI upstream the new version is published, but we still seem to get some caching weirdness:

```
$ pio pkg search ESPUI
Found 1 packages (page 1 of 1)

s00500/ESPUI
Library • 2.2.3 • Published on Sun Aug 20 15:39:30 2023
ESP32 and ESP8266 Web Interface Library

$ pio pkg install
Warning! Ignore unknown configuration option `upload_url` in section [env:esp32]
Resolving esp32 dependencies...
Library Manager: Installing s00500/ESPUI @ ^2.2.3
UnknownPackageError: Could not find the package with 's00500/ESPUI @ ^2.2.3' requirements for your system 'linux_x86_64'
```

I will retry in a bit 😊

christian-heusel, 2023-08-20:
Now works flawlessly! 😊 ✅

closed 2023-08-20 by christian-heusel