dewenni/ESP_Buderus_KM271 #45: Probleme mit v.3.2.4

number91982, 2023-09-28:
Nach Update auf v3.2.4 mit KM271-Wifi v0..0.6 lässt sich nicht mehr auf das Webinterface zugreifen.
Pingen lässt sich sich das ganze jedoch noch.

Nach zurücksetzen der Einstellungen lässt sich das Board wieder konfigurieren. Sobald jedoch die GPIO Zuweisung gespeichert wird treten die Probleme wieder auf.

dewenni, 2023-09-30:
Also wenn du die Konfiguration zurück setzt, dann läuft es und du kannst auch auf das Webinterface zugreifen?
Und wenn du dann die GPIO´s konfigurierst, speicherst und neu startest, dann geht es in eine Bootschleife?
Liegt das tatsächlich an den GPIO Einstellungen oder passiert das auch wenn du andere Einstellungen änderst?

Das ist insgesamt etwas komisch weil ich an dem ganzen Konfigurationshandling bewusst nichts geändert habe.
Es wäre hilfreich wenn du hier noch ein paar Details liefern könntest. Vielleicht hast du auch die Möglichkeit das Board über einen seriellen Adapter am PC zu betreiben und den Monitor Modus zu nutzen?

number91982, 2023-09-30:
Ich bin leider jetzt am WE nicht Zuhause um Tests durchzufüjren, aber am Montag kann ich das gerne machen.

Vorab paar weitere Infos:
Mein erster Versuch war einfach das Update einzuspielen, danach hab ich mich gewundert dass das Webinterface nicht mehr laden wollte. Der Webservice scheint jedoch zu laufen, bekomme keine Timeouts oder der gleichen.
Ping war weiter möglich.
Dachte mir da ist was beim Flaschen schief gelaufen und hab die Platine aus der Logamatic genommen und per USB mit Strom versorgt.
Sofort konnte ich aufs Webinterface zugreifen. (Ohne reset)

Ich dachte mir evtl hat sich die konfig verschluckt, also einfach Mal einen Reset durchgeführt. Komplett neu eingerichtet und das Board wieder in die Logamatic eingebaut. Selbiges Verhalten.

Also nochmal resettet im eingebauten Zustand. Diesmal nur wlan konfiguriert.
GPIO alles auf -1 gestellt und nach dem Neustart war erstmal alles i.o. natürlich ohne auf die Heizung zugreifen zu können.
Dann GPIO Pins manuell eingetragen, erstmal nur RX und TX und das war's dann.
Hab dann erstmal wieder vorgängersoftware per Update eingespielt, hier ist wieder alles i.o.

Bin leider am WE nicht Zuhause um weitere Tests durchzuführen, aber kann am montag gerne Mal wieder updaten und versuchen seriell darauf zuzugreifen.

dewenni, 2023-10-01:
_Dachte mir da ist was beim Flaschen schief gelaufen und hab die Platine aus der Logamatic genommen und per USB mit Strom versorgt. Sofort konnte ich aufs Webinterface zugreifen. (Ohne reset)_

Das klingt schon fast danach als ob es nur ein Problem mit der Spannungsversorgung seinen könnte.
Die Spannungsversorgung scheint bei manchen Logamatic für die Platine schon grenzwertig zu sein.
Es wäre also interessant ob es funktioniert, wenn du die Platine in der Logamatic extern versorgst.
Du kannst dafür ja den Jumper ändern und die Platine über ein USB Netzteil versorgen.

number91982, 2023-10-01:
So gerade wieder zuhause angekommen

Also an der Spannungsversorgung liegts definitiv nicht, heute nochmal die selbe Prozedur mit deinem Update durchgeführt bei externer Spannungsversorgung -> selbes Resultat.
Settings zurückgesetzt und zurück zu 3.2.3
zurück internen Versorgung

Dann einfach mal selbst den Compiler angeworfen und deine Quellen 1:1 durchlaufen lassen, Update eingespielt -> alles funktioniert

dewenni, 2023-10-01:
Also nachdem du die Version 3.2.4 über PlatformIO per Download auf das Board aufgespielt hast, hat es funktioniert?
Würde bedeuten mit dem Image stimmt was nicht.
Welche Image Datei hast du denn wie benutzt?
Es gibt ja einmal das Image für das initiale Flashen (..esp32_flash..) und einmal das Image für das Update über die Weboberfläche (..ota_update..)

number91982, 2023-10-02:
hab natürlich in beiden Fällen das ota_update eingespielt.
Das Update habe ich nicht per PlatformIO eingespielt sondern das Update in beiden Fällen per Web-Oberfläche durchgeführt.

Ich hab dir hier mal das ota_update angehängt, welches mein Compiler ausgespuckt hat.
[buderus_km271_ota_update_v3.2.4.bin.zip](https://github.com/dewenni/ESP_Buderus_KM271/files/12779613/buderus_km271_ota_update_v3.2.4.bin.zip)

dewenni, 2023-10-02:
ok, danke für die Infos.
Plausibler macht es das allerdings nicht.

Falls sonst noch jemand das gleiche Problem hatte oder hat, bitte mal melden.
Im Moment kann ich es leider noch nicht nachvollziehen.

number91982, 2023-10-02:
ich hab jetzt mal noch ein bisschen rumprobiert um den Fehler etwas eingrenzen zu können:

platform-espressif32 von 5.4 hoch gezogen auf 6.4.  und neu kompiliert - laufe sofort wieder in den Fehler

zurück auf platform-espressif32 5.4 - neu kompiliert - Fehlerfrei.

dewenni, 2023-10-02:
Ok, ich habe glaube vor dem letzten Update auch verschiedene Updates in PlatformIO gemacht und bin auch auf 6.4
Warum das damit jetzt nicht geht ist dennoch eine gute Frage.
Prinzipiell bin ich an der Speichergrenze von dem Standard Partitionsschema.
Vielleicht geht ihm da irgendwie der Speicher aus.
Für das nächste Update an dem ich gerade arbeite musste ich auch das Parititonsschema anpassen. Mit den geplanten Erweiterungen war ich bei 99,1% Speicher. Das konnte man noch kompilieren, beim Starten ist er aber dann abgestürzt.
Mit dem anderen Schema bin ich jetzt bei 66% und es läuft stabil.
Wenn man das Partitionsschema ändert, geht aber die Config verloren, deswegen wollte ich mit der 3.2.4 die Funktion für Backup und Restore von der config.json implementieren.
Wenn jetzt mit dem nächsten Update die Config verloren geht, kann man sich die vorher noch sichern und danach wieder einspielen.

number91982, 2023-10-02:
Ich versteh das leider noch viel weniger als du, da ich so gut wie keine Ahnung davon habe.

Wenn du aber irgendwie Versuchskaninchen brauchst stehe ich gerne zur Verfügung!
Danke für deine tolle Arbeit!

pcsv17, 2023-11-17:
Hallo, habe mir auch eine Platine gekauft, angeschlossen (auch extra mal mit USB Stromversorgung) und mit dem WLAN verbunden. Nun habe ich vermutlich auch das Problem, dass er nicht genug Speicher hat. Erst Funktioniert bis auf das Dashboard fast alles. Danach kommt nach kurzer Zeit nur noch das Logo der Webseite oben.

the78mole, 2023-11-18:
Kleiner Hinweis meinerseits. Die aktuelle Charge (ab Anfang November) hat wieder einen ESP mit 8 MB Flash. Die Charge von Juli hatte aufgrund Lieferengpässen nur einen 4 MB-ESP32 drauf.

solutionfound, 2024-01-17:
ich hatte genau das gleiche Problem und ein Downgrade auf 3.2.3 hat das Problem gelöst. Irgendwo scheint bei der 3.2.4 der Wurm drin zu sein.

Franck78, 2024-01-27:
Maybe move the config file to the serial eprom.
The partition is ~ 1M or 2M for a small file that does not really need to be readable by human (more code removed).
'DoubleResetDetector' also uses the filesystem but should be removed (replaced by a long press detection)  ;)
With these this change, ESP with 4M should be salvageable.

dewenni, 2024-01-31:
> ich hab jetzt mal noch ein bisschen rumprobiert um den Fehler etwas eingrenzen zu können:
> 
> platform-espressif32 von 5.4 hoch gezogen auf 6.4. und neu kompiliert - laufe sofort wieder in den Fehler
> 
> zurück auf platform-espressif32 5.4 - neu kompiliert - Fehlerfrei.

@number91982 
Ich habe das Release v3.2.5 jetzt mal wieder basierend auf espressif32 5.4 erstellt.
Teste mal on die Version bei dir jetzt damit auch stabil läuft.

number91982, 2024-01-31:
Habe die 3.2.5 installiert - läuft bisher einwandfrei. Wenn ich Mal Zeit habe jage ich sie mit der aktuellen espressif32 durch den Compiler und teste damit und gebe dir Rückmeldung!
Vielen Dank für dein Engagement!

closed 2024-01-31 by dewenni

number91982, 2024-02-08:
Heute mal den Compiler angeworfen mit espressif32 6.5 - Probleme wie in der Version 3.2.4 also zurück zur espressif32 5.4 und schon ist wieder alles i.O.

dewenni, 2024-02-08:
ok, dann versuche ich mal soweit möglich auf der 5.4 zu bleiben.
Kann aber natürlich sein, dass andere BugFixes mich irgendwann zwingen auf eine neuere Version zu gehen.
Die aktuelle v3.3.1 ist auch auf 5.4 kompiliert. Läuft die bei dir auch?

dewenni, 2024-02-08:
Ok, danke!