dewenni/ESP_Buderus_KM271 #50: Version 3.2.4 startet nicht (loop)

MisterButcher, 2023-11-28:
Hallo,

ich habe das KM271 Board von Daniel Version 0.0.7

Bei Daniel liegen die "dewenni" Versionen 3.2.3 und 3.2.4 mit dabei. 3.2.3 scheint zu laufen und startet (das WLAN wird angezeigt). 

3.2.4 startet nicht sondern loopt mit folgenden Meldungen: 

```
doubleResetDetected
Saving config file...
Saving config file OK
LittleFS Status: 1
[    36][E][vfs_api.cpp:105] open(): /littlefs/config.json does not exist, no permits for creation
Failed to read file, using default configuration and start wifi-AP

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
```

Die selben Fehlermeldungen kommen auch, wenn ich das 3.2.4 bin File nehme das mir von der IDE erzeugt wird ... 

Das KM271 Board ist im Moment über einen CP2102 per USB an den Computer angeschlossen.

Irgendeine Idee? Mache ich was falsch? In der IDE finde ich keine config.json ... 

Danke + Gruss

Stefan

dewenni, 2023-11-29:
Hallo Stefan,

leider kann ich dir da spontan nicht helfen.
Der Fehler deutet eigentlich auf ein Problem mit der Spannungsversorgung hin.
Da die 3.2.3 jedoch läuft, ist das auch komisch.

Ich würde sagen du nutzt erstmal die v3.2.3
Ich versuche demnächst nochmal ein Update zu bringen, kämpfe aber auch da derzeit noch mit Stabilität und Speicher Themen. Irgendwie ist das zuletzt etwas "instabil" geworden und ich weiß nicht so recht warum.

Grüße Sven

MisterButcher, 2023-12-03:
Hallo Sven, 

danke für die Antwort, Ich habe mir daraufhin die 3.2.3 runtergeladen, und mit plattformio aufgemacht. Wenn ich hier allerdings einen build mache laufe ich am Ende beim merge auf einen Fehler. 
Leider kenne ich mich mit plattformio überhaupt nicht aus, mein "gebastel" hat bisher keinen Erfolg gebracht.
Hast du dazu vielleicht einen Tipp?
Danke + Gruss
Stefan

```
 *  Executing task: C:\Users\user1\.platformio\penv\Scripts\platformio.exe run 

Processing esp32 (platform: espressif32; board: upesy_wroom; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/upesy_wroom.html
PLATFORM: Espressif 32 (6.4.0) > uPesy ESP32 Wroom DevKit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- ArduinoJson @ 6.21.3
|-- muTimer @ 0.1.1
|-- ESPUI @ 2.2.3
|-- ESP_DoubleResetDetector @ 1.3.2
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient @ 0.9.0
|-- LittleFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\esp32\src\basics.cpp.o
Compiling .pio\build\esp32\src\config.cpp.o
Compiling .pio\build\esp32\src\km271.cpp.o
Compiling .pio\build\esp32\src\main.cpp.o
Compiling .pio\build\esp32\src\mqtt.cpp.o
Compiling .pio\build\esp32\src\oilmeter.cpp.o
Compiling .pio\build\esp32\src\webUI.cpp.o
Building .pio\build\esp32\bootloader.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generating partitions .pio\build\esp32\partitions.bin
Compiling .pio\build\esp32\libbb9\AsyncTCP\AsyncTCP.cpp.o
In file included from src/basics.cpp:2:
.pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https: [-Wcpp]
 #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https://docs.elegantota.pro/async-mode/     
  ^~~~~~~
In file included from src/main.cpp:14:
.pio/libdeps/esp32/ESP_DoubleResetDetector/src/ESP_DoubleResetDetector.h:114:10: warning: #warning Using ESP32 Core 1.0.6 or 2.0.0+ [-Wcpp]
         #warning Using ESP32 Core 1.0.6 or 2.0.0+
          ^~~~~~~
Compiling .pio\build\esp32\lib19f\FS\FS.cpp.o
Compiling .pio\build\esp32\lib19f\FS\vfs_api.cpp.o
Compiling .pio\build\esp32\lib96e\WiFi\WiFi.cpp.o
Archiving .pio\build\esp32\libbb9\libAsyncTCP.a
Compiling .pio\build\esp32\lib96e\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32\lib96e\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32\lib96e\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32\lib96e\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32\lib96e\WiFi\WiFiSTA.cpp.o
[... 108 lines cut ...]
Linking .pio\build\esp32\firmware.elf
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  22.6% (used 74132 bytes from 327680 bytes)
Flash: [========= ]  86.5% (used 1133641 bytes from 1310720 bytes)
Building .pio\build\esp32\firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
merge_bin([".pio\build\esp32\firmware.bin"], [".pio\build\esp32\firmware.elf"])
C:\Users\user1\.platformio\penv\Scripts\python.exe C:\Users\user1\.platformio\packages\tool-esptoolpy\esptool.py --chip esp32 merge_bin --flash_size 4MB -o F:\MC\SVEN MQTT\KM271-3.2.3\.pio\build\esp32/firmware_merged.bin 0x1000 F:\MC\SVEN MQTT\KM271-3.2.3\.pio\build\esp32\bootloader.bin 0x8000 F:\MC\SVEN MQTT\KM271-3.2.3\.pio\build\esp32\partitions.bin 0xe000 C:\Users\user1\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 0x10000 F:\MC\SVEN MQTT\KM271-3.2.3\.pio\build\esp32/firmware.bin
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}]
                         [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                         [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET]
                         [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument <address> <filename>: Address "MQTT\KM271-3.2.3\.pio\build\esp32/firmware_merged.bin" must be a number
*** Error 2
*** [.pio\build\esp32\firmware.bin] F:\MC\SVEN MQTT\KM271-3.2.3\.pio\build\esp32/firmware_merged.bin: No such file or directory
```

closed 2024-01-18 by dewenni