dewenni/ESP_Buderus_KM271 #52: plötzlich falsche Aussentemperatur?

kalleick, 2023-12-02:
Hallo, 

ich habe plötzlich 255 Grad Außentemperatur, was könnte das sein, Sensor defekt?

[image]

Iam-klm, 2023-12-02:
Das Problem ist bekannt und ist nur auf dem Dashboard falsch. Unter dem Reiter allgemeine Anzeige stimmen die Werte.
Schönes Wochenende.

Heusel, 2023-12-04:
Kann Iam-klm bestätigen unter MQTT/ Grafana / allgemeine Werte stimmen die Temperaturen.
 
Ich habe Zeile 1140/1141 aus der Datei WEBUI.cpp in Verdacht. Hier sollte anstelle von uint8ToString ein int8ToString stehen. 

bisher:
-  ESPUI.updateLabel(id.dash.tmp_out_act,  uint8ToString(kmStatusCpy.OutsideTemp));
-  ESPUI.updateLabel(id.dash.tmp_out_act_d,  uint8ToString(kmStatusCpy.OutsideDampedTemp));

geändert:
+  ESPUI.updateLabel(id.dash.tmp_out_act,  int8ToString(kmStatusCpy.OutsideTemp));
+  ESPUI.updateLabel(id.dash.tmp_out_act_d,  int8ToString(kmStatusCpy.OutsideDampedTemp));

Hab ne Testversion für mich gebaut, allerdings hab ich jetzt keine Minusgrade mehr und kein Kältespray im Haus. Somit noch keinen abschliessenden Test .  
Wer also heute eine kalte Nacht erwartet kann gerne mit testen :-)

dewenni, 2023-12-04:
Ja da ist der Fehler.
Das habe ich auch bereits korrigiert und ist somit im nächsten Release gefixed.
Ich wollte jetzt nur alles deswegen ein neues Release machen.
Ist ja wirklich nur die Anzeige im Dashboard der WebUI.

MQTT Werte sind davon nicht betroffen.

Grüße Sven

Heusel, 2023-12-04:
Super danke für den schnellen fix und das coole Projekt  von euch.
Dann freunen wir uns auf das nächste Release.  Desween lohnt es sich nicht wirklich. Ist ja jetzt ausdiskutiert

Franck78, 2024-01-27:
@Heusel ,
"I built a test version for myself, but now I no longer have temperatures below zero and no cold spray in the house. So no final test yet."

 ```
 // Outdoor Temp
  ESPUI.updateLabel(id.dash.tmp_out_act,  int8ToString(kmStatusCpy.OutsideTemp - 20));

```
ah ah, no need to freeze, just subtract something and your are done. So yes, **int8** is good fix.
The thing is, are there others to find ;)

dewenni, 2024-01-30:
solved with v3.2.5

closed 2024-01-30 by dewenni