dewenni/ESP_Buderus_KM271 #54: KM271 mit FM244 Solar - Web frontend not working

solutionfound, 2023-12-03:
Hi,
wahrscheinlich bin ich der einzige, der die KM271 wifi v0.0.7 mit dem FM244 Solar Modul verwendet.
Ich habe hierzu ein "Breakout" Stecker angelöste und die Karte extern angesteckt.

Es sind zusätzlich noch zwei Heizkreise installiert und ein einstufiger Brenner.

Ich habe folgenden Effekt: Kurz nach dem Starten wird die Oberfläche mit einigen Werten angezeigt (allerdings nicht im ersten Tab), aber auf dem 2. und 3. Tab. Dann refesht die Seite und es werden nur noch Werte als Text angezeigt und die Oberfläche nicht mehr richtig "gerendert". In dem Bereich, wo Einzelwerte stehen sollten ist ein ganzer String mit aneinandergefügten Werten sichtbar. Dann erscheint nur noch die Kopfzeile: Buderus Logamatic Connected

Leider ist die Heizung 100km entfernt und ich habe am Freitag nur die Karte installiert, so dass ich nicht vor Ort gehen kann und einen Reset oder ähnliches oder eine neue Firmware auspielen. Am 16.12. wäre ich vsl. wieder vor Ort.

Ich hatte mal die FM244 herausgezogen, was mit einem Fehler im Display quitiert wurde, aber nicht an dem Verhalten händerte.
Leider bin ich nicht auf die Idee gekommen das Solarmodul in der Konfig auszuschalten und dann zu testen. Das kann ich aber das nächste mal machen.

Per MQTT bekomme ich einige Werte von der Karte, aber nicht alle, die im Param.txt hinterlegt sind. Kann das auch damit zusammenhängen?

Kannst Du mal schauen, ob Du den Fehler nachvollziehen kannst (mit den wenigen Informationen und ohne FM244).

Grüße
Marco

heiner1976, 2023-12-09:
Hallo Marco, 
ich habe zwar nicht die FM244 aber hatte mit der KM271 v0.0.7  die gleichen Probleme. Ich habe über OTA ein downgrade auf v3.2.3 geamcht uns alles funktionierte.
Vielleicht hilft es Dir ja auch.
LG
Henrik

solutionfound, 2023-12-10:
Hallo Henrik,
danke für Deinen Tipp. Ich werde nächstes Wochenende ein Downgrade versuchen.
Nutzt Du MQTT und haben bei Dir alle Parameter Wert?
LG
Marco

solutionfound, 2024-01-17:
Mit dem Downgrade auf 3.2.3 funktioniert die Oberfläche und es scheint das Fehlerbild zu sein, welches schon andere beschrieben haben und es hat nichts mit der Solarfunktion zu tun.
Allerdings hatte ich arge Probleme die Firmware über das Web-frontend einzuspielen weil nach Auswahl der Firmware der linke Bereich einfach leer blieb ohne das der Upload durchgeführt wurde.

closed 2024-01-18 by dewenni

dewenni, 2025-09-27:
@solutionfound kannst du mir bei Gelegenheit mal ein paar Details liefern, wie du das mit dem breakout an der FM244 gemacht hast?
vielleicht ein Foto und eine kleine Beschreibung.

Danke

solutionfound, 2025-10-01:
@dewenni - die Info Deines Posts ist bei mir im Spam gelandet, daher erst jetzt eine kurze Antwort.
Ich war damals mit the78mole per Email im Austausch und habe ihm inklusive Bilder berichtet wie ich es gemacht habe.
Das könnte ich recht einfach weiterleiten, wenn ich eine Email hätte oder soll ich hier etwas zusammenfassen?

Kennst Du noch jemanden, der die Anlage mit Solarmodul im Einsatz hat oder woher kommt das Interesse?

dewenni, 2025-10-01:
Es gab hier im Forum eine Anfrage zu dem Thema:
https://community.simon42.com/t/esp-buderus-km271-anbindung-einer-logamatic-r2107-hs-2105/40906/11?u=dewenni

Ideal wäre, wenn du hier unter "Discussions" einen Beitrag veröffentlichen könntest mit deiner Lösung.
Dann kann man spätere Interessenten auch darauf verweisen.