dewenni/ESP_Buderus_KM271 #56: Restarts trotz 1000 uF, aber Hotspot, static IP

chickbi, 2023-12-12:
Hallo, tolles Projekt, Respekt und Dank (an HW und FW)!
Hab seit ein paar Tagen endlich ein KM271-pcb 0.0.6 in Betrieb (ohne Bauteile gekauft, als es keine gab).
Einstellungen im Hotspot-UI 192..., danach unter statischer IP 172... im Web-UI erreichbar.
Soll per MQTT , NodeRed, InfluxDB, Grafana (bin hier aber nicht festgelegt).
Aber zu MQTT-Tests komme ich meist nicht, da immer mal wieder nicht mehr unter 172... erreichbar. Nach 'Speichern und Neustart' ohne Änderungen wieder unter 192.. erreichbar. Aber auch nach Trennen der Spannungsversorgung meist direkt im gespeicherten 172... Hmm, komisch. Spannungen 5,0..5,1 V und 3,30 V, perfekt. Mit Oszilloskop allerdings etwas schlechter: 5 V haben mehrere 100 mV bei jeder LED (Blinken sichtbar) aber gefühlt auch jeder Rechenvorgang im ESP (schnellere Wechsel): Menü-Tab-Wechsel usw. Die 3V3-Regelung filtert das meiste hiervon, nur deutlich kürzere und kleinere Einbrüche, sieht eigentlich ok aus. C7 gegen 1000 uF (low-ESR) getauscht, besser. Nach ein paar Minuten, spätestens ein paar Stunden kommt aber doch wieder ein Wechsel zu 192... Klar kann man jetzt im Netzteil weiter optimieren (3V3 mehr, Buderus 2107-Netzteil ist ja Trafo und dann sieht's aus wie Längsregler, den Elko mal checken usw.). 

Frage: 
Weshalb startet der der ESP in den Safe 192... und nicht wieder in den 172... ? Kann man hier was optimieren. Kenne mich hier nicht so aus, Hat man da überhaupt ne Chance zwischen Brownout zu unterscheiden? Nach längerer Spannungstrennung startet er ja auch wieder normal (172..). 
           
Feste IP / static IP:
Wäre es möglich das Menu hierfür nachzupflegen? Habe hier nur statische IPs und die Fritzbox 7490 (FW aktuell) hat ein Problem mit dem Verbindungsaufbau: die MAC-Adresse der IP zuzuweisen klappt fast wie gewohnt (etwas umständlicher), aber bleibt "bis zur nächsten Anfrage" inaktiv: kein Ping, nicht erreichbar. Erst nach DHCP-Server mit nur der gewünschten einen IP und dessen ersten Anfrage klappt dann alles wie gewünscht (wahrscheinlich nur sicher bis Ablauf der DHCP-Leasetime). Aber die Abbrüche sind ja häufig, also muß der DHCP-Server für diese IP leider anbleiben, damit der Reconnect klappt. Andere ESP-Projekte haben oft neben DHCP auch die static-IP Option, die Verbindung zurFritzbox klappt da deutlich einfacher und ohne DHCP (Tasmota, ESP-CAM, 8266, 32, aber auch Raspi usw.).
Bin hier neu, wie kann ich mich einbringen? Messen? Netzteil weiter untersuchen? FW? (mag mich eingentlich nicht aus dem Fenster lehnen, ob ich da zeitlich Erwartungen erfüllen kann. Versuch's aber gern). 
Lieben Gruß aus dem Sauerland...

dewenni, 2023-12-12:
Hallo,

ich habe aktuell leider wenig Zeit mich mit dem Projekt zu beschäftigen.
Vielleicht klappt das zwischen den Jahren etwas besser.

Aber zum Thema Neustart und IP. Im Projekt gibt es diesen "Setup-Mode" in dem der ESP einen eigenen Hotspot aufmacht und eine statische IP hat.
Der Setup-Mode wird ausgeführt, wenn innerhalb 10 Sekunden der Reset Taster zwei mal gedrückt wird.
Gleiches passiert aber auch, wenn der ESP innerhalb von 10 Sekunden von sich aus zwei mal neu startet. Z.B. wegen Spannungseinbrüchen.
Das erklärt vielleicht ein wenig das beobachtete Verhalten.

Die Möglichkeit eine statische IP anstelle von DHCP in der Config zu berücksichtigen, nehme ich mal als Anforderung auf.

Ich habe noch ein paar Dinge vorbereitet, bin aber noch nicht dazu gekommen das zu ende zu bringen und in ein stabiles Release zu verpacken.

Franck78, 2024-01-27:
Somewhat similar to what I add on my own version (esp32-S3). 
I think getting rid of this automatic 'DoubleResetDetector' is a good thing.

On another ESP project, to go to reset mode, I just watch a long press on button 'boot' or 'prog' (not sur which one was easy to read) and that's it. No weirdos.

@chickbi, not convinced power is the problem (for me it was simple line 50 in main.cpp. 

For DHCP, try to find in your DHCP server setup "lease time" and set it to 1 day or more.

dewenni, 2024-02-03:
@chickbi : manual ip settings added in v3.3.0

closed 2024-02-03 by dewenni