dewenni/ESP_Buderus_KM271 #60: Möglichkeit den Configwert Reglereingriff Raumtemp auch bei MQTT setzen zu können

HMMike, 2024-01-07:
Hi vllt könntest du noch folgendes Implementieren:

Kommando um den Reglereingriff der Raumtemperatur einstellen zu können

die Codefolge wäre für HK1;   (b"\x07\x15" + Temp + b"\x65\x65\x65\x65\x65")

bzw für HK2;   (b"\x08\x15" + Temp + b"\x65\x65\x65\x65\x65")

bzw generell vllt, das in einer art expertenmodus du ein Expert Publish mit (bwispielsweise Hex string mit 16 Stellen entsprichen den 8 Bytes eines Kommandotelegramms, damit wäre auch "ausprobieren" möglich und den als KOmmandojob an die L2107 schickst

Gruss, Mike

dewenni, 2024-01-07:
Hi, 
danke für den Hinweis.
Kannst du mir etwas genauer erklären was man damit einstellt und woher du die Info zu dem Parametern hast?

HMMike, 2024-01-08:
Hi. Ich habe die L2107 mit der funkuhr Fernbedienung, daher fahre ich raumregelung.

Der genannte Parameter beschreibt den Eingriff des Reglern der Raumtemperatur auf die vorlauf-solltemperatur in grad C. Ich benutze unterschiedliche Werte in der Übergangszeit und im winter.

Woher habe ich die Infos.. Von mir stammte seinerzeit aus dem Jahre 2015/2016 die python Implementierung einer L2107 Anbindung, die auch lesen und schreiben konnte. Der thread im homematic forum ist von mir. Daher ist mir die Kommunikation und die telegramme bekannt. Da ich aber an der Stelle einiges umbauen muss wegen Solar bin ich nun auch auf die km271 wifi gewechselt. Ist ein super Projekt und ein Kasten samt pi ,Netzteil pegelwandler und switch etc an der Wand kann weg.

Als expertentelegramm dachte ich an ein topic in der Art /setvalue/experttelegramm was als message dann den 16 Zeichen hexstring für das telegramm innehat.

Gruss Michael

HMMike, 2024-01-21:
Hi,

das passende Rückmeldetelegramm ist das, welches ich einen Post drüber als BUG gemeldet hatte, da bei dem Rücklesen der Offset nicht stimmte. (im Quellcode die 0+4 anstatt der 0

der Defaultwert ist 3 Grad

EInstellbar an der Logamatik ist dieses nur, wenn du eine Fernbedienung angeschlossen hast

Kapitel 4.14 der Serviceanleitung

[image]

dewenni, 2024-01-21:
ah, ok!
Das hatte ich nicht in Verbindung gebracht. Bin gerade dabei das zu implementieren und wollte das dann auch gerne testen.

HMMike, 2024-01-21:
@dewenni 

via MQTT musstest dieses über das config Telegramm senden können. Nur die Logamatik wird den Wert nicht beachten und ausser beim Rücklesen selber wirst du das nicht an der Logamatik einstellen oder kontrollieren können

Bei mir ist das ja mit der Tag/Nachttemperatur so, ich habe eine Fernbedienung, da kommen die beiden Werte zwingend von der FB. Ich kann zwar die beiden werte beschreiben über ein Konfigtelegramm (mit der Original KM271 und meinem Python Programm) oder auch über MQTT mit deiner nun installierten SW,aber die FB überschreibt diese werte dann wieder bei ihrem nächsten Telegramm. Eigentlich müsstest du die beiden Werte auf der http Bedienseite rausnehmen, wenn die Fernbedienung aktiviert ist, weil die eh keinen Effekt dann haben. Dafür fann die Aufschalttemperatur anzeigen und änderbar machen, wenn FB aktiviert ist. Aber ich denke, das geht zu weit, hier werden alle ja wahrscheinlich eine Implementierung in ihr Automatisierungssystem gemacht haben und dort die werte Visualisieren und Änderbar haben.

Bekommst du auch meinen Wunsch nach dem selbst konfiguriertbaren expertentelegramm umgesetzt ?

dewenni, 2024-01-21:
Ich habe mal die Serviceschnittstelle und auch die zwei Befehle für den Reglereingriff implementiert.
Scheint auch prinzipiell zu funktionieren.

Service Schnittstelle erstmal nur per mqtt.
Beispiel:
```
Topic: esp_heizung/cmd/service
Payload: 00_11_22_33_44_55_66_77_88
```

Payload sind dann 8 Hex Werte durch "_" getrennt. Ich habe das auch so gut es geht auf Plausibilität getestet.
Es müssen immer die 8 Hexwerte sein und die Gesamtanzahl an Zeichen muss auch stimmen.

Für die beiden neuen Kommandos gibt es folgende Syntax:

```
command:    heating circuit 1: controller intervention
topic:      {"setvalue/hk1_reglereingriff", setvalue/hc1_ctrl_intervention"}
payload:    Resolution: 1 [°C] - Range: 0 ... +10 [°C]

command:    heating circuit 2: controller intervention
topic:      {"setvalue/hk2_reglereingriff", setvalue/hc2_ctrl_intervention"}
payload:    Resolution: 1 [°C] - Range: 0 ... +10 [°C]
```

Bitte mal testen und Rückmeldung geben.
[beta_3.2.x.zip](https://github.com/dewenni/ESP_Buderus_KM271/files/14002863/beta_3.2.x.zip)

dewenni, 2024-01-21:
Ich werde aber vermutlich den Begriff "Reglereingriff" nochmal in "Aufschalttemperatur" ändern.
Dann wird deutlicher, dass der Sollwert mit dem existierenden Config (Istwert) zusammengehört.

HMMike, 2024-01-22:
@dewenni 
könntest du mir die beta bitta als ova compilieren, dann teste ich gerne. Ich komme immo nicht an die Logamatic ran via kabel und hab so auch noch nie geflashed.

dewenni, 2024-01-22:
@HMMike in dem Archiv im Beitrag zuvor sind beide .bin Files drin. Ich habe sie nur zugegeben nicht getestet.

HMMike, 2024-01-22:
Ok. Dann kurze, dumme Frage. Ich mache das heute abend dann über vpn. Bin auf Dienstreise immo

Die für ova flash, ist das die die mit
Date_3.2.x.bin endet oder die
Flash_v3.2.x.bin ?

dewenni, 2024-01-22:
Die mit "Flash" in Namen ist für das Flashen per Kabel. Die andere Datei müsste auch ota im Namen tragen.
Wenn du nicht zuhause bist, würde ich das Update aber nicht einspielen. Es besteht immer die Gefahr das es nicht klappt. Dann hast du keine Werte.

HMMike, 2024-01-22:
ich hab das ota hochgeladen und update durchgeführt.

[image]

er meldet sich aber noch mit folgender firmware:
[image]

die Aufschalttemperatur wird immer rnoch fehlerhaft ausgelesen
änderung via mqtt nicht möglich

dewenni, 2024-01-22:
@HMMike dann hat er den Upload oder das Update abgebrochen.
Das gleiche hatte ich eben auch bei meinen Tests. Nach mehreren Versuchen ging es dann plötzlich und auch mehrfach hintereinander.

Möglicherweise liegt es daran, dass der Flashspeicher mit der Default Partitionierung nahezu voll belegt ist.
Daher werde ich mit dem nächsten Update das Partitionsschema anpassen um mehr Platz für das Programm und den OTA Upload zu bekommen. Das geht dann aber nur in dem man das Programm per Kabel aufspielt.

HMMike, 2024-01-22:
ok, nach dem hinweis habs auch ein paar mal probiert... ok, über vpn so ne sache, aber.. nach 4 versuchen hat er es gefressen.

status:
1. Wert aufschalttemperatur kommt nun richtig
2. wert lässt sich verstellen über MQTT
3. Servicetelegramm funktionierte auch (du hast in deinem Beispiel aber 9 Hexwerte, dürfen aber nur 8 sein)

also in der Form, alles supi

Gruss, Michael

dewenni, 2024-01-22:
Prima, dann gibt's die Tage ein offizielles Release dazu.
Danke für die Hilfe und falls du noch weitere Werte kennst die nicht umgesetzt sind, gib Bescheid.

HMMike, 2024-01-22:
DU kannst eigentlich jeden Wert de Konfig Tabelle setzen (Obs Sinn macht ist die andere Sache)

HK1 setzen:
07 Offset 6 Byte folge wie gehabt

HK2 setzen
08 [Offset des HK1] 6 Byte folge wie gehabt

der wwv beginnt mit 0C und folgt leider nicht der schönen regel,  
Ich weiss dass es da einen Parameter geben muss, der einen WWB Ladezyklus ausführt. Grund: an der FB kann man einen Taster anschliessen, der dann einen Ladezyklus anstösst. und da die FB mit ihren Telegrammen die Tag/Nachttemp überschreibt,  wird diese da in der gruppe wohl auch einen Wert setzen

dewenni, 2024-01-30:
solved with v3.2.5

closed 2024-01-30 by dewenni