dewenni/ESP_Buderus_KM271 #61: Aufschalttemperatur wird falsch gelesen

HMMike, 2024-01-10:
Hi,

Bei der IBN ist mir noch etwas aufgefallen. Zu dem Wunsch, die Aufschalttemperatur ändern zu können (siehe vorheriges Issue) fiel mir auf, das die Aufschalttemperatur schon falsch gelesen wird.

Im MQTT steht der Wert "AUS", definitiv eingestellt ist bei mir 2 Grad (Kontroliert an der Logamatik selber und nochmal mit der testweise rückgebauten KM271 und meinem alten Pi unter python

die Aufschalttemperatur findet sich in dem Telegramm: (vllt auch direkt im HK2 gucken)

HK1:
00 15 Aufschalttemp xx xx xx xx xx xx xx

HK2
00 4D AUfschalttemp2 xx xx xx xx xx xx xx

Gruss Michael

dewenni, 2024-01-10:
Hallo Michael,

danke für den Hinweis. Ich habe es mir auch gerade angeschaut und du hast völlig recht, auch in deiner Analyse von meinem Code.

data[2+4] muss geändert werden in data[2+0]

Das mit dem 2+x habe ich nur wegen der Lesbarkeit gemacht. In den ersten zwei Bytes steht die "Adresse" drin und danach kommen die eigentlichen Nutzdaten.

Kannst du es für dich selbst jetzt erstmal direkt selbst anpassen im Code und die geänderte SW auf das Board spielen, oder bist du auf ein Release angewiesen was du flashen kannst?

Aktuell fehlt mir noch etwas die Zeit an dem Projekt weiter zu machen. Die Zeit kommt sicher wieder, aber im Moment habe ich noch ein paar andere Dinge fertig zu machen.

HMMike, 2024-01-10:
Ich hab leider nicht die Erfahrung mit den ESPs, um mir da selber das Programm neu zu compilieren. Hab ich für die ESPs noch nie gemacht

Franck78, 2024-01-27:
@HMMike 
If you are able to build something for Arduino, this is exactly the same process for ESP32.
More, it is easy to use this code with Arduino IDE, that's what I did.

Tip: when viewing source code in github, click at the line number, click the three dot, use it to link the line.

dewenni, 2024-01-30:
solved with v3.2.5

closed 2024-01-30 by dewenni