dewenni/ESP_Buderus_KM271 #63: Mischer-Werte springen

solutionfound, 2024-01-17:
[image]

Ich sehe über MQTT in Operhab sehr komische Sprünge in en Mischerwerten.
Der Mischer selber springt physisch nicht so wild.
Wie muss man die Mischerwerte interpretieren oder liegt hier ein Fehler vor?

Grüße
-Marco

dewenni, 2024-01-18:
Gute Frage. Ich habe keinen Mischer verbaut um das zu Prüfen.
Ich habe gerade nochmal geschaut und laut der Doku die ich genutzt habe, sollte zumindest die Stelle im Telegramm passen. Das müsste ein Prozentwert sein und wird genauso ausgewertet die die Pumpenleistung von HK1 oder HK2.
HK1_Pumpe springt bei mir zwischen 0 und 100 und passt soweit.
Was macht bei dir der Wert für die Pumpe?

solutionfound, 2024-01-18:
Die Pumpenwerte springen von 0 zu 1 bzw. 100 und habe ich jetzt als Switch off/on interpretieren lassen in Openhab. Das passt wunderbar.
Wie man an der Grafik sieht, gehen die Werte von 0 vis 255. Es sieht so aus als springen die Werte von 255 auf 0 als würde eine 8 Bit Speicherzelle überlaufen.
Auf jeden Fall sind es keine Prozentwerte.

Iam-klm, 2024-01-27:
Hi Sven.
Ich habe zwei KM271 V0.0.6 am laufen, einmal ein Bud mit HK1 minimal'st und einen Bud mit HK2 mit WW und Mischer.

Habe im Moment deine 3.2.x, am Laufen, habe aber mir im UI auch schon mal selbst Anpassungen (minus Werte) gemacht.

Das mit den Mischerwerten kann ich bestätigen, habe das Verhalten mehrfach verfolgt.Zum Beispiel:
[image]
Die übertragenen HEX Werte waren ok. Habe mir die auch die Sourcen angesehen, ist mir aber nichts aufgefallen.
[image]
Wenn die Ladepumpe zuschaltet und natürlich die Umwälzpumpe abschaltet, zeigt der Mischermotor immer 156%

Auch das Brenner ein/aus (1/0) war beim Bud HK1 Ofen, schon für Stunden auf on obwohl an der Temperaturkurve zu sehen war, dass dem nicht so war und meine Ohren haben es bestätigt. Nach einem reset per Hand war es denn mal wieder gut. Hat sich dann aber später wieder mal eingestellt.

Für mich sieht es aus, als ob im Speicher Werte überschreiben werden, da das unschöne Verhalten bei den tiefen minus Werten häufiger als Jetzt auftrat.
Aber das hast du ja schon irgenwo erwähnt, dass du den Speicher umorganisieren möchtest.

Werde mal die #811b bei Gelegenheit mitschreiben.

Schon mal DANKE für deine Zeit.

Guß Klemens

Iam-klm, 2024-01-27:
OK Sven.

Es hat mir keine Ruhe gelassen.

Wenn ich jetzt über Tag den Bud per UI auf Hand Nacht bringe, knallt die #811b schlagartig durch.

[AM-Buderus-KM271_debug message.csv](https://github.com/dewenni/ESP_Buderus_KM271/files/14072071/AM-Buderus-KM271_debug.message.csv)

Da sind wirklich Werte höher als #64.
Der Mist kommt, so wie es aussieht aus dem Bud, warum auch immer, oder die Datenflut wird nicht richtig aufgearbeitet.
Nach dem Zurückschalten auf Automatik sind es dann #64

Klemens

Franck78, 2024-01-28:
@Iam-klm 
Hi, on my 2105, this hk2_Mischerstellung has no 'logical behaviour'.
I would say it's a bit field indicating something instantaneous ?
Probably not a bug on 2105 and 2107.

dewenni, 2024-01-30:
ich sehe hier erstmal nichts was ich an der Software tun könnte, daher schließe ich das Issue

closed 2024-01-30 by dewenni