dewenni/ESP_Buderus_KM271 #64: Solar-Werte in Oberfläche anzeigen und MQTT implementieren

solutionfound, 2024-01-17:
Hallo,
das ich auch die FM244 im System habe, wollte ich fragen, ob die Solar-Werte auch implementiert werden können.

Danke und Grüße
-Marco

dewenni, 2024-01-18:
Hallo,

mir sind leider keine weiteren Werte aus dem Protokoll bekannt.
Es gibt zwar noch ein paar Telegramme die noch nicht umgesetzt sind, aber dazu kenne ich keine Interpretation.
Wenn sich darin noch Werte für das Modul FM244 verbergen, dann könnte man dass gegebenenfalls noch durch probieren herausfinden.

Unbekannt sind noch folgende Teile: #6 #5 #4 #7 

Du könntest Werte für das FM244 ändern und dann schauen ob sich in diesen Telegrammen etwas verändert.
Ich habe dazu eine Debug Ausgabe im MQTT eingebaut die man aktivieren könnte

[image]

solutionfound, 2024-01-18:
Zu den unbekannten Teilen: wenn du kein Solar und FM244 installiert hast, halte ich es für unwahrscheinlich,  dass diese damit zu tun haben.

Das mit dem Ändern der Werte kann ich mal versuchen. Wie du den Debug mode einschaltest, ist mir nicht ganz klar.
Sofern der debug Modus an ist, kommen dann alle Werte als Debug_message oder nur die unbekannten?

dewenni, 2024-02-04:
@solutionfound In der neuesten Version kannst du den Debug Modus einfach selbst aktivieren. Zusätzlich kannst du auch einen Filter setzten um nur bestimmte Telegramme per mqtt zu senden.
Du könntest also mal einzelne Solar Werte an der Logamatic verändern und schauen welche Telegramme gesendet werden und ob du innerhalb des Telegramms deinen eingestellten Wert wiederfindest. Beachte aber das im Telegramm Hex Werte enthalten sind.

solutionfound, 2024-02-04:
@dewenni : cool! 
Allerdings habe ich noch ein paar Hürden zu überwinden: Ich habe gelesen, dass mindestens eine Version ab 3.2.5 per Kabel geflasht werden muss - leider habe ich kein Kabel dazu.
In weiser Voraussicht hatte ich aber eine Version mit eingelötetem Debug Header von Daniel geordert.
Hast Du einen Tipp für ein Kabel? 

Nächste Hürde ist: Ich muss erst mal wieder zur Wohnung meiner Mutter kommen - die ist gerade auf Reha, d.h. vor Mitte Februar komme ich sicher nicht vorbei - was mir auf der anderen Seite  Zeit gibt ein Kabel zu organisieren ;-)

Also nicht wundern, wenn es etwas länger dauert.

Daniel hatte  ja geschrieben, dass die letzten Boards mehr Speicher haben, ist die Speichereinschränkung trotzdem relevant?

dewenni, 2024-02-04:
Ich hatte bisher ein sehr einfachen Adapter wie diesen hier: https://eckstein-shop.de/32C3V2F5VTypAUSBtoTTLKonverterUART5PinSerialSTCChipCP2102ModuleArduino

Du findest da im Shop aber auch noch einige andere Varianten die vermutlich genauso funktionieren.
https://eckstein-shop.de/search/?qs=usb+ttl

Bei dem 4MB Boards war der Speicher bei dem Default Partitions-Schema schon >90% voll.
Mit dem anderen Partitions-Schema bin ich jetzt wieder bei ca. 56%
Bei einem 8MB Board ist das vermutlich kein Problem, aber ich habe auch keines hier um das nachzuvollziehen.

the78mole, 2024-02-04:
Naja, das Problem ist, wenn wir das Partitionsschema von 4 auf 8 MB ändern, dass dann einige der älteren Module nicht mehr updatebar sind. Eine gangbare Varainte wäre, verschiedene Varianten der Firmware zu bauen, eine Full-Blown für die User mit 8MB und eine feature-stripped-Variante, die noch ins 4 MB-Flash passt und auch das andere PArtitionsschema verwendet. Aber der Release-Aufwand ist halt deutlich höher und das bleibt dann leider bei dewenni hängen.
Nun könnte man als erstes mal einen GitHub Actions workflow aufsetzen, damit das Release wenigsten automatisch funktioniert. Das würde dewenni vermutlich schon stark entlasten. Daraus dann mehrere Outputs mit unterschiedlichen Settings/Partition-Schemes zu erstellen ist wahrscheinlich kein großer Aufwand mehr.
@dewenni, was denskt Du?

dewenni, 2024-02-04:
Ich denke das ist nicht nötig.
Die 4MB genügen ja aktuell auch.
Nur werden die 4MB mit dem Standard Partitions-Schema so aufgeteilt, dass nur ca 1,3MB für Applikation frei sind. Das wurde jetzt eben schon knapp.
Ich habe das Schema daher auf "min_spiffs" geändert und jetzt gibt es auch bei den 4MB bis zu 1,9MB für die Applikation.
Diese 1,9MB sind jetzt aktuell nur zu 56% belegt. Der mehr Speicher kommt daher, das jetzt die SPIFFS Partition kleiner ist. Das ist aber egal, da liegt nur die Config Datei.

Um das Schema zu ändern muss man halt einmal neu flashen.
Danach kann man ja wieder OTA machen.

solutionfound, 2024-05-18:
Heute habe ich 4.1.0 geflasht - von Version 3.2.5. Interessanterweise muss ich OTA immer 2x den Flashvorgang durchführen und beim 2. Mal bootet das System in der neuen Version. Aber das nur als Randbemerkung.
Mit der neuen Version habe ich gleich das Feature der unbekannten Datagramme genutzt und eine Korrelation mit den Solar-Werten in der Oberfläche zu machen.
Folgendes habe ich herausgefunden:
91_46 ist wohl die Kollektor-Temperatur mit dem Sensor, der im Modul verbaut ist. Es kommen viele Diagrammen, die sich nur um einen Wert bei dem dritten Hex-Wert verändern und dazu noch mit der Anzeige übereinstimmen. Ich denke das passt ;-)
91_46_37_00_00_00_00_00_ff_00_00 --> Kollektor Temperatur 55 °C

Anderer Wert ist 91_44 der mit WW-Solar übereinstimmt.
91_44_32_00_00_00_00_00_ff_00_00 --> WW-Solar 50 °C

Ansonsten habe ich gerade nur noch 
91_42 der nur 01 oder 00 ist und daher vsl. ein Schaltzustand abbildet. Vielleicht die Ladungspumpe für WW-Solar? Da muss ich mal die Werte etwas genauer mit den Temperaturwerten korrelieren.

91_47 --> da läuft der Wert von Hex 10 über die Zeit herunter bis auf 00. Nur die Werte Hex 09, 06, 03 werden nicht aufgeführt. Vielleicht irgendein Temperatur offset? Ich habe die Anlage so eingestellt, dass wenn es WW-Solarertrag gibt, der untere Schwellwert für die WW-Aufbereitung weiter herunter gesetzt wird.
Da da die Werte gegen Abend aber auf 00 gehen ist es vielleicht auch irgendetwas mit "Solar-Ertrag in Grad"? 

91_4a --> ist an einigen Zeiten fast identisch mit 91_46 Kollektor Temperatur, aber wird dann irgendwann nicht mehr gesendet. Gerade als die Sonne stark scheinte ist 91_4a recht gering und steigt dann an bis auf Dec 60 und wird dann nicht mehr gesendet.

Da ich nur kurz Zeit hatte, habe ich noch keine Werte verändert und dann überprüft welche Datagramme gesendet werden. Das versuche ich das nächste Mal, da die Anlage bei meiner Mutter steht.

@dewenni : ist es für Dich aufwändig zumindest schon mal die beiden Datagrammtypen zu verarbeiten und per MQTT und ggf. auch über die Oberfläche bereitzustellen?
Wenn die nicht mehr als unbekannte Datagramm in der Liste auftauchen wird es auch übersichtlicher beim "forschen".

Viele Grüße

dewenni, 2024-05-19:
@solutionfound 

cool, danke für die Analyse. Gerne füge ich weitere Werte hinzu. In der Oberfläche würde ich dafür am besten noch eine extra Seite machen. Bei bisher "nur" zwei Werten wäre das jetzt noch etwas wenig. Aber wenn du noch ein paar mehr Werte herausfindest, dann machen ich das gerne.

Hier sind die Werte vom Typ 91 auch erwähnt, allerdings ohne weitere Erklärung
https://www.mikrocontroller.net/topic/141831?page=single#1526149

Als MQTT Werte kann ich die beiden aber gerne schon mal aufnehmen.
Dann müssen wir noch noch herausfinden warum das bei dir mit den [Einstellungen](https://github.com/dewenni/ESP_Buderus_KM271/issues/98#issuecomment-2118821663) so komisch ist.

dewenni, 2024-05-19:
Ich könnte dir auch eine "spezielle" Version machen, wo alle 91er Werte per mqtt gesendet werden.
Dann könntest du die ggf. besser auswerten und vielleicht auch aufzeichnen, um dann besser die Zusammenhänge herauszufinden?!

dewenni, 2024-05-19:
Die Einstellungen wären dann auch noch interessant herauszufinden.
https://www.manualslib.de/manual/15526/Buderus-Logamatic-2107.html?page=42#manual

Solar EIN/AUS - default EIN
MAX Solar - 60..90°C - default 75°C
MIN Solar - 30..45°C AUS - default AUS

Du könntest also bitte mal die 3 Werte verändern und schauen welche Telegramme das sind.
Ist vielleicht eines dieser Telegramme #4 - sollte denke ich jedenfalls mit Adresse 0x anfangen.

solutionfound, 2024-05-19:
> Ich könnte dir auch eine "spezielle" Version machen, wo alle 91er Werte per mqtt gesendet werden. Dann könntest du die ggf. besser auswerten und vielleicht auch aufzeichnen, um dann besser die Zusammenhänge herauszufinden?!

Das wäre sehr cool - denn die Temperaturwerte erzeugen viel "traffic" und ich könnte die mit Wassertemperatur und Außentemperatur in Diagramme aufnehmen.

Vielleicht kannst Du die noch nicht komplett verstanden Werte irgendwie mit einem Draft-Topic in MQTT aufnehmen, dann kann ich die zusammen in Graphen darstellen, das würde helfen zu versehen was da gemessen wird bzw. bei 91_42 was bzw. wann der Wert von 00 auf 01 geht.

Ich hoffe, dass das Risiko mittlerweile gering ist, dass ich den Zugriff verliere, da die Version 4.1.0 ja zu funktionieren scheint. Davor musste ich immer wieder mit dem seriellen Adapter zurück flashen.

Bzgl. Verändern der Einstellungen: Das werde ich sicher das nächste Mal machen wenn ich vor Ort bin. Die Anlage ist halt 100km weit entfernt - daher verbinde ich das eben wenn ich bei meiner Mutter bin.
Zusätzlich zu den von Dir erwähnten drei Settings, die man im Service-Modus einstellt, kann man auch Solar auf manuell und auto stellen. Denn Sinn habe ich noch nicht verstanden, aber das wird auch noch Datagramme triggern schätze ich.

Anekdote: Was ich nicht geschafft habe ist die Solar-Anlage in Betrieb zu bekommen wenn der Sensor im Kollektor zu hohe Werte hat. Ich glaube ich hatte es sogar mit dem "Relaistest" nicht geschafft die Pumpe in der Solaranlage zu starten. Ist aber vielleicht auch nur ein "Randthema". Konkret ist mir das in zwei Fällen passiert als ich einmal die Heizung wegen Wartung (Wechsel der Zirkulationspumpe) und einmal Ende des Urlaubsmodus nicht in Dauerbetrieb hatte. Beide Male war es sehr sonnig und daher der Temperaturwert hoch im Kollektor, der logischerweise sofort runter gehen würde, wenn die Ladungspume das Fluid umpumpen würde. Konsequenz war das an dem Tag das Brauchwasser unützerweise mit dem Brenner auf Temperatur gebracht wurde und erst am nächsten Tag durch Abkühlung in der Nacht die Anlage wieder wie gewünscht funktionierte.

dewenni, 2024-05-24:
Ich hab dir jetzt mal eine Testversion gebaut, wo die vermeintlichen Solarwerte als MQTT gesendet werden

[image]

Die tauchen also jetzt nicht mehr bei den "undefined_messages" auf.

Außerdem habe ich am Webserver mal testweise noch was ergänzt, dass den Browser dazu zwingen soll, dass er keinen Inhalt cached. Bitte berichte mal ob das was an deinem beobachten Verhalten bezüglich der Settings ändert.
Ich konnte das bisher nicht nachvollziehen.

[buderus_km271_v4.1.7000.zip](https://github.com/dewenni/ESP_Buderus_KM271/files/15436887/buderus_km271_v4.1.7000.zip)

solutionfound, 2024-05-24:
Danke Dir. Ich habe die neue Version installiert und sie lief auch ca 2,5 Stunden - jetzt ist die Verbindung abgebrochen. Ich war gerade dabei MQTT neu einzurichten in OpenHab um die Daten anzeigen zu lassen. Ob das damit etwas zu tun hat, kann ich aber nicht sagen.
Ich werde moren mal mein Mutter bitten die Heizung aus und wieder einzuschalten. Wenn er dann nicht mehr ins Netz kommt muss ich warten bis ich wieder vor Ort bin.

Was ich aber sagen kann, dass es keine Änderungen bzgl. der Einstellungen gab. Wenn ich das Device im Browser im Private-Modus aufgerufen habe, sind die Einstellungen der Logamatic nicht da gewesen. Sehr komisch, dass Du das nicht nachstellen kannst.

solutionfound, 2024-05-25:
@dewenni 
Mit einem Powercycle konnte ich wieder auf die den esp zugreifen und habe bekomme auch die MQTT Daten.
91_4a scheint ein Zähler zu sein, da er ziemlich genau im 60 Sekundentakt um eins hoch zählt. Nach 255 startet er wieder von 1.

Leider ist nach ca. 4,5 Stunden der esp wieder eingefrohren.
Ich habe meine Mutter vorhin nochmals die Heizung aus und einschalten lassen worauf ich wieder Zugriff hatte, aber diesmal nur 35 mins - dann war der esp wieder nicht erreichbar.
Ich habe gelesen unter den issues #104, dass die Stabilität Probleme bereitet. Die 4.1.0 lief bei mir fast eine Woche ohne Probleme.

Einziger Unterschied, der mir einfällt. Ich habe die 4.1.0 2x geflasht und die von die bereitgestellte 4.1.7 nur 1x... keine Ahnung, ob das etwas ausmacht.

Mit der aktuellen Stabilität wird es aber schwierig mal längere Werte zu erfassen und ich kann meine Mutter nicht fortwährend in den Keller schicken... vielleicht gehe ich zurück auf die 4.1.

dewenni, 2024-05-26:
Tja, das mit der Stabilität ist auch seltsam. Ich habe da überhaupt keine Probleme mit. Mein Board seit 4.x noch kein einziges mal "eingefroren".

In der 4.1.7000 die ich dir gegeben habe, sind auch nur die zusätzlichen MQTT Nachrichten für das FM244 und eine Browser Option bezüglich Caching dazu gekommen. Ich wüsste nicht warum das etwas an der Stabilität ändern sollte. Das war jetzt nur eine "Spezialversion" basierend auf der 4.1.0.
Das muss noch irgend einen anderen Zusammenhang haben.

Hast du das Board per USB versorgt oder nur über die Buderus?
Falls du nur über die Buderus versorgst, könntest du ja mal testen ob es über USB besser wird.
Da könntest du zur Not auch mal eine schaltbare Steckdose zwischen schalten um aus der Ferne einen PowerCycle durchzuführen. Oder du machst einen Shelly oder ähnliches an die Heizung. Ich habe einen Shelly in der Verkleidung der Logamatic versteckt. Damit kann ich den Stromverbrauch messen und ggf. auch mal neu Starten ohne in den Keller zu müssen.

solutionfound, 2024-05-26:
Das Board wird über die Buderus versorgt und ich kenne die Thematik mit bei manchen Modellen nicht ausreichender Stromversorgung. Ich hatte bei früheren Versionen, wo das Board nicht mehr stabil lief zeitweise per USB versorgt. Das brachte aber keine Besserung, daher habe ich die wieder abgebaut, weil 3.2.5 rock solid lief.

Ein Shelly oder ähnliches wäre die ultima ratio. 
Ich probiere heute die 4.1.7 nochmals ein zweites mal aus der ferne upzudaten. 

Ist es nicht so, dass der esp da irgendwie immer zwei Versionen gleichzeitig im Dateisystem/Partitionen liegen hat und bei Problemen auf die andere zugreift? Das war zumindest mal ein Thema bei dem Device, das ich für meine Gasheizung verwende: emsesp.github.io/docs
Da war der Effekt, dass nach einem Reboot bei manchen die alte SW wieder da war. Mich hat das auch mal betroffen.

Ich habe im Keller schwaches WLAN. Gestern habe ich gesehen, dass als reboot Grund keine Verbindung zu MQTT angegeben war - vielleicht bootet das Gerät und nimmt dann irgendwann die andere SW, die korrupt ist.
Ich kenne mich mit dem esp und der SW-Architektur leider nicht aus, daher alles nur Gedanken, die ich mir zusammenreime.

dewenni, 2024-05-26:
hmm, ok.

> Ich habe im Keller schwaches WLAN. Gestern habe ich gesehen, dass als reboot Grund keine Verbindung zu MQTT angegeben war

Wenn die MQTT Verbindung verloren geht, dann versuche ich 5x neu zu verbinden und starte dann den ESP neu.
Könnte auch sein, dass er sich dann aus irgend einem Grund gar nicht mehr verbindet.

Der Plan ist, demnächst auch eine Unterstützung für einen Ethernet Anschluss zu haben. Hat zwar auch nicht jeder im Keller, aber es wird dem Einen oder Anderen sicher auch helfen. Ich würde dann selbst auch auf Ethernet gehen. Ab und an habe ich nämlich auch Probleme mit dem WLAN im Keller.

solutionfound, 2024-05-26:
Wie ich unter #104 geschrieben habe, hat das 2. Mal flashen wohl geholften. Der esp läuft seitdem und daher konnte ich ein paar Weitere Werte identifizieren:

91_42 Solar Ladung ein=1 / aus=0 
Schaltet sich ein wenn Kollektortemperatur 10°C höher ist als Solar-WW und schaltet aus wenn Kollektortemperatur auf 5°C über Solar WW fällt

Es ist auch konsistent mit 91_4a - wenn die Ladung ein ist, dann zählen die Minuten dort hoch. Wenn 91_4a von 255 auf 0 springt, dann wird 91_49 um eins hochgezählt. Was mich zu er Annahme brachte, dass es die üblichen 3 Werte für die Betriebsstunden sind. Ich habe die Werte mal in Stunden umgerechnet und das passt mit dem Betriebsstunden zusammen, die ich letzte Woche abfotografiert hatte. Daher ist
91_49 --> Multiplikator für 256 Minuten
91_48 --> Multiplikator für 65536 Minuten 

Ich konnte heute keine weiteren unbekannten Datagramme sehen. 
91_4b bis 91_4d sind bis jetzt nicht aufgetreten.

91_47 konnte ich noch nicht zuordnen. Anbei der Verlauf des heutigen Tages - vielleicht hast Du oder jemand anderes eine Idee.

[image]

dewenni, 2024-05-31:
ok, dann lass uns mal zusammenfassen:

**Status Werte:**

| Adresse | Beschreibung           | Einheit       |
|---------|------------------------|---------------|
| 9142    | Solar Ladung           | ein=1 / aus=0 |
| 9143    |                        |               |
| 9144    | WW-Solar               | °C            |
| 9145    |                        |               |
| 9146    | Kollektor Temperatur   | °C            |
| 9147    |                        |               |
| 9148    | Laufzeit Minuten 65535 | min*65535     |
| 9149    | Laufzeit Minuten 256   | min*256       |
| 914a    | Laufzeit Minuten       | min           |
| 914b    |                        |               |
| 914c    |                        |               |
| 9145    |                        |               |

**Config Werte:**

TBD...

solutionfound, 2024-05-31:
Hallo, 

komme gerade vom Besuch bei der Heizung ;-) 
Ich habe übrigens keinen einzigen Absturz seit letztem Sonntag. Das nochmalige Flashen half nachhaltig.
Ich habe Deine Tabelle etwas ergänzt. Dass kein Wert gesendet wird bis dato und dass etwas gesendet wird, aber die Interpretation offen ist, hat auch etwas Informationswert.

Adresse | Beschreibung | Einheit
--|--|--
9142 | Solar Ladung | ein=1 / aus=0
9143 | _(bis jetzt keine Werte empfangen)_  |  
9144 | WW-Solar | °C
9145 |  _(bis jetzt keine Werte empfangen)_ |  
9146 | Kollektor Temperatur | °C
9147 | _Werte empfangen; Bedeutung unklar_  |  
9148 | Laufzeit Minuten 65535 | min*65535
9149 | Laufzeit Minuten 256 | min*256
914a | Laufzeit Minuten | min
914b | _(bis jetzt keine Werte empfangen)_ |  
914c |  _(bis jetzt keine Werte empfangen)_ |  
914d | _(bis jetzt keine Werte empfangen)_ |  

**Weitere Erkenntnisse:**
Schalten des "Heizkreises Solar" über die Tasten Auto, Tag und Nacht analog wie bei HK1 und HK2 und WW
01_d9_28_0a_01_ff_ff_ff_ff_00_00 --> Solar Tag=Manuell (5. Hex-Paar 01)
01_d9_28_0a_02_ff_ff_ff_ff_00_00 --> Solar Automatik ((5. Hex-Paar 02) ist auch der Default-Wert - daher wird der bei Dir wahrscheinlich nach dem Einschalten gesendet - siehe #5 
01_d9_28_0a_00_ff_ff_ff_ff_00_00 --> Solar Nacht=Aus nicht im Setup-Menü (5. Hex-Paar 00)

**Im Setup-Modus:** --> erklärt auch ein weiteres Datagramm in #5 
01_d2_78_0a_05_26_46_00_ff_00_00 --> Solar aus im Setup-Menü (8. Hex-Paar 0)
01_d2_78_0a_05_26_46_01_ff_00_00 --> Solar ein im Setup-Menü (8. Hex-Paar 1)

01_d2_78_0a_05_26_44_01_ff_00_00 --> Max Solar 68 °C (7. Hex-Paar 44)
01_d2_78_0a_05_26_45_01_ff_00_00 --> Max Solar 69 °C (7. Hex-Paar 45)
01_d2_78_0a_05_26_46_01_ff_00_00 --> Max Solar 70 °C (7. Hex-Paar 46)

01_d2_78_0a_05_25_46_01_ff_00_00 --> Min Solar 37 °C (6. Hex-Paar 25)
01_d2_78_0a_05_26_46_01_ff_00_00 --> Min Soart 38 °C (6. Hex-Paar 26)

Mehr Einstellungen kenne ich für die Solarfunktion nicht. 
Damit bleibt aktuell nur offen was 9147 bedeutet.

Meinst Du dass die noch nicht empfangenen Datagramme wirklich existieren? Die könnten auch einfach nicht verwendet sein.

dewenni, 2024-08-31:
@solutionfound 
gibt es hier noch neue Erkenntnisse?  
Ansonsten würde ich demnächst mal schauen das ich das in ein Release packe.

solutionfound, 2024-09-03:
@dewenni 
ich habe jetzt nicht mehr intensiv nach noch neuen unbekannten Datenpaketen geschaut - gestern hatte ich das Logging mal wieder eingeschalten, aber dann hat sich das System aufgehängt.
Gerade habe ich alle oben als (bis jetzt keine Werte empfangen) gekennzeichneten Werte überprüft, da ich diese per MQTT eingebunden hatte, aber da ist bis jetzt auch keine einziger Wert empfangen worden.

Bei 91_47 empfange ich stets Werte, die nur größer 0 sind, wenn auch die Kollektortemperatur hoch genug ist um das Wasser zu heizen, aber ich habe bis dato noch keine echte Interpretation der Werte gefunden.

Soll ich hier mal ein paar weitere Grafiken mit dem Kurvenverlauf posten, damit vielleicht jemand anderes noch eine Idee hat?

Die Tabelle von oben ist damit noch der letzte aktuelle Stand und Du könntest damit das nächste Release erstellen.
Sofern Du Namen einführst für die Werte, bitte diese nennen, damit ich Openhab entsprechend anpassen kann.

Ich bin gespannt, was das neue Release noch bringt und das komische Verhalten, dass die Konfig verloren geht jedes mal habe ich immer noch mit jedem Rechner / Browser.

dewenni, 2024-11-15:
@solutionfound 
ich habe das jetzt mal für das anstehende Release eingebaut.

**Einstellungen:**  

[image]

**neue Solar Seite:**  

[image]

**Dashboard:**  

[image]

MQTT (Beispiel deutsch):
```
esp-km271-test/config/Solar_Aktivierung
esp-km271-test/config/Solar_Betriebsart
esp-km271-test/config/Solar_Maximaltemperatur
esp-km271-test/config/Solar_Minimaltemperatur
```

```
esp-km271-test/status/Solar_Warmwasser
esp-km271-test/status/Solar_Laufzeit
esp-km271-test/status/Solar_Ladung
esp-km271-test/status/Solar_Kollektor
esp-km271-test/status/Solar_9147
```

Den Wert aus 0x9147 habe ich mal mit aufgenommen und ihn unbekannterweise auch 9147 genannt.
Vielleicht ergibt sich ja noch was das ist.

Passt das so aus deiner Sicht?

solutionfound, 2024-11-15:
@dewenni  - sieht sehr cool aus - das Icon finde ich auch sehr passend.

Ein paar Nachfragen / Anmerkungen:
Bei MQTT - rechnest Du die Laufzeit schon um, d.h. es wird nur einen Wert geben, der gesendet wird? Welches Format hat der? Sollte genauso sein wie der sonstige (Brenner-)Laufzeitwert.

Zum Wert Solar_Minimaltemperatur - der wird in der Anzeige / Anleitung auch Solar Min genannt. Aber ich finde den Begriff semantisch etwas unglücklich. Der Wert bewirkt, dass erst bei unterschreiten dieser Temperatur mit dem Kessel nachgeladen wird. D.h. es ist eine Festlegung auf eine untere Warmwassertemperatur des sonst (hartkodierten) delta -5°C Schwellwertes gegenüber der WW-Solltemperatur für das Aufheizen des Warmwassers. Vielleicht könnte man den zumindest in der Oberfläche anders nennen, da der Wert die WW-Termperaturwerte "überschreibt". Vielleicht so ähnlich wie "WW Minimaltemperatur" bei Solarbetrieb und bei dem Max-Wert entsprechend. Aber das ist wirklich nur eine Feinheit. Du kannst es gerne auch so lassen.

Nächste Woche am Samstag bin ich wieder bei der Anlage, wenn Du bis dahin fertig bist mit dem Release, könnte ich es vor Ort flashen - ist vielleicht sicherer bei so einem "major" Release ;-)

dewenni, 2024-11-16:
@solutionfound 
ja die Laufzeit habe ich bereits zusammengerechnet. Die einzelnen Werte kann man ja nicht wirklich gebrauchen.
Es ist also eine Summe in Minuten.

```
kmStatus.SolarOperatingDuration_Sum =
          kmStatus.SolarOperatingDuration_2 + (kmStatus.SolarOperatingDuration_1 * 256) + (kmStatus.SolarOperatingDuration_0 * 65536);
```

Am besten ich nenne es genauso wie in der Bedienungsanleitung
- Solar EIN
- Max Solar
- Min Solar

dewenni, 2024-11-23:
@solutionfound 
vielleicht kommt es ja noch rechtzeitig heute: release [v5.0.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v5.0.0)

solutionfound, 2024-11-24:
@dewenni  - gerade installiert OTA - hat wunderbar funtioniert.
MQTT muss ich noch umbauen, damit die Soarwerte wieder funktionieren.

Folgendes ist mir aufgefallen: Es werden bei der Konfiguration keine Werte angezeigt.
[image]

dewenni, 2024-11-24:
@solutionfound 
Danke für das schnelle Feedback.
Kommen die Werte auch nicht wenn du einen Browser Refresh machst?

solutionfound, 2024-11-24:
@dewenni 
Wenn ich einen Browser-Refresh mache, wird die Seite mit dem Dashboard geladen, aber immer noch Solar im Handbetrieb Nacht und die Solarwerte bei Konfiguration fehlen weiterhin.

Auch ist positiv, dass jetzt der Fehler verschwunden ist, dass die Einstellungen verloren gehen. Bis jetzt mit dem Laptop und Handy überprüft, wird Solar und alle andere Heizkreise angezeigt. Das war sogar direkt nach dem Update so. Ich musste nur Solar zusätzlich einschalten, weil es das ja vorher nicht gab.

solutionfound, 2024-11-24:
@dewenni 

Ich sehe gerade dass ich nach dem Upgrade ca. 40 sek. eine 31 unbekannten Datageramme bekommen habe - soll ich die hier oder wo anders bereitstellen?

dewenni, 2024-11-24:
Interessant.
Kannst du mal prüfen ob sich die Werte aktualisieren wenn du Einstellungen direkt an der Heizung änderst?
Also Solar Min, Solar Max oder Solar Betriebsart ändern.

Ich habe eben an meinem Testsystem nochmal getestet und manuell ein Telegramm für Solar Betriebsart gesendet und dann ändert sich auch die Anzeige im WebUi.

Bevor du wieder weg fährst, wäre es auch ganz hilfreich wenn ich einen Log von den Telegrammen nach einem Neustart bekommen könnte. Entweder per MQTT abgreifen oder in der WebUI den Mode auf Logamatik Werte stellen und nach einem Neustart kopieren.

solutionfound, 2024-11-24:
@dewenni 
Erst mal super vielen Dank, dass Du Dich so kümmerst!!!

Wenn ich die Werte für Auto, Tag und Nacht ändere werden sie auch erkannt und angezeigt - sowohl im Log wie auch in der Oberfläche.
Dann habe ich die Heizung vom Strom getrennt und wieder angeschaltet und dann waren alle Werte da. Daher habe ich jetzt Min und Max nicht noch verstellt und getestet.
Wahrscheinlich hätte ich die Heizung einmal Aus- und Einschalten müssen, damit die Werte gesendet und initialisiert werden nach dem Update.

Das Log hatte ich eingeschaltet und habe am Handy viele Werte gesehen. Jetzt wollte ich am Laptop die Werte speichern, aber da waren dann nur die neuen Werte und am Handy waren die älteren auch weg... wahrscheinlich muss ich das mal nur mit dem Rechner machen.
Das kommt dann am Nachmittag und sicher bevor ich fahre.

Die unbekannten Datagramme von heute Nacht kann ich schon schicken, weil ich gespeichert hatte:
[24.11.2024 - 00:53:31] undef msg : 91_4d_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:31] undef msg : 91_4c_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:31] undef msg : 91_4b_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:29] undef msg : 91_45_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:28] undef msg : 91_43_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:28] undef msg : 89_41_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:26] undef msg : 88_3b_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:26] undef msg : 88_3a_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:25] undef msg : 88_39_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:24] undef msg : 88_35_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:24] undef msg : 88_34_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:20] undef msg : 81_23_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:19] undef msg : 81_22_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:19] undef msg : 81_21_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:18] undef msg : 81_1d_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:18] undef msg : 81_1c_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:15] undef msg : 80_11_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:15] undef msg : 80_10_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:15] undef msg : 80_0f_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:14] undef msg : 80_0b_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:13] undef msg : 80_0a_00_00_65_65_65_65_00_00_00
[24.11.2024 - 00:53:02] undef msg : 00_b6_0f_0f_64_05_01_04_00_00_00
[24.11.2024 - 00:53:02] undef msg : 00_af_0f_0f_08_07_07_08_00_00_00
[24.11.2024 - 00:53:01] undef msg : 00_8c_65_65_65_65_65_65_00_00_00
[24.11.2024 - 00:53:00] undef msg : 00_77_65_65_65_65_65_01_00_00_00
[24.11.2024 - 00:52:59] undef msg : 00_62_28_0a_0a_11_65_65_00_00_00
[24.11.2024 - 00:52:59] undef msg : 00_5b_65_65_65_65_65_65_00_00_00
[24.11.2024 - 00:52:58] undef msg : 00_3f_65_00_05_05_2d_01_00_00_00
[24.11.2024 - 00:52:57] undef msg : 00_2a_28_0a_0a_11_65_65_00_00_00
[24.11.2024 - 00:52:57] undef msg : 00_23_65_65_65_65_65_65_00_00_00
[24.11.2024 - 00:52:56] undef msg : 00_07_65_00_05_05_2d_01_00_00_00

dewenni, 2024-11-24:
Ok, das ist einigermaßen verständlich.

Wenn der ESP startet wird ein "LogCmd" an die Heizung gesendet worauf diese dann Inder Regel alle Daten sendet (Config und Status Werte)
Ich speichere diese Daten dann in internen Strukturen.

Die Config Werte werden dann auf Änderung überwacht und wenn sich ein Wert ändert, dann werden alle Config Werte aus der internen Struktur an die Webclients gesendet.

Jetzt könnte es sein, dass nach dem Update die Solar Werte erstmal noch nicht von der Heizung gesendet wurden.

Aber eigentlich sollten die auch bei einem Neustart vom ESP durch das LogCmd gesendet werden.

Schau also auch nochmal ob es jetzt passt wenn der ESP neu startet oder ob du immer die Heizung neu starten musst.

Bisschen komisch ist das noch.

solutionfound, 2024-11-24:
Hier die Werte, wenn ich den ESP neu starte - da werden eigentlich die Parameter für Solar mitgeschickt und das Web-Ingerface zeigt auch alles an. Komisch - ich glaube wir werden den singulären Fehler direkt nach dem Update nicht (mehr) finden... Aber jetzt läuft ja auch alles. 

[24.11.2024 - 18:19:20] Status: HK2_Mischerstellung = 249
[24.11.2024 - 18:19:16] Status: HK2_Mischerstellung = 246
[24.11.2024 - 18:19:16] Status: HK2_Vorlaufisttemperatur = 33
[24.11.2024 - 18:19:12] Status: HK2_Mischerstellung = 27
[24.11.2024 - 18:19:09] Status: HK2_Mischerstellung = 24
[24.11.2024 - 18:19:09] Status: HK2_Vorlaufisttemperatur = 32
[24.11.2024 - 18:19:08] Status: HK1_Vorlaufisttemperatur = 38
[24.11.2024 - 18:19:05] Status: HK2_Mischerstellung = 3
[24.11.2024 - 18:19:01] Status: HK2_Mischerstellung = 2
[24.11.2024 - 18:18:54] Status: WW_Isttemperatur = 55
[24.11.2024 - 18:18:51] Status: Aussentemperatur = 7
[24.11.2024 - 18:18:47] Status: Aussentemperatur = 6
[24.11.2024 - 18:18:47] Status: WW_Isttemperatur = 54
[24.11.2024 - 18:18:46] Status: HK2_Mischerstellung = 255
[24.11.2024 - 18:18:41] Status: Solar_Laufzeit = 1255739
[24.11.2024 - 18:18:40] Status: Solar_9147 = 0
[24.11.2024 - 18:18:40] Status: Solar_Kollektor = 7
[24.11.2024 - 18:18:39] Status: Solar_Warmwasser = 30
[24.11.2024 - 18:18:39] Status: Solar_Ladung = 0
[24.11.2024 - 18:18:38] Status: Modulkennung = 14
[24.11.2024 - 18:18:38] Status: Versionsnummer_NK = 0
[24.11.2024 - 18:18:37] Status: Versionsnummer_VK = 4
[24.11.2024 - 18:18:37] Status: Aussentemperatur_gedaempft = 8
[24.11.2024 - 18:18:37] Status: Aussentemperatur = 7
[24.11.2024 - 18:18:36] Status: Brenner_Laufzeit_Summe = 1442249
[24.11.2024 - 18:18:36] Status: Brenner_Laufzeit_Minuten = 201
[24.11.2024 - 18:18:36] Status: Brenner_Laufzeit_Minuten256 = 1
[24.11.2024 - 18:18:35] Status: Brenner_Laufzeit_Minuten65536 = 22
[24.11.2024 - 18:18:35] Status: Abgastemperatur = 255
[24.11.2024 - 18:18:34] Status: Brenner_Ansteuerung = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_BetriebStufe2 = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Leistung_hoch = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Leistung_frei = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Unter_Betrieb = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Kesselschutz = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Betrieb_Stufe1 = 0
[24.11.2024 - 18:18:34] Status: Kessel_Betrieb_Abgastest = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Externe_Stoerung = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Sicherungskette = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Abgas_max_Grenzwert = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Abgasfuehler = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Kessel_bleibt_kalt = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Zusatzfuehler = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Kesselfuehler = 0
[24.11.2024 - 18:18:34] Status: Kessel_Fehler_Brennerstoerung = 0
[24.11.2024 - 18:18:33] Status: Brenner_Ausschalttemperatur = 46
[24.11.2024 - 18:18:33] Status: Brenner_Einschalttemperatur = 32
[24.11.2024 - 18:18:33] Status: Kessel_Vorlaufisttemperatur = 39
[24.11.2024 - 18:18:32] Status: Kessel_Vorlaufsolltemperatur = 39
[24.11.2024 - 18:18:32] Status: WW_Pumpentyp_Absenkung_Solar = 0
[24.11.2024 - 18:18:32] Status: WW_Pumpentyp_Zirkulationspumpe = 0
[24.11.2024 - 18:18:32] Status: WW_Pumpentyp_Ladepumpe = 0
[24.11.2024 - 18:18:32] Status: WW_Einschaltoptimierung = 0
[24.11.2024 - 18:18:32] Status: WW_Isttemperatur = 55
[24.11.2024 - 18:18:31] Status: WW_Solltemperatur = 55
[24.11.2024 - 18:18:31] Status: WW_BW2_Vorrang = 0
[24.11.2024 - 18:18:31] Status: WW_BW2_Warm = 1
[24.11.2024 - 18:18:31] Status: WW_BW2_Tag = 1
[24.11.2024 - 18:18:31] Status: WW_BW2_Einschaltoptimierung = 0
[24.11.2024 - 18:18:31] Status: WW_BW2_Ausschaltoptimierung = 0
[24.11.2024 - 18:18:31] Status: WW_BW2_Nachladen = 0
[24.11.2024 - 18:18:31] Status: WW_BW2_Manuell = 0
[24.11.2024 - 18:18:31] Status: WW_BW2_Laden = 0
[24.11.2024 - 18:18:31] Status: WW_BW1_Fehler_Anode = 0
[24.11.2024 - 18:18:31] Status: WW_BW1_Fehler_WW_bleibt_kalt = 0
[24.11.2024 - 18:18:31] Status: WW_BW1_Fehler_Fuehler = 0
[24.11.2024 - 18:18:31] Status: WW_BW1_Fehler_Desinfektion = 0
[24.11.2024 - 18:18:31] Status: WW_BW1_Ferien = 0
[... 106 lines cut ...]
[24.11.2024 - 18:18:10] Config: HK2_Fernbedienung = An
[24.11.2024 - 18:18:10] Config: HK2_Temperatur_Offset = 0.0 °C
[24.11.2024 - 18:18:09] Config: HK2_Heizsystem = Fussboden
[24.11.2024 - 18:18:09] Config: HK2_Absenkungsart = Raumhalt
[24.11.2024 - 18:18:09] Config: HK2_Aussenhalt_ab = 0 °C
[24.11.2024 - 18:18:09] Config: HK2_Aufschalttemperatur = 2 °C
[24.11.2024 - 18:18:09] Config: WW_Vorrang = An
[24.11.2024 - 18:18:09] Config: HK2_Auslegung = 43 °C
[24.11.2024 - 18:18:09] Config: HK2_Max_Temperatur = 49 °C
[24.11.2024 - 18:18:08] Config: HK2_Urlaubtemperatur = 17.0 °C
[24.11.2024 - 18:18:08] Config: HK2_Betriebsart = Automatik
[24.11.2024 - 18:18:08] Config: HK2_Tagtemperatur = 21.5 °C
[24.11.2024 - 18:18:08] Config: HK2_Nachttemperatur = 20.5 °C
[24.11.2024 - 18:18:08] Config: HK2_Sommer_ab = 16 °C
[24.11.2024 - 18:18:08] Config: HK1_Frost_ab = 0 °C
[24.11.2024 - 18:18:08] Config: HK1_Fernbedienung = Aus
[24.11.2024 - 18:18:08] Config: HK1_Temperatur_Offset = 0.0 °C
[24.11.2024 - 18:18:07] Config: HK1_Heizsystem = Heizkoerper
[24.11.2024 - 18:18:07] Config: HK1_Absenkungsart = Aussenhalt
[24.11.2024 - 18:18:07] Config: HK1_Aussenhalt_ab = 9 °C
[24.11.2024 - 18:18:07] Config: HK1_Aufschalttemperatur = 3 °C
[24.11.2024 - 18:18:07] Config: HK1_Auslegung = 50 °C
[24.11.2024 - 18:18:07] Config: HK1_Max_Temperatur = 90 °C
[24.11.2024 - 18:18:06] Config: HK1_Urlaubtemperatur = 17.0 °C
[24.11.2024 - 18:18:06] Config: HK1_Betriebsart = Automatik

dewenni, 2024-11-24:
Ok prima - wenn es jetzt passt dann ist ja gut 👍 
Ich schließe das dann hier mal ab.
Wenn Du noch herausfindest was "9147" ist kannst du dich ja nochmal melden.

closed 2024-11-24 by dewenni