dewenni/ESP_Buderus_KM271 #65: Außentemperatur Anzeigefehler

floschad, 2024-01-19:
Hi,

unter 0°C wird die Außentemperatur falsch angezeigt, siehe Fotos.
[image]
[image]

dewenni, 2024-01-19:
Danke für den Hinweis - das ist bereits bekannt und gefixed. #51
Ich habe es nur noch nicht geschafft ein neues Release zu erstellen da ich immer noch versuche zu verstehen warum so viele user Probleme mit der letzten Version haben (V3.2.4)
Bei dir scheint die aber ja sonst zu funktionieren, oder?
Welche Hardware hast du verbaut? das Board von Daniel oder ein normalen ESP mit der original KM271?

floschad, 2024-01-24:
Ich habe das Modul von Daniel verbaut, funktioniert alles ohne Probleme.
Ich hatte vorher ESPeasy als ser2net Server mit Fhem (KM271-Modul) in Betrieb, ebenfalls ohne Probleme.

closed 2024-01-24 by floschad