dewenni/ESP_Buderus_KM271 #66: Sollwerte Heizkreise

floschad, 2024-01-19:
Hi,

es wäre toll, wenn man auch den Sollwert der Heizkreise für Tag und Nacht einstellen könnte. Hier sind die Werte alle von Rudolf König zu finden: [[](https://github.com/mhop/fhem-mirror/blob/master/fhem/contrib/km271.pl)](url)
Danach Raumsollwert.

dewenni, 2024-01-19:
Hallo,

die FHEM Implementierung war zum Großteil auch meine Vorlage zu Umsetzung.
Das was du verlinkt hast sind aber alles nur Istwerte.
Auch die Raumsolltemperatur ist ein Istwert der gelesen wird.

Das habe ich auch schon implementiert
[image]

Der Wert für den "Raumsollwert" kommt aber von einem Raumthermostat was ich nicht verbaut habe.
Den Wert dafür vorzugeben wurde hier schon mal diskutiert #44

floschad, 2024-01-24:
Top, vielen Dank!

closed 2024-01-24 by floschad