dewenni/ESP_Buderus_KM271 #69: Webinterface bzw. WLAN Conntection

joeflower, 2024-02-04:
Hallo !
Ich habe gestern die V3.3.1 ausprobiert. Ist lange her, dass ich das Board via Interface geflasht habe. Flashvorgang hat funktioniert, WLAN Daten via Handy eingegeben. Speichern -> nix. Also kein Webinterface. Nochmal.. das Gleiche. Erase all hilft auch nicht. Nach etlichen Versuchen habe ich das Board eingebaut und siehe da. WLAN steht und Webinterface ist erreichbar. Kann man das evtl auch im ausgebauten Zustand anzeigen, damit man den erfolgreichen Flashvorgang überprüfen kann?

Noch etwas: Man kann bei der GPIO Zuweisung die Board Rev einstellen. Nach dem Speichern sieht man jedoch nicht mehr was man eingestellt hat. Kann das jeweils Eingestellte im Drop-Down erhalten bleiben?
Das sind beides keine wirklichen Fehler. Nur Vorschläge.

P.S: Heute Morgen war der ESP nicht mehr erreichbar. Da half nur Neustart. Werde es weiter beobachten. Mit der 3.2.4 hatte ich seit dem Release keine Probleme.

dewenni, 2024-02-04:
Hi,

danke für die Rückmeldung.
Das es im abgebauten Zustand nicht startet ist etwas komisch und zumindest kein bewusstes Feature oder Verhalten.
Ein anderer User hat berichtet, es könnte an der Seriellen Schnittstelle liegen, die dann keine Verbindung hat.
Ich konnte das selbst aber noch nicht bestätigen.

Das mit dem DropDown für das Board sollte nur eine Möglichkeit sein, um eine Vorauswahl an GPIO´s zu laden.
Danach kann man es ja dennoch ändern so wie essen für einen passt. Beispielsweise auch den Ölzähler.
Dann müsste ich das Verhalten ganz ändern und bei Auswahl von einem Board Typ keine Änderung erlauben. Änderungen dann nur wenn man es auf "generic" stellt. Ich würde es aber so lassen wollen.

Bitte gib aber Feedback ob es sonst stabil läuft.

joeflower, 2024-02-09:
War wohl nur eine einmalige Sache mit dem Absturz. Läuft jetzt 1 Woche ohne Probleme und stabil.

closed 2024-02-11 by dewenni