dewenni/ESP_Buderus_KM271 #71: ArduinoJson warning

Franck78, 2024-02-07:
Not blocking the build of the final binary, but not working ;)

As you know , I'm (still) with Arduino IDE 1.8.
The library manager says "ArduinoJSON 7.0.3" is available.

When I select it, repeated warning is issued:

```
/Arduino/km271_3.31/config.cpp:152:28: warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]

```
Result is webserver unable to get any data and displaying only "Connected".

Need to use arduino_json 6.12.5

Franck

Franck78, 2024-02-07:
this is strange : "WebUI still uses the functions that are no longer available from V7"

When I replace as suggested, I don't have any undefined symbols! Using WebUI 2.2.3

dewenni, 2024-02-11:
my web interface is based on the library ESPUI and this still use `DynamicJsonDocument` what is no longer available in V7
https://github.com/search?q=repo%3As00500%2FESPUI%20dynamicjson&type=code

dewenni, 2024-03-28:
@Franck78 
Since I no longer have a dependency in the current release candidate #80, I was able to switch to ArduinoJSON V7.
I am happy for any feedback of the new version

dewenni, 2024-04-07:
fixed since [release 4.0.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.0)

closed 2024-04-07 by dewenni

Franck78, 2024-04-14:
Hello,
trying to compile the thing. 

stuck on

```
#include <gzip_c_css.h>
#include <gzip_js.h>
#include <gzip_login_html.h>
#include <gzip_m_css.h>
#include <gzip_main_html.h>

```
even google has nothing to say about those files ! Which lib is it ??

dewenni, 2024-04-14:
Hi @Franck78 
These are header files that are generated in PlatformIO with a python script. I use a script to generates these files based on HTML, CSS and JS source files to a compressed gzip format.
If you are using Arduino IDE this could be a problem...

I have set these files to the .gitignore because the compressed data change on every build.
But I can change this and add the files to the repository.

As a quick help, files attached
[Archive.zip](https://github.com/dewenni/ESP_Buderus_KM271/files/14971497/Archive.zip)

Franck78, 2024-04-15:
built and sent to my test ESPS3, ok.

Just add some comment in the header file on how to build the gzipped files. 

No time to really read changes in the code. I hope you separeted mttq/interface translated strings;)
I see no mini light show at startup / doubleReset not easily deactivated (with #define #undef help)
I also convert burner time in something readable (only for the display, not mqtt messages).
[image]

next thing to do (for ESP-S3 plenty of room), have some graphics of interesting data, served directly, some day.

dewenni, 2024-04-15:
_I hope you separeted mttq/interface translated strings;)_ **=> yes**
_I see no mini light show at startup / doubleReset not easily deactivated_ **=> no, sorry**
_I also convert burner time in something readable (only for the display, not mqtt messages)._ **=> not a bad idea**