dewenni/ESP_Buderus_KM271 #73: MQTT Topics and translation

Franck78, 2024-02-08:
Hi,

I'm new to mqtt so maybe I'm wrong but I don't think so ;)

It's a design error to make 'Topics' labels dependent on the LANG.

They should remain as stable as possible. And short.
This is low level communication. Far far from the GUI level / Human interface !

Each translation 'fix' will blow any setup , up to inside a database where a Topic label can be used as a column name.
If someone add a new LANG, it's setup is kaput.

Each 's_mqtt_cmds' 's_cfg_topics' 's_stat_topics'  's_error_topics' etc should be limited to English (most common low denominator).

dewenni, 2024-02-08:
Hi @Franck78 
as you already wrote, it is a question of design.
I got your point, but I will not change it.
I started this software for my own and used only german topics because it was easier to understand for me.
After some time I decided to go publish on GitHub and as a realized that there are also users outside germany, I have added english as well. At this time there were already some users including me, that are using databases and it would break my database if I would switch all topics to englisch.

If I would start from scratch I would fix the mqtt messages to english. But now it is what it is.
The language setting is also not to be understood as a language for the WebUI, but is a setting to change the language of the entire software. This is something that you configure only once. Users in Germany can use German and all others can use English.

Sorry, but I'm not a professional software engineer and the software has grown over time. I would solve some things differently today, but now there is also the issue of compatibility.

closed 2024-02-08 by dewenni

Franck78, 2024-02-09:
"but I'm not a professional software engineer"
Profesionnal in the meaning 'earning money' maybe. Skilled enough to write good code, yes.

Fixing this is not really complicated.

First, make LANG=0 the 'english'. I have done it in this attachment language file.

The next step is having LANG and MQTTLANG. This need a small update in the settings, something saying
'German MQTT messaging'  and a limited search and replace.
When unchecked, MQTTLANG will be zero .

The user interface will use the LANG index, and mqtt related stuff MQTTLANG (wich will be only 0 or 1).

Optionally depending on space available on your ESP, the arrays could be

["english mqt label", "german translation", "English trans", "French trans" ]
the first field being more a short mqtt name.

This little awk prog do 90% of the job for swapping the english/german in language.h
It can also be use by anyone to add a new lang by just addnig a m[1] field for example.
The remaining 10% is esay by hand.

```
gawk '
{
   if (match($0,/^(const.+= *\{)(.+), *(.+)\};/,m))
   {
     # reformat as correct syntax
     printf("%s%-50s,%s};\n", m[1], m[3], m[2] )
   } else {
     #do nothing to the line
     print $0
   }
}
'
```
[language.zip](https://github.com/dewenni/ESP_Buderus_KM271/files/14222920/language.zip)