dewenni/ESP_Buderus_KM271 #81: ESP32 mit LAN Anschluß ?

Kenaschon, 2024-04-03:
Hallo zusammen,

gibt es eine Möglichkeit, das alles auch per Lan-Kabel zu realisieren?

Ich habe die 3.3.1 drauf. Leider verliert sie immer noch regelmäßig die Wlan Verbindung. Trotz Wlan-Repeater im Keller.

Gibt keinen ESP32 mit Lan Anschluß auf dem man die Software aufspielen kann?

Grüße
Manfred

dewenni, 2024-04-03:
Hallo Manfred,

es gibt spezielle ESP32 Board mit LAN Anschluss, aber es gibt auch Erweiterungsmodule die man per SPI-Bus an einen ESP32 anbinden kann.
Ich habe mir mal spontan so ein Erweiterungsmodul bestellt und werde die Implementierung mal testen.
Das könnte generell ganz interessant sein. Mein Empfang ist im Keller auch nicht der Beste und LAN wäre kein Problem.
Ich teste das mal und melde mich.

Grüße Sven

Kenaschon, 2024-04-03:
Vielen Dank für die schnelle Reaktion.

the78mole, 2024-04-03:
Hi all,

just want to join the discussion as the HW developer of the KM271-WiFi :-P 
The KM271-WiFi-Board gives access to the SPI signals on the extension header J7, as depicted in the following images/screenshots.
[image]
[image]
[image]

If they do not collide with other stuff (r.g. internal SPI-Flash), you could easily attach one of the various Ethernet-SPI controllers to it (e.g. the [LAN8720](https://www.roboter-bausatz.de/p/lan8720-netzwerkmodul-lan-tcp-ip-ethernet)) and use the [arduino ESP-example](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/examples/ETH_LAN8720) with it.

If this works, I would tend to design some "official" extension for the KM271-WiFi.

Regards,
Daniel

Kenaschon, 2024-04-04:
Englisch - Deutsch - Englisch - Deutsch. Kann ich verstehen.

Ich verstehe auch, das github international sein soll, aber in diesem Fall stelle ich mir immer die Frage, wie viele haben im Ausland eine Buderus Öl-Heizung?

the78mole, 2024-04-04:
Hi Kenaschon,

tatsächlich verschicke ich ca. 15-20% an Nutzer im Ausland, überwiegend aus der EU (Rumänien, Ungarn, Polen, Litauen,,...), aber auch einige z.B. in die USA und die Ukraine (vor 2022). Ich hab mir diese Frage vorher auch gestellt und war sehr überrascht, wo Buderus überall seine Kessel stehen hat. Besonders außerhalb der EU ist das ziemlich nervig, da man dann den ganzen Zoll-Kram ausfüllen muss.

Grüße,
Daniel

dewenni, 2024-04-04:
Hi,

ich habe gestern spontan mal einen [W5500](https://www.roboter-bausatz.de/p/w5500-mini-netzwerkmodul-lan-tcp-ip-ethernet) bestellt. Der macht einen guten Eindruck und kann per SPI angebunden werden.
In der Bauform ist er auch sehr kompakt und könnte ggf. dann zukünftig auf das Board mit aufgesteckt werden?

Ich teste das mal in der Software und Berichte.

Grüße Sven

PS: bezüglich englisch. Es gibt scheinbar mehr internationale Interessenten als man denkt. Daniel hat auch schon einige Boards ins Ausland verkauft. Hätte ich anfangs auch nicht gedacht.

the78mole, 2024-04-04:
Hi @dewenni,

tatsächlcih scheint der LAN8720 selbst kein SPI zu haben, der W550 allerdings schon.

Bin gespannt, was Du berichtest.

Grüße,
Daniel

Kenaschon, 2024-04-04:
Ich dachte das so etwas geht.

https://www.roboter-bausatz.de/p/wt32-eth01-esp32-modul-mit-ethernet-bluetooth-wifi

JieffeEl, 2024-04-04:
@Kenaschon Buderus is also sold in Belgium where, despite German being one of the three national languages, almost nobody speaks it :) I never learned it at school, whereas French (my maternal language) and Dutch are mandatory ...  Also, Buderus is (was?) sold under the label Bosch abroad.

@the78mole @dewenni  I once bought a LAN8720 module but it seems to need an awful lot of pins to function, as well as a few hardware modifications. The W5500 is certainly a wiser solution in this regard. In the past, I successfully used ENC28J60 modules with Arduinos and ESP8266s but that chip relies a lot on the microprocessor for the network protocols, contrary to the W5*00 that does most of the work internally.

the78mole, 2024-04-04:
Hi @JieffeEl, @Kenaschon 

@JieffeEl  the LAN8720 has a default RMI, which has some, but not awfully lot signals. The ESP32 could also handle this, but it needs to be connected to certain pins, which are not routed to a pin header on my board. SPI-to-Ethernet is simply the only solution, my board can be extended with.

@Kenaschon if you want to design your own Buderus KM271, you are free to do so 😜, but the board you suggested does not fit on my board and simply makes no sense to be added, too, since mine already has an ESP32 populated. 😅

Regards,
Daniel

dewenni, 2024-04-04:
As is so often the case, it's not as easy as you think. The W5500 does not work out of the box with the "standard" ESPAsync web server that I am using. There are already several issues here.
However, there are examples where this was generally done with a web server. But apparently always with various adjustments.

this example [WebServer_ESP32_W5500](https://github.com/khoih-prog/WebServer_ESP32_W5500) works, but I have to check how I can adapt this to my project.

the78mole, 2024-04-04:
@dewenni Autsch! Klingt nach unschöner Arbeit und nach einem Haufen KB Speicher... Soweit ich weiß ist da nicht mehr viel Platz übrig, oder?

dewenni, 2024-04-04:
ja, sieht leider nicht so aus, als könne man das einfach parallel zum Standard WiFi und Standard WebServer betreiben.

Wenn ich es aber [hier](https://github.com/espressif/arduino-esp32/issues/8605#issuecomment-2034383284) richtig deute, dann könnte sich das ab Arduino 3.0 bessern.

Wann das kommt, ist mir aber nicht bekannt. Wenn man [hier](https://github.com/espressif/arduino-esp32/milestone/4) schaut, könnte es aber bald verfügbar sein.

the78mole, 2024-04-04:
Hm, klingt vielversprechend... Somit bleibt mir ja noch etwas Zeit für die Umsetzung der HW 🤣

dewenni, 2024-04-05:
So, ich habe es mal grundsätzlich mit Arduino 3.0 (alpha) in meinem Projekt am laufen.
Ich musste aber auch ein paar Dinge deaktivieren und auch in der AsyncTCP provisorisch IPV6 deaktivieren.
Auch parallel zum WiFi funktioniert es noch nicht. Zumindest nicht mit den bisherigen WiFi Funktionen. Die muss man ggf. auch durch neue Funktionen ersetzen. die bisherige OneWire Funktion ist auch nicht mehr kompatibel.
Also irgendwie wird das schon funktionieren, aber ganz so einfach ist es nicht.
Ich warte mal auf das release von Arduino 3 und dann werde ich mich damit nochmal weiter beschäftigen.

dewenni, 2024-04-05:
@the78mole 
konkret habe ich dieses [W5500 Modul](https://www.az-delivery.de/products/w5500-ethernet-netzwerk-internet-modul-fuer-arduino) gekauft. Das müsste man dann mit einem Kabel anschließen und ggf. nochmal in ein extra Gehäuse packen.

[image]

Als Header für dein Board wäre aber [diese Bauform](https://www.roboter-bausatz.de/p/w5500-mini-netzwerkmodul-lan-tcp-ip-ethernet?srsltid=AfmBOopXO3m2K_GUOOebsRd9pD0J_G6smNyHzLr0dfZDWPB8LJPnvJZ5iMg) vielleicht interessanter. 

[image]

Das Teil könnte man ja ggf. direkt auf dein Board aufstecken wenn du die Stiftleisten entsprechend vorsiehst.

Man benötigt aber auch fast alle Pins (6+2) in der Software:

- (01) ETH_PHY_CS
- (02) ETH_PHY_IRQ
- (03) ETH_PHY_RST
- (04) ETH_SPI_SCK
- (05) ETH_SPI_MISO
- (06) ETH_SPI_MOSI
- (07) VCC
- (08) GND

Soweit ich das bisher verstanden habe, könnte man das WiFi sogar parallel zum Ethernet betreiben.
Aber das muss man dann mal ausprobieren. Aktuell habe ich mein Projekt mit Arduino 3 und dem Ethernet provisorisch am laufen. Dafür bekomme ich in Arduino 3 das WiFi generell nicht zum laufen. Auch nicht in einem neuen leeren Testprojekt.
Da passt scheinbar irgendwas mit meinem Platformio Setup nicht. Aber ich bleibe da mal dran und berichte wenn ich beides am laufen habe.

the78mole, 2024-04-05:
Hi @dewenni 

in ESPhome sind folgende Pins angegeben:

    # Example configuration entry for SPI chips
    ethernet:
      type: W5500
      clk_pin: GPIO19
      mosi_pin: GPIO21
      miso_pin: GPIO23
      cs_pin: GPIO18
      interrupt_pin: GPIO36
      reset_pin: GPIO22

Sieht fast so aus, als könnte man das halbwegs frei konfigurieren:
SPI based chips do not use [SPI Bus](https://esphome.io/components/spi). This means that SPI pins can’t be shared with other devices.

Allerdings steht auch auf der ESPhome-Seite: This component and the Wi-Fi component may not be used simultaneously, even if both are physically available.

Ich werde einfach mal ein sauberes Board entwerfen und das dann testen. Jetzt müssen erst mal die 100 neuen Module endlich bei mir ankommen. Die hängen schon die ganze Woche im deutschen Zoll fest und der meldet sich einfach nicht :-(

Grüße,
Daniel

dewenni, 2024-04-05:
kurzes update: In einem einfachen Testprojekt in der Arduino IDE, läuft auch WiFi mit dem W5500 parallel.
Ich habe einen kleinen Webserver in dem Projekt hinterlegt und den kann ich sowohl über die WiFi IP und auch über die LAN IP erreichen. Das wäre also sehr komfortabel.

the78mole, 2024-04-13:
Sag mal @dewenni ,

könntest Du spaßeshalber mal die Stromaufnahme und die Stromspitzen des gesamten Boards messen?
- nur Ethernet
- nur WLAN (dürfte bei WLAN-Aktivität ca. 170 mA sein)
- WLAN + Ethernet

Am einfachsten geht das, wenn man ein USB-Power-Messgerät hat und das Board über den USB versorgt.

Aktuell tippe ich schwer darauf, dass es mit Ethernet deutlich stromsparender sein sollte und geschätzt dann weniger als 100 mA in der Spitze zieht.

Das wäre dann eine gute Alternative für die wenigen Leute (3-5%), wo die Spannungsversorgung der Buderus-Steuerung ein wenig schwachbrüstig geraten ist und das Board manchmal einen Power-On-Reset ausführt.

Grüße,
Daniel

akno2205, 2024-05-03:
Hallo Daniel, ich mir diese Woche Dein KM271 Board zugelegt und bin sehr begeistert. Leider gehöre auch ich zu den Menschen mit schlechtem Wlan im Keller wo sich die Heizung befindet. Da ich sehr gute Erfahrungen mit ESP32-WROOM-32U mit externer Antenne gemacht habe. Habe ich mich gefragt, ob man dies nicht als Option nutzen könnte. Somit wäre es möglich einen besseren Empfang zu bekommen. Diese wäre nicht so umständlich, wie einen Lan-Port zu verbauen. Man könnte nur den ESP tausche, sofern dieser den Anforderungen entspricht.
Gruß Andy

the78mole, 2024-05-04:
Hi Andy,

habe Dir ja schon per Email geantwortet. Für andere, die das hier lesen, noch mal kurz die Passage zum ESP32 mit externer Antenne:

Grundsätzlich ist es natürlich möglich und ich habe mir das auch bereits überlegt. Allerdings hätte ich dann zwei Varianten des Moduls, was die Bevorratung dann ganz schnell teurer für mich macht. Auch der Skaleneffekt ist dann natürlich nicht mehr so groß, wenn ich nicht mehr die gleiche Menge einer Variante bestelle. Aktuell habe ich eine Variante und löte die Optionen dann nach Nutzerwunsch auf. Da das nur "grobe" THT-Stecker sind, geht das recht flott. Wenn ich das mit dem ESP-Modul ebenfalls so machen wollte, dann müsste ich, je nach Nutzerwunsch, ein anderes Modul per Hand auflöten. Das ist mir viel zu aufwendig. 

Nun gäbe es noch zwei Möglichkeiten: Entweder ein ESP-Modul umlöten oder ein blankes PCB komplett neu bestücken. Ersteres ist schwierig, selbst mit Heißluft-Lötkolben. Für Letzteres habe ich gar nicht alle Bauteile da (z.B. die Levelshifter oder die größeren ELKOs).

Grüße,
Daniel

dewenni, 2024-11-23:
@the78mole @Kenaschon @akno2205 @JieffeEl 

Ich habe nun im aktuellen Release v5.0.0 die Unterstützung für einen Ethernet Abschluss implementiert.
Ausgelegt ist die Implementierung für ein W5500 Board. Das kann man einzeln kaufen und sowohl an einem normalen ESP32 als auch an dem Board von Daniel anschließen.
Doku findet ihr in der Readme.md
Das ganze läuft bei mir an einem Board v0.0.6 zuverlässig.
Wichtig ist nur, das das Kabel zwischen Board und W5500 so kurz wie möglich ist (ca 10cm). Ein längeres Kabel hat bei mir Probleme gemacht.

Kenaschon, 2024-11-23:
Hallo Dewenni,

vielen Dank für die tolle Arbeit. Habe mir direkt ein W5500 bestellt. Melde mich wenn es da ist.

Grüße
Manfred

dewenni, 2024-11-25:
@Kenaschon und alle anderen Interessenten
@the78mole hat hierzu auch schon was geplant - Ihr könnt euch zumindest schon mal auf die Warteliste setzten:
https://www.tindie.com/products/the78mole/km271-wifi-ethernet-extension/

[image]

Kenaschon, 2024-11-25:
Nette Idee.

Ist auch ein "nur" Board km271 mit Netzwerkanschluss geplant?

the78mole, 2024-11-25:
Hi Kenaschon,

> Ist auch ein "nur" Board km271 mit Netzwerkanschluss geplant?

Meinst Du, direkt auf der Einsteckkarte ein Ethernet? Nein, eher nicht. Die Resonanz auf die Ethernet-Extension ist weit hinter den Erwartungen, da rentiert sich ein KM271-Ethernet nicht bzw. ist den Aufwand nciht wert. Ich werde ohnehin nicht mehr ewig Support machen können. Die olle Buderus-Heizung fliegt in den nächsten 1-2 Jahren raus und dann wird es schwierig mit dem Real-World-Testen. Da kann ich mir die Buderus nur noch neben den Schreibtisch stellen und schauen, ob die Kommunikation grundlegend funktioniert...

Grüße,
Daniel

Kenaschon, 2024-11-26:
Bevor ich es direkt alles verbinde, ist das so ok?

[image]

dewenni, 2024-11-26:
Ich hatte zuletzt folgende Belegung genutzt:
ESP32-Mini
|  | GPIO |
|------|----|
| CLK   | 18 |
| MOSI | 23 |
| MISO | 19 |
| CS     | 5   |
| INT    | 16 |
| RST   | 17 |

closed 2024-11-27 by dewenni

Kenaschon, 2024-11-28:
"Wichtig ist nur, das das Kabel zwischen Board und W5500 so kurz wie möglich ist (ca 10cm). Ein längeres Kabel hat bei mir Probleme gemacht."

Ich habe die Kabel durch einen Ferrit-Kern gelegt. Danach war alles ohne Störungen und Probleme.

Grüße
Manfred