dewenni/ESP_Buderus_KM271 #83: v4.0.0 Ölzähler und Pushover

kalleick, 2024-04-05:
Hallo Sven, 

erst mal einen großen Dank und Lob für die neue Version.

Eben installiert und mir sind noch zwei Dinge aufgefallen: 

a) der Ölzähler (ich hatte den virtuellen genutzt) ist plötzlich bei 0.0 und der alte Wert fehlt. Dauert es noch, bis der Wert kommt? 
Werte für Verbrauch Kg/h (1,5) und Öl-Dichte KG/L (0,8) sind hinterlegt, hat er übernommen: 
  "oilmeter": {
    "use_hardware_meter": false,
    "use_virtual_meter": true,
    "consumption_kg_h": 1.549999952,
    "oil_density_kg_l": 0.839999974
  },
b) Pushover ist großartig. Hab es auf Alarme und Info eingestellt und nach Neustart hat er mir diesen per Pushnachricht gemeldet. Super. Allerdings bekomme ich bei einer Testnachricht nichts gemeldet. 

Vielen Dank
Patrick

dewenni, 2024-04-05:
@kalleick 
Vielen Dank für das Feedback!

**virtueller Ölzähler**
Hier hatte ich in der Tat noch einen Fehler. Es wurde beim virtuellen Ölzähler auch der Wert vom HW-Zähler ausgegeben.
Ich habe es korrigiert.

**Pushover Testnachricht**
Das sollte schon funktionieren. Das Problem ist aber, dass der Eingabetext nicht übernommen wird, wenn man direkt von der Eingabe auf den Button drückt. Wenn man nach der Eingabe woanders hin klickt oder Enter drückt und dann erst den Senden Button, sollte es klappen.
Ich habe aber auch das noch optimiert und jetzt wird mit dem Drücken vom Sende Button der Text vom Eingabefeld aktiv gelesen. Sollte also auch gelöst sein.

Ich werde noch etwas auf weiteres Feedback warten um ggf. noch weitere Bugfixes zu machen und dann werde ich ein neues Release machen. Die zwei Themen hier sollten ja für dich auch nicht so dringend sein.

dewenni, 2024-04-07:
fixed with [release 4.0.1](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.1)

closed 2024-04-07 by dewenni