dewenni/ESP_Buderus_KM271 #84: Unterbrechung nach einer gewissen Zeit

Kenaschon, 2024-04-10:
Hallo,

ist es normal, das nach einer gewissen Zeit (ca. 1 Stunde) immer diese Anzeige kommt?

[image]

Kenaschon, 2024-04-10:
Das passiert auch, wenn der Browser offen ist und sonst nichts am Rechner gemacht wird. Ruhezustand habe ich nicht am Rechner.

dewenni, 2024-04-13:
ich habe im aktuellen [Release v4.0.2](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.2) ein re-connect eingebaut. Kannst du mal testen

Kenaschon, 2024-04-13:
Habe das Update eingespielt - werde berichten. Vielen Dank für die schnelle Umsetzung.

Kenaschon, 2024-04-16:
Es scheint zu funktionieren. Danke dafür.

closed 2024-04-17 by dewenni