dewenni/ESP_Buderus_KM271 #87: feature-request: set time with NTP after power loss

mutex1980, 2024-04-17:
Would it be possible to have an option to automatically set the time and weekday from NTP after the Logamatic had a power loss? Potentially some automatic NTP sync if the difference between the Logamatic time and the NTP time shows significant difference?

As far as I understood, this can be only done manually in the web UI currently?

Thanks, Mirko

dewenni, 2024-04-17:
Unfortunately it is not possible to read the actual time from the Logamatic. At least I don´t know how to read it.
The only way is to set/write date and time from the ESP to the Logamatic.
I have already implemented, that the Time will be set when the "daylight saving time" has changed.

It is also possible to set the date and time on the logamatic with an mqtt command.
In that case the actual ntp time is used and will be set to the logamatic.

```
mqtt  topic:  ../setvalue/setdatetime
payload: none
```

So you could do this by your own, but you should wait some seconds or better some minutes after restart before you trigger the command, because it took some time to get the ntp time from the ntp Server after restart.

dewenni, 2024-04-17:
but I can also think about an additional option to set the date and time will be set automatically after restart.
This could be an option in the "date and time" settings.

mutex1980, 2024-04-18:
> but I can also think about an additional option, that date and time will be set automatically after restart. This could be an option in the "date and time" settings.

Yes, this would be a solution as the ESP itself is supplied by Logamatic power. Thanks for your work Sven.

Actually this my primary use case for making the Logamatic smart at all. I often realize only after days it again lost time synchronization after power loss and the water is not hot enough during day time.

dewenni, 2024-04-23:
will be implemented within next release

dewenni, 2024-04-27:
implemented in [v4.0.4](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.4)

closed 2024-05-31 by dewenni

mutex1980, 2024-06-30:
Sorry for the late reply - I tested today with 4.1.0 and it works. Thanks for implementing.