dewenni/ESP_Buderus_KM271 #91: Kleinigkeiten

joeflower, 2024-04-20:
1. Ich weiß nicht warum, aber ich muss OTA update immer 2x machen. Schlägt beim ersten mal immer fehl. Dann sofort nochmal und es funktioniert. Beobachtet habe ich das seit 4.0.0
2. Den Hardware Ölzähler und den Virtuellen kann man gleichzeitig auswählen. Macht in meiner Welt keinen Sinn. Oder versteh ich da was falsch ?

Ansonsten TOP ARBEIT ! Danke für die tollen Funktionen.... läuft stabil durch (zumindest von Update zu Update)

dewenni, 2024-04-20:
Danke für dein Feedback.

Das mit dem OTA habe ich auch manchmal. Ich konnte aber leider noch nicht herausfinden woran das liegt. Ich glaube es hat auch einfach was mit der WiFi Verbindung zu tun.
Manchmal klappt es beim ersten Versuch und dann braucht es plötzlich wieder mehrere Versuche. Damit muss man jetzt leider erstmal leben. Irgendwann klappt es ja dann schon 😉

Die Ölzähler kann man jetzt beide unabhängig aktivieren. Ob das nun sinnvoll ist...
Aber es schadet ja nicht. Das zu verriegeln wäre aus meiner Sicht ein unnötiger Konfigurationsaufwand. Daher würde ich das so lassen.

joeflower, 2024-04-20:
ok damit gebe ich mich zufrieden.... Dann kann man ja hier zu machen.

closed 2024-04-20 by joeflower