dewenni/ESP_Buderus_KM271 #94: Unable to connect to wifi

Kev0123, 2024-04-23:
I am unable to connect to wifi, its a 40 digit password with serveral special characters. These are the ones contained: 
{($<~%\}_!
at least one of them is causing an issue with connection.

Kev0123, 2024-04-23:
The issue is not affecting e.g. the WLED project.

dewenni, 2024-04-23:
ok, I will test it.

dewenni, 2024-04-23:
Id did some tests and found the reason for that. It is not related to WiFi function itself.
It is a problem of handling the user input from my side.
I use snprintf to copy the user input from the webUI to the config structure.
And in printf function the `%` is a special formatting character. If you have a `%` in the string, it will be removed.

It's a bit unlucky how I currently use the function. Will be fixed with the next release.
Thanks for reporting.

dewenni, 2024-04-27:
fixed with [v4.0.4](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.4)

closed 2024-04-27 by dewenni