dewenni/ESP_Buderus_KM271 #96: Beschreibung des Setup-Mode

floschad, 2024-04-24:
In der Beschreibung steht Reset drücken und nach 5 Sekunden nochmal Reset drücken.
Der Setup Modus wird nach 5 Sekunden Rest halten aktiviert.

P.S. Ich war sehr froh, dass es diesen Modus nun gibt :-)
Vielen Dank für die tolle Arbeit an dem Projekt!

dewenni, 2024-04-25:
Hi,

Bin mir jetzt nicht ganz sicher was das Ziel von deinem Beitrag ist.

Den Setup Mode gibt es ja schon fast von Beginn an.

Ab der 4.0 habe ich nur die Zeit für die Erkennung des "Double Reset" von 10s auf 3s reduziert.
Bei 10s konnte das auch schon mal reagieren obwohl man es nicht wollte.

Wolltest du mich nur darauf hinweisen, dass die Beschreibung noch nicht angepasst ist und dort noch von 10s ausgegangen wird?
Das muss ich in der Tat noch anpassen.

floschad, 2024-04-26:
Hi,

sorry für die Verwirrung.

Ziel ist die Beschreibung. Mit dem beschriebenen vorgehen bin ich nicht weiter gekommen.
Das Gerät ist bei mir in den Setup-Mode gewechselt, als ich den Reset Taster 5 Sekunden lang gehalten habe.

dewenni, 2024-04-27:
fixed with [v4.0.4](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.0.4)

closed 2024-04-27 by dewenni