dewenni/ESP_Buderus_KM271 #98: Settings / UI are dependend on browser history / hostname not changable - Ver. 4.0.3

solutionfound, 2024-04-28:
Hello,
as I upgraded from 3.2.5 for some hours to 4.0.3 and 4.0.4 - now back to 3.2.5 - I want to report some UI behavior where I am not sure if they are planned or a bugs.

1. After the OTA update the card was fully working thus has stored all the configuration. In the web UI all settings have been empty. In the Tools menu to export the settings when selecting view all settings are shown with correct values. But when apply restart in the "Einstellung" section the board starts and looses all setting, i. e. wifi password is lost and the board is not accessible any longer.
2.  I did an export of the settings and the hostname of the board was saved in a wrong category - if I recall it was either SSID or wifi password - definitely in a wrong category - unfortunately I cannot it check any longer as I have modified the jason file manually immediately when I realized that. A second export after some time and some restarts do not show this issue.
3. I was not able to change the hostname - it is always ESP-Buderus-KM271 - importing a jason nor changing it in the UI did not work.
4. Strange behavior of the settings. When opening the first time from a browser the UI is in it's default mode - no settings visible, all features, i.e. warm water and heating circuits are off. When selecting the features and revisit the web front end they are visible and stored, but if you use another browser from another device the UI is again in the default mode. I tested this with three devices (desktop, laptop, mobile). Especially with the behavior reported unter 1. it has the consequence if you select restart the values you have not put into the settings are lost / erased. 
5. No confirmation on changing settings - for me it was due to the behavior described above not clear when the settings are applied to the board as there is no "save" button or confirmation in the dialogue. Some show immediate effect like selecting the features warm water or heating circuit, but whether this applies to the fields where text is given like hostname, password, ssid, NTP server, etc. is not clear and also if a restart is required.

I hope the report helps to identify some misbehavior and avoid others having the same trouble.
Thank you Daniel for you work so far.

Viele Grüße
-Marco

dewenni, 2024-05-11:
Hi @solutionfound 

sorry that the update was not working as expected.

regarding 1)
that is something that is difficult to further analyze. I will do some tests and see if I can reproduce something like this.

regarding 2 and 3)
You are right, there was still a mistake in my code and I save the hostname to the config.wifi.ssid.
Unfortunately a copy paste mistake.... I will fix it soon.

regarding 4)
this is also something that I haven't seen so far. I will do some tests and see if I can reproduce something like this.

regarding 5)
All settings are automatically stored when they changed. I thought this is more comfortable than before.
But they are only saved when you changed it or at least click inside the input field.
Most of the values are only used on startup (example: WiFi, GPIO, ...)

the78mole, 2024-05-11:
Little sidenode/question. It could be easier to reproduce some brwoser-cache behaviour, when using a fresh incognito mode of your browser. This should not take over any cache.

solutionfound, 2024-05-18:
Just updated to 4.1.0 where all settings are empty. When I enable HC 1 and 2 etc. they are shown in UI.
When I open a browser window in private mode (Firefox) then again all settings are empty and the Logamatic options like HC1 and HC 2 are not selected.
So it seems the settings are not stored persistent on the board but somewhere in the browser (chache?) ?

solutionfound, 2024-05-31:
When the browser looses connection - identified on the lower part of the window - I reload and then all settings like the switches for the Logamatic are off and all other settings are in the default mode.

One information: I have setup the board without access control i.e. without a user. Could this be related to the issue?

dewenni, 2024-06-01:
no I also use it without authentication, but I am still not able to reproduce this behavior.
I have testet with Safari, Chrome and Firefox and for me everything looks good.
strange....

dewenni, 2024-11-03:
@solutionfound I found some possible issue when a web client connects. This is fixed in the new v4.1.4
Maybe you can also check if this solves your problem with the update of the settings in webUI.

solutionfound, 2024-11-03:
@dewenni thank you for that hint. I am still running the version 4.1.7 you have made some moths back specifically with the additions for the solar values to be sent. 
I was reluctant to update to any other version you offered as a general release as I did not found any information in the release notes stating you have included the changes for the solar values made for me.
So what will happen with the solar values when I upgrade to your suggested 4.1.4 version?

dewenni, 2024-11-04:
ah you are right - I haven't implemented this up to now.
I will include this at the latest for the planned update (Arduino3 + Ethernet support).
I'll probably manage that this month

dewenni, 2024-11-25:
should be solved with V5.0.0

closed 2024-11-25 by dewenni