dewenni/ESP_Buderus_KM271 #99: Not booting on powerup / LED status

Kev0123, 2024-04-29:
After cutting the power to the Buderus the board won't come on without a push of the reset button. Is this intended?

Where is the status of the LEDs documented? The 2nd is orange, 3rd is blinking green and last one is constant on.

Kev0123, 2024-04-29:
As I see right now the webserver and Wifi keeps crashing as well...

dewenni, 2024-05-11:
Hi @Kev0123 
the LED status depends on your settings and your Hardware. You can configure the LED´s in the GPIO Settings
[image]

That the "software" crashes is something that is hard to analyze for me, because I haven't seen this so far with the 4.x Software. My board has now been working for 13 days.
[image]

the78mole, 2024-05-11:
Try supplying it through USB (don't forget to replace the one PWR_SEL Jumper). If it stops crashing, than you are one of the few Buderus owners with a weak/crappy internal power supply. See my [YouTube-Video](https://www.youtube.com/watch?v=h_pQlpXaQ1I&t=810s) if not sure about the Jumpers.

dewenni, 2024-05-26:
@Kev0123 any news?

Kev0123, 2024-05-26:
Sry for the late response.

After re-plugging the device into the logamatic port, the booting works fine now. I just wanted to watch if the crashed will occure after some time. 
The random reboot seems to be fixed with the latest version and/or because of reinserting. Don't know if its a co-incident or just because of the previous bad connection between the board and the logamatic.

For now everything is working fine.

I couldn't figure out how to aggregate the infos from of the gerber files and the pinout to see which LED does what, so if anyone can give a basic overview whats the "stock" config if GPIO is set to the latest version ( thing its <0.0.6 or so labeld) would be great. Otherwise the topic can be closed.

Thanks for the support!

the78mole, 2024-05-26:
Hi Kev,

sorry for not replying, was on vacations with family for some days...

You could do it the hard way using gerber files, but you could also use the interactive BOM for easy access to your infromation. If you have the 0.0.5 e.g., use this link:
https://htmlpreview.github.io/?https://github.com/the78mole/km271-wifi/blob/main/KM217-WiFi/bom/ibom%200.0.5.html

Here you can select the appropriate net and see it highlighted in total:
[image]

You can replace the URL with any from here to get the iBOM of the different versions, from that file list:
https://github.com/the78mole/km271-wifi/tree/main/KM217-WiFi/bom

Now it should be quite straight forward to track down the GPIOs together with the schematics, which could be found here in PDF format:
https://github.com/the78mole/km271-wifi/tree/main/KM217-WiFi/Export

In dewennis firmware, you can assign different functions to the LEDs' GPIOs.

Hope this help :-)

Regards,
Daniel

dewenni, 2024-05-26:
maybe I can add this information in the webui settings as well.

dewenni, 2024-05-31:
@Kev0123 @the78mole 
I have added some information dialog to the settings page - will be available with the next release
[image]

mutex1980, 2024-06-30:
I have Rev 0.0.8 and the Board does not (at least in 9 of 10 cases) boot even when USB powered. 
Inside the Logamatic it never booted without pressing reset.

Jumpers must be correct as booting works after pressing reset.

https://github.com/the78mole/km271-wifi/issues/6

dewenni, 2024-11-25:
I close this topic here, because I think it is nor related to the software.
Go ahead here: https://github.com/the78mole/km271-wifi/issues/6

closed 2024-11-25 by dewenni