dewenni/ESP_Buderus_KM271 #104: Bei v4.1.0 wieder sehr viele W-Lan abbrüche

Kenaschon, 2024-05-23:
Hallo, 

ich habe auf V4.1.0 upgedatet. Jetzt habe ich wieder sehr viele W-Lan Verbindungsunterbrechungen.

Auch ein drücken der Resettaste führt nicht immer zu Erfolg. Rote LED bleibt dann dauerhaft rot. Erst Stecker raus und wieder rein kann das Board wieder erwecken.

Zurück auf V4.0.0 habe ich das nicht.

Grüße
Manfred

dewenni, 2024-05-26:
Hallo Manfred,

ich denke das muss irgendein Seiteneffekt sein. Ich glaube nicht das es an der Software direkt liegt.
In der 4.1.0 habe ich lediglich etwas mehr MQTT Nachrichten für HomeAssistant ergänzt. Diese Nachrichten werden aber auch nur einmalig beim Start gesendet. Danach werden von HA die bereits existierenden Nachrichten ausgewertet.

Vielleicht gibt es allgemein ein Problem mit den Updates. Das die Updates machmal nicht sauber funktionieren.
Das wäre aber auch schwer zu analysieren....

Ich bin ja auch weiter an einer Version basierend auf Arduino 3.0 dran. Die hat dann auch Unterstützung für einen Ethernet-Anschluss.

solutionfound, 2024-05-26:
Ich hatte am Samstag ein Update auf eine neue Version durchgeführt, die dewenni freundlicherweise für mich erstellt hatte, Ab dann hatte ich das Problem, dass immer die Verbindung zum WLAN verloren ging. 
Gleiches ist mir schon mal bei einem früheren Versionen passiert.
Was ich aber dann (außer gestern) gemacht habe ist die Firmware 2x zu installieren.

Heute früh habe die oben genannte Version nochmals OTA geflasht und seitdem - 10,5 Stunden - läuft der esp ohne einen Abbruch.
Das würde die These von dewenni unterstützen
Daher mal der Tipp das upgrade 2x durchzuführen und zu sehen, ob es dann besser wird.

Kenaschon, 2024-05-26:
Werde ich mal testen.
Danke für die Info

solutionfound, 2024-06-30:
Mein System lief jetzt über 30 Tage ohne Fehler durch.

Gerade habe ich gesehen, dass es vor ca. 2 Tagen wohl automatisch neu gestartet ist: Grund kenne ich nicht, aber es ist erreichbar und liefert Werte.      
System zeigt Folgendes:      
Laufzeit | 1d 21h 42m 04s
Neustart Grund | Software panic reset

dewenni, 2024-11-25:
Ich schließe das mal hier - bitte v5.0.0 testen und bei Problemen erneut ein issue öffnen

closed 2024-11-25 by dewenni