dewenni/ESP_Buderus_KM271 #107: Offline every three days

dremeier, 2024-07-11:
I have the problem that the interface is not accessible every three days. The two green LEDs on the right light up. After a reset everything is OK again.
What is needed for further debugging?

blacksheepinc, 2024-08-03:
I have a similar issue, for me the MQTT might be the problem. 
I can still reach/ping the device however in Home assistant, it show unavailable, after restart everything is ok for a day or two.

gtsy7, 2024-11-12:
I have a similar issue, for me the MQTT might be the problem.
I can still reach/ping the device however in Home assistant, it show unavailable, after restart everything is ok for a day or two.

dewenni, 2024-11-12:
please be patient - I am working on a new version with a different implementation

dewenni, 2024-11-25:
@blacksheepinc  @dremeier 
please try the new release v5.0.0
If there are still stability problems - please create a new issue or use the discussion

closed 2024-11-25 by dewenni