dewenni/ESP_Buderus_KM271 #109: Die Weboberfäche kann nach begrenzen des Zuganges nicht mehr geöffnet werden

stefangrosse, 2024-08-04:
Hi.
Nachdem ich den Zugang in den Einstellungen begrenzt habe, kann ich nicht mehr auf die Weboberfläche zugreifen:
[image]

Im Mqqt Explorer sehe ich, das die Karte noch erreichbar ist:
[image]

Auch kommt der Ping in HomeAssistent im Log an.

dewenni, 2024-08-15:
Hi,

das Problem ist mir bekannt und ich habe das auch schon behoben. Ich versuche die nächsten Tage ein Update zu machen.
Das Update müsstest du dann zur Not nochmal per Kabel einspielen. Sorry!

Grüße
Sven

DanMike, 2024-08-30:
Ich habe dasselbe Problem. Gibt es ein Telnet- oder MQTT-Kommando, um das wieder zu deaktivieren?

dewenni, 2024-08-31:
Hallo,

ich habe das Problem  behoben und ein neues release dazu bereitgestellt.
Das Update müsstest du dann zur Not nochmal per Kabel einspielen. Sorry!

DanMike, 2024-08-31:
Hallo @dewenni ,
danke für das Update, es läuft jetzt! 👍 

Ich hatte schon vorher per Kabel die 4.1.0 nochmal geflasht, damit ich wieder auf das Web UI komme.
Das Update jetzt auf 4.1.1 habe ich per OTA eingespielt. Beim ersten Versuch gab es bei 99% den Fehler "Update nicht erfolgreich". Habe es dann einfach nochmal ausgeführt und das hat geklappt.
Kein Sorry nötig, was Du hier machst ist ganz große Klasse! 🥇

closed 2024-10-03 by dewenni