dewenni/ESP_Buderus_KM271 #111: logging in the web error

gtsy7, 2024-10-08:
The problem when logging in the web interface set a password now I can't log in 
[image]
help me solve the problem

gtsy7, 2024-10-10:
I activated the check mark and set the username and password and clicked reload. And after that I can't log in.
v4.1.1

gtsy7, 2024-10-10:
A
Status: 500. Message: InfluxDB returned error: error parsing query: found FROM, expected identifier, string, number, bool at line 1, char 9
[image]

dewenni, 2024-10-14:
> I activated the check mark and set the username and password and clicked reload. And after that I can't log in. v4.1.1

good question. I am not aware of this problem.
I think you have to flash it again if you can not login.

dewenni, 2024-10-14:
> A Status: 500. Message: InfluxDB returned error: error parsing query: found FROM, expected identifier, string, number, bool at line 1, char 9 ![grafan](https://private-user-images.githubusercontent.com/104444938/375427238-b08c67b9-5554-4a64-8a9e-3c4e1c150ee7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4ODYzOTAsIm5iZiI6MTcyODg4NjA5MCwicGF0aCI6Ii8xMDQ0NDQ5MzgvMzc1NDI3MjM4LWIwOGM2N2I5LTU1NTQtNGE2NC04YTllLTNjNGUxYzE1MGVlNy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxNFQwNjA4MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMzM1ZDRmZGIxYTNjMDY3OGIwYTllZTU4MzAxODFjZmU1MmMwYzVmZGM3NTQzNGZhZTMyNzcwMWFmYjc4NzRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hcnC2UdVBg-rJ8ltz2-6NU8WYCEPDpL_CRPXUKTxYBg)

Grafana is not a plug and play solution.
You have to be sure that your setup fits to the grafana export file
- influxdb 2.0 (Flux language)
- german mqtt names

closed 2024-11-09 by dewenni

Alacika, 2024-11-13:
I can't log in either after setting the login/password and rebooting.

dewenni, 2024-11-13:
@Alacika strange...
as a woraround you can connect via telnet and use command: "config reset"

Alacika, 2024-11-13:
> @Alacika strange... as a woraround you can connect via telnet and use command: "config reset"

Thanks for the suggestion, I'll try it this afternoon!

dewenni, 2024-11-13:
after reset the ESP will start in Setup-Mode and open the Access Point

Alacika, 2024-11-13:
I managed to reset it with telnet. I set everything up again. But when I activated authentication, after rebooting I couldn't log in again. Telnet reset again, I'm not setting up authentication for now. The version is 4.1.4, the language setting is English.

Alacika, 2024-11-13:
If I downgraded the firmware to version 3.x and set up authentication, then updated to 4.1.4, the login works fine!

dewenni, 2024-11-14:
interesting. Then I would imagine that in v4 the user name and password is not copied correctly from webUI to the config.json
Could you provide me the information how your username and password look like? 
For example which special characters are used?

Alacika, 2024-11-15:
@dewenni I was unable to reproduce the login error again. I did everything as you asked, and the config.json contained the entered password correctly before and after the restart. The login was also flawless. I don't know what it was...