dewenni/ESP_Buderus_KM271 #116: Modul geht nach 2-3 Stunden offline und muss wieder geflashed werden

DonCanDo, 2024-10-13:
Hallo,

ich bin langsam mit der Software am Verzweifeln. Auch nach dem mehrmaligen neu flashen geht sie nach 2 3 Stunden einfach so aus dem Netz. Ändern tut sich derweil nichts es passiert einfach.

Ich habe schon die externe Stromversorgung dran. Der Wifi Empfang ist auch stetig bei über 75%. Ich weiß nicht mehr weiter.

Wenn es sich festgerammelt hat leuchtet D1 grün D2 grün und D3 orange. 

Strom habe abgeklemmt und wieder dran dann haben D1 und D2 grün geleuchtet. Offline war es dennoch.

Dann den Strom noch mal weg gemacht für mehr als 5 Minuten.......nichts außer schnelles blinken mit einer kleinen Pause nach 11 mal blinken. Gute 15 Minuten später war die Oberfläche wieder da. Als Fehler stand natürlich Power-on reset da.

DonCanDo, 2024-10-13:
Und wieder war es offline, dass ich den Strom 10 Minuten unterbrechen musste.

DonCanDo, 2024-10-14:
Kann es sein, dass der Zugriff von mehreren Geräten auf die WebUI zum Verbindungsabbruch führt? Ich habe zu Hause ein HD Fire hängen dass die Seite permanent offen hat. Jetzt wollte ich grade von unterwegs für Frau die Heizung einschalten und keine 2 Minuten später ist es offline. Die Nacht ist es soweit online geblieben.

dewenni, 2024-10-14:
Ich kann dir da leider im Moment nicht helfen.
Mein System läuft recht stabil. Vor einigen Tagen hat es mal von alleine neu gestartet, lief dann aber wieder normal seit dem.

Das es sich so "aufhängt" das nix mehr geht, kenn ich so nicht.
Es ist dann aber auch nicht in dem Setup-Mode wo es einen Access-Point aufmacht, oder?

paxrsi, 2024-10-14:
> Kann es sein, dass der Zugriff von mehreren Geräten auf die WebUI zum Verbindungsabbruch führt?

Den Eindruck hatte ich nach der Installation meines Moduls auch (Version 4.1.0). Ich stand neben der Heizung, war mit dem Smartphone auf dem Webfrontend des Moduls, ging an den PC, um mir das Webfrontend von dort aufzurufen und kurze Zeit später ging gar nichts mehr. Das ist mehrfach so passiert. Ich habe da jedes Mal die Heizung aus- und wieder eingeschaltet. Irgendwann habe ich es dann vermieden, mit zwei Geräten auf dem Webfrontend zu sein und wenn ich mal drauf bin, dann immer nur sehr kurz. Das meiste läuft bei mir über Home Assisstant und das läuft bis auf das hier im Diskussionsbereich beschriebene reproduzierbare Problem beim Neustart von Home Assisstant auch so gut, dass ich keinen weiteren Handlungsbedarf gesehen habe. Das ist bei mir auch fast der einzige Grund, warum ich auf das Webfrontend gehe, weil ich nach jedem Neustart von Home Assisstant jedes Mal auch das Buderus-Modul neu starten muss. Das mache ich dann übers Webfrontend des Moduls und starte nicht die ganze Heizung neu.

DonCanDo, 2024-10-14:
Moin,

danke dass du deine Erfahrungen hier auch einbringst. Ich war ja dann gestern so dermaßen genervt, dass ich die externe Stromversorgung über 3 Verlängerungskabel gemacht habe wo ich dann in der Küche den Stecker ziehen kann weil ich nicht alle 10 Minuten mehr in den Keller wollte. Der Flashapadter hängt mit USB Verlängerung auch schon dauerhaft an der Regeleinheit.

Also dann bräuchte ich mich nicht wundern. Ich habe im Flur ein Fire Tab hängen als Dashboard. Dort ist die Seite bislang die ganze Zeit offen weil es ja über den HA nicht ansteuern kann. Aber wenn das tatsächlich ein Problem ist, dann bin ich recht lost.

Kannst du mir mal sagen, ob du entgegen der Anleitung etwas anders gemacht hast bei den Einstellungen zum Homeassistant?
Das Problem habe ich ja in einem anderen Thread auch, dass die HA Integration auf Teufel komm raus das Board nicht findet. Ich schätze wenn ich es endlich in den HA bekäme wäre die WebUI auch nicht mehr zwingend notwendig. 

Ich werde mal schauen heute Abend ob ich das Board via 2 Zugriffe bewusst himmeln kann oder mit einem Zugriff bewusst nicht.

paxrsi, 2024-10-14:
> Kannst du mir mal sagen, ob du entgegen der Anleitung etwas anders gemacht hast bei den Einstellungen zum Homeassistant?

Hhmm... Bin mir nicht sicher, ob ich das noch alles rekapitulieren kann. Aber ich kann dir sagen, wie meine Einstellungen im WebUI sind:
Unter Einstellungen > MQTT:
- Aktivieren auf AN
- Config Nachricht als retain auf AUS (hhmmm... vielleicht sollte ich das mal auf AN setzten, das könnte mein Problem lösen)
- Home Assistant auf AN
- Server: <Homeassistant Hostname> (Der Server hat in der Fritzbox diesen Namen und eine Fixe IP V4 Adresse)
- Port: 1883
- Topic: ESP-Buderus-KM271
- Benutzer: mosquitto (der muss in Home Assistant für die MQTT-Integration angelegt werden, frag mich aber bitte nicht, wo... Der User kann auch anders heißen.
- Passwort: Passwort halt
- Home Assistant Topic: homeassistant
- Home Assistant Device Name: Logamatic

DonCanDo, 2024-10-23:
Hey Sven,

hier kann erstmal pausiert werden. Ich hatte noch die Version 4.1.1 drauf. Habe jetzt auf 4.1.3 geupdated und gucke jetzt mal welche Probleme noch Bestand haben.

DonCanDo, 2024-10-30:
Kannst du mal deine aktuelle Version prüfen. Mir geht schon wieder das Messer in der Tasche auf. Das Modul geht einfach so offline heute schon das vierte Mal. Den Fehler kann ich nicht nachforschen weil ich es nur mit Strom weg zum laufen bekomme.

gonium, 2024-11-01:
Moin! Erst einmal: Danke für das Projekt, ich hab diese Woche die Hardware (ein KM271-Modul + MAX3232 + ESP32) an meine Logamatic angeschlossen. Lief erstmal stressfrei, modulo ein USB-Netzteil, welches nicht genug Strom lieferte. Das ist ja aber einfach zu lösen.

Nun, mit stabiler Stromversorgung, bekomme ich auch Werte im Webinterface angezeigt -- allerdings nur kurz. Nach längstens einer Stunde ist der ESP32 tot bzw. reaktionslos. Ein Strom aus/Strom an hilft dann für kurze Zeit. Das tritt bei der aktuellen 4.1.3 auf (ich hab das Binary aus den Releases via esptool.py geflasht). 

Testweise hab ich nun die Version 3.3.1 geflasht, auch aus den Releases. Damit läuft das Ganze nun seit ein paar Stunden stabil. Ich beobachte das mal weiter.

dewenni, 2024-11-01:
Das ist echt komisch.
Ich habe zwei Systeme laufen und kann so etwas bisher nicht beobachten.

Wenn das bei dir so zuverlässig und häufig auftritt, könntest du vielleicht mal schauen ob du hier noch ein paar Debug Daten liefern kannst? Wäre es möglich das du dich parallel mit dem PC mal ranhängst und schaust ob der ESP irgendwelche Meldungen rausgibt wenn der Zustand eintritt?

Mir fehlt hier leider bisher jeglicher Anhaltspunkt was passiert wenn der ESP nicht mehr erreichbar ist.
Ich habe auch keine Möglichkeit gefunden den Backtrace im Falle eines Crashs über den Neustart hinweg zu speichern.

dewenni, 2024-11-01:
Es wäre auch in solchen Fällen mal interessant zu wissen ob nur der Webserver nicht mehr reagiert oder der ganze ESP nicht reagiert.
Man könnte z.B. mal testen ob der ESP noch auf einen Ping reagiert
Oder ob er sich noch per Telnet bedienen lässt.
Könnt ihr das bitte auch mal testen?

gonium, 2024-11-01:
Zum Verhalten vorher: Ich konnte den ESP32 weder pingen noch via Telnet darauf zugreifen. Auch der DHCP-Lease ist abgelaufen ohne Renewal. Kommunikationstechnisch war der komplett tot. Ich hab nun einen Laptop im Heizraum deponiert, dort läuft ein ``screen`` mit, der auf die serielle Konsole schaut. Tjoa, fun fact: Es ist ein Heisenbug. Bislang alles unauffällig, ich beobachte das mal so -- ist aber natürlich keine Dauerlösung. Irgendwo gibt es anscheinend einen Unterschied in den Installationen. 

## Hardware?

Ich verwende [dieses Devboard](https://www.az-delivery.de/en/products/esp-32-dev-kit-c-v4) mit diesem [MAX3232 Breakout Board](https://www.amazon.de/dp/B09L1BB6F8). Verkabelt ist das Ganze mit den Dupont-Kabeln vom Breakout Board, das serielle Kabel ist ungeschirmt. EMI-technisch ein ziemlich schlechter Aufbau, allerdings würde ich bei EMI-Problemen erwarten, dass sich der ESP32 neu startet. Ist ein Watchdog aktiv?

## Software?

Vielleicht verwenden wir ja andere Images. Ich hab eben ``buderus_km271_esp32_flash_v4.1.3.bin`` runtergeladen und geflasht. Hat dieses Binary einen anderen SHA256-Hash als das, was Du verwendest? Ist auf Deinen Boards irgendwo sonst ein Überbleibsel einer alten Installation, die ich nicht habe?

gonium, 2024-11-01:
Hm, noch ein Unterschied: Ich hab eben unter MacOS geflasht, vorher hab ich das unter Windows gemacht. Beide Male mit ``esptool.py``.

gonium, 2024-11-01:
Nope, passiert wieder. Seriell sieht es so aus:

```
Saving config file...
Saving config file OK
LittleFS successfully started
WiFi Mode STA - Trying connect to: <WLAN-ESSID>
Telnet Server: running!
Connected to AP successfully!
WiFi connected
IP address:
192.168.<x.y>
MQTT - connection attempt: 1/5
MQTT connected
Stop doubleResetDetecting
Saving config file...
Saving config file OK
New Client connected
New Client connected
Received - Element ID: p10_log_refresh_btn, Value: true
New Client connected
New Client connected
New Client connected
New Client connected
```

Das Modul reagiert nicht mehr übers Netz, weder ``ping`` noch ``telnet``. Sieht so aus:

```
$ ping 192.168.107.87
PING 192.168.107.87 (192.168.107.87): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
^C
--- 192.168.107.87 ping statistics ---
4 packets transmitted, 0 packets received, 100.0% packet loss
```

```
$ telnet 192.168.107.87
Trying 192.168.107.87...
^]

]^]^]^]telnet: connect to address 192.168.107.87: Network is unreachable
telnet: Unable to connect to remote host
```

Auch in Home Assistant kommt nix mehr an:

[image]

gonium, 2024-11-01:
Nach Reset lief das Setup nun etwa 'ne Stunde. Ich mache ein Downgrade auf 4.0.0.

gonium, 2024-11-01:
Nope, auch diese Version lief vielleicht 15 Minuten.

paxrsi, 2024-11-01:
Hast du mehrere Webclients offen, die gleichzeitig auf die Webseite des KM271 Moduls zugreifen? Damit hatte ich gefühlt auch regelmäßig Abbrüche.

gonium, 2024-11-01:
Bin mir gerade unsicher. Neues Experiment: 4.0.0, gar keine Browser offen, ich gucke nur ob in Home Assistant was ankommt.

gonium, 2024-11-01:
Nope, wieder weg. Ich hab keine Idee, was ich sinnvolles probieren könnte.

Iam-klm, 2024-11-01:
Hallo
Bei meinen beiden z.Z. mit 4.1.3 laufenden Systemen ist es auch so. Zugriffe  oder offenen Browser und das Modul ist dann kurz oder lang tot.
Wenn ich es nicht anspreche liefert es problemlos Wochen lang über mqtt alle Werte.
Ich persönlich hatte das Verhalten bei den Vorgängerversionen auch.
Zugriff über Browser ja, aber wenn nachgesehen Verbindung sobald wie möglich schließen.
Gruß Klemens

dewenni, 2024-11-02:
> Bin mir gerade unsicher. Neues Experiment: 4.0.0, gar keine Browser offen, ich gucke nur ob in Home Assistant was ankommt.

Zur Info: HomeAssistant gibt es erst ab 4.1.0

Ich bin gerade dabei ein Update auf das neue Arduino Framework V3 zu machen und auch von der AsyncWebserver Library gibt es einige neuere Versionen. Allerdings ist Arduino V3 auch nochmal vom Speicherbedarf angewachsen und es wird zunehmend knapp bei einem 4MB Board.
Vielleicht kann ich aber in den nächsten Tagen hier mal eine Testversion für euch zu Verfügung stellen.

DonCanDo, 2024-11-02:
Moin,

hier mal die neusten Infos aus meiner Warte.

Das Modul läuft jetzt erstmal grundsätzlich unter externem Netzteil. Heute morgen ging es wieder offline nachdem ich auf Bedienung ging auf der Weboberfläche. Mittlerweile habe ich sicher gestellt, dass die Weboberfläche immer nur einmal geöffnet ist, denn dass dabei zum Crash kommt kann ich mittlerweile bestätigen. 

Derzeit hatte ich einmal ein Downgrade auf 4.1.2 gemacht. Dabei hatte sich an den Problemen aber auch nichts geändert. Die WLAN Verbindung ist stets stabil.

gonium, 2024-11-02:
Also die Version 3.3.1 läuft nun stabil seit gestern Abend. Insofern gehe ich davon aus, dass es nicht an meiner Hardware liegt, sondern das Problem mit V4.0.0 eingeführt wurde.

dewenni, 2024-11-03:
So, ich habe im Zuge der Arbeiten dem nächsten größeren Update ein Problem entdeckt, welches für manche der hier genannten Probleme verantwortlich seien könnte.
Bitte testet mal das aktuelle Release [v4.1.4](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.1.4) und gebt mir Rückmeldung ob es damit besser läuft.

Zur weiteren Erklärung:
Ich habe in potenzielles Problem entdeckt und beseitigt, was dazu führen konnte, dass es zu Problemen kommt wenn sich ein WebClient verbindet.

gonium, 2024-11-04:
Danke für Deine Mühe -- ich hab die v4.1.4 nun installiert. Ich melde mich!

DonCanDo, 2024-11-05:
Moin,

lieben Dank für das Kümmern. Habe die Version gestern Nachmittag via ota-update rauf gemacht und dann mal bewusst nicht neu gestartet, damit ich sehe wenn was anderes als reset Grund drin steht. Und heute morgen war es dann bereits wieder offline. D1 hat durchgängig grün geleuchtet und D3 durchgängig orange. Mittlerweile habe ich eine USB Steckdose an der Heizung dran wo das Modul permanent extern Saft bekommt. Als Eingriff habe ich nur mein Fire 10HD im Flur das die Weboberfläche 24/7 anzeigt. Ich schau mal wie das heute den Tag über läuft. Kann vielleicht auch ein Problem auftreten, wenn die Web Oberfläche auf dem Fire HD permanent offen ist und ich vom Homeassistant Befehle schicke?

gonium, 2024-11-05:
Ist etwas weniger wackelig, aber noch weit entfernt von stabil. Gestern Abend lief das Ganze soweit, etwa um Mitternacht fiel mir aber auf, dass die Werte nicht mehr aktualisiert werden. Die Webseite war allerdings erreichbar. Ich hab das Modul dann neu gestartet. Seitdem hab ich keinen Webbrowser mehr offen gehabt.

In Home Assistant kommen nur sporadisch Werte an:

[image]

gonium, 2024-11-08:
Kurzes Update: Da ich sowieso ESPHome-Komponenten verwende, nutze ich nun die ESPHome-Komponente von themole78: https://github.com/the78mole/esphome_components. Damit läuft mein Setup seit ein paar Tagen stabil.

dewenni, 2024-11-08:
> Kurzes Update: Da ich sowieso ESPHome-Komponenten verwende, nutze ich nun die ESPHome-Komponente von themole78: https://github.com/the78mole/esphome_components. Damit läuft mein Setup seit ein paar Tagen stabil.

das ist natürlich auch eine gute Alternative. Durch meine Home Assistant Unterstützung haben sich die Anwendungsfälle beider Software angefangen zu überlappen.

Ich bin im Hintergrund dabei meine Software zu überarbeiten und bin natürlich bemüht die aktuellen Probleme zu beheben.
Ich kann es zwar derzeit nicht wirklich nachstellen, aber ich versuche dennoch einige Dinge anzupassen und zu optimieren.
Vielleicht magst du die überarbeitete Version ja dann trotzdem später nochmal testen.

paxrsi, 2024-11-10:
> Bitte testet mal das aktuelle Release [v4.1.4](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v4.1.4) und gebt mir Rückmeldung ob es damit besser läuft.

Danke für das Update! Ehrlich gesagt: Im Moment macht das Modul alles was ich brauche und da wir gerade am Anfang der Heizsaison stehen, bin ich da eher vorsichtig mit Experimenten. Daher habe ich für mich beschlossen, erst mal keine Änderungen vorzunehmen, so lange es nicht unbedingt sein muss. Ich will diesen Winter möglichst viele Realdaten sammeln, mit denen ich dann später hoffentlich die energetische Sanierung präziser planen kann. Wenn die Heizsaison rum ist, vielleicht so ab Mai, bin ich in dieser Hinsicht vielleicht wieder etwas mutiger :)

dewenni, 2024-11-25:
@gonium @Iam-klm @DonCanDo @paxrsi 
Gerne mal das neue Release V5.0.0 testen und hier Rückmeldung geben.
Ich hoffe damit das Thema "mangelnde Stabilität" behoben zu haben.
Ich habe hier einiges in Bezug auf die Verbindung zwischen ESP und WebClient überarbeitet was vermutlich in der Vergangenheit die Ursache für die Probleme war.

Iam-klm, 2024-11-25:
Rückmeldung 5.0.0
Bei meinen beiden Systemen lief Update auf 5.0.0 OTA problemlos, zuvor von feste IP auf DHCP umgeschaltet.
So wie es aussieht, ist die ESP - WebClient Erneuerung der Erfolg. Konnte bis aktuell keine Abstürze feststellen.
Auch nach dem heutigen Update des mqtt Servers auf dem ioBroker, fing sich diese Version wieder durch Neustart und meldete sich wieder am Server an.
Dass der ibfo/client Name nicht mehr dem Hostname entspricht, hattest du ja schon gesehen. Der esp32-??????????? ist leider für uns Menschen immer eine kleine Hürde.

Danke für die vielen Stunden um die stabile 5.0 zu liefern. Diese Version hat schon mal ihren Stellenwert wieder.👍✔

Ich will bestimmt nicht meckern, nur ein Vorschlag und eine Schönheitssache:

Bei rot scheint noch ein Leerzeichen voran zu stehen.
Bei gelb wäre es schön, wenn der angewählte Menüpunkt nach dem Verlassen mit der Maus, noch seinen Fokus behalten würde.

[image]

Bei den Systeminformationen sind die Prozente schon beeindruckend. 🤣

[image]

Grüße

Iam-klm, 2024-11-26:
Hi Sven.
Nachmeldung zu 5.0.0.
Unter Menue Brenner ist eine Englisch - Deutsch Mischung.

[image]

Gruß Klemens

dewenni, 2024-11-27:
@Iam-klm 
ich habe die Punkte soweit alle berücksichtigt, bis auf den ersten. Das ist aber auch kein relevanter Punkt.
Es ist aber noch ein anderer wichtiger Punkt aufgetaucht , der es nötig macht auf die neue Version zu aktualisieren #131 

[Release v5.1.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v5.1.0)

dewenni, 2024-11-27:
ich schließe das jetzt mal hier - sollten weitere neue Themen auftauchen, dann gerne für jeden Punkt ein issue anlegen oder die Diskussion nutzen.

closed 2024-11-27 by dewenni