dewenni/ESP_Buderus_KM271 #117: Ideen: zeitgesteuerter Reboot oder watchdog und Fehlerspeicher?

Kallrick, 2024-10-21:
Hallo, 

hatte merkwürdiges beobachtet: mein ESP lief monatelang stabil, plötzlich innerhalb einer Woche ist er mehrfach ausgestiegen, keine Reaktion, keine Daten über mqtt mehr, Webinterface nicht erreichbar. 
Da ich dann nur den Stecker ziehen konnte, konnte ich keine ursachliche Fehlermeldung mehr sehen.
a) Daher die Frage, ob man nicht die letzten 5 Fehler Gründe in einen Speicherbereich (littlefs?) eintragen kann, der sich nicht durcheinen Reboot löscht. So könnte man nach einem Stecker ziehen auch noch den Grund des Problems davor feststellen
b) ich glaub das gibt es schon, ein Watchdog, der bei nicht Reaktion einen eigenen Neustart macht. Aber scheinbar macht er das noch nicht bei dem Ereignis, von dem ich betroffen war, vielleicht kann man das konfigurieren und noch was wie einen ping aufnehmen, wenn von da nichts zurück kommt, dass er dann neu startet oder so
c) ggf einen Timer, den man in der Oberfläche konfigurieren kann wie "nächtlicher Reboot um 00:00"

Mir ist aufgefallen, dass das Einfrieren erst aufgekommen war, als ich öfter mal wieder auf die Weboberfläche geschaut hatte, vielleicht war dadurch was überlastet oder ein Speicherüberlauf entstanden? Nachdem ich jetzt "alles in Ruhe" gelassen habe, läufts wieder stabil durch seit 2 Wochen.

Viele Grüße
Patirck

ferengi82, 2024-11-02:
Ich weis nicht ob es zur Fehlerlösung beiträgt, ich habe die Platine (0.6) schon eine ganze Zeit lang eingebaut (mit der ESP-Home Firmware) aber mich wegen Zeitmangel nie richtig damit beschäftigt, mir ist nur aufgefallen das sie sich in unregelmäßigen Abständen aufgehangen hat.
Vor ein paar Wochen habe ich dann die Firmware gewechselt, aber hier das gleiche Problem.
Löse jetzt ganz pragmatisch 1x um Mitternacht einen reset über Homeassistent aus => seit dem läufts eigentlich ohne Probleme.

dewenni, 2024-11-23:
@Kallrick bitte das neue Release testen [v5.0.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v5.0.0)

closed 2024-11-25 by dewenni