dewenni/ESP_Buderus_KM271 #118: WW Zirkulation Entität auf einmal weg

DonCanDo, 2024-10-24:
Ich verstehe nicht, wieso heute auf einmal die Entität weg war und weg ist.

[image]

DonCanDo, 2024-11-01:
Kann mir hier jemand weiter helfen?

dewenni, 2024-11-01:
Ist das denn die neue Entität zum Bedienen oder noch die alte Entität als Sensor?

Die alte als Sensor ist ja weggefallen weil es ja jetzt eine Entität zur Bedienung gibt vom Typ "Option"

DonCanDo, 2024-11-08:
Also die Einstellung geht. Aber da wird mir nicht angezeigt und getrackt wenn die Zirkulationspumpe läuft

dewenni, 2024-12-29:
ist hier noch was offen?

DonCanDo, 2024-12-29:
Kannst du zu machen die Entität ist wieder da

Grüße Steven

closed 2024-12-29 by dewenni