dewenni/ESP_Buderus_KM271 #119: Probleme beim Speichern von Einstellungen

maddinmm, 2024-10-30:
Ich habe aktuell Probleme, dass Einstellungen dauerhaft gespeichert werden. 

Versucht wurde schon die Version von 4.1.1 bis 4.1.3 wobei es bei allen gleich ist. 
Auch die Browser (Firefox, Safari,  Edge, Chrome) wurden getestet. 
Im Serial Monitor werden die Änderungen angezeigt die ich eintippe. 

Fehlerbeschreibung: 
Beim ändern der GPIO bleiben dauerhaft die Ports auf folgende Werte
LED-Heartbeat -> 21
LED-Logmode -> 17
LED-WiFi -> 21
LED-Ölzähler -> 22
Impuls-Ölzähler -> Funktioniert genau so wie es sollte. 

Beim einstellen vom MQTT Retain Flag wird dieser immer auf "AUS" gestellt. Laut config.json ist dieser aber auf "True", jedoch werden die Topics nicht auf "retain" gesendet. 

Hat jemand noch das selbe Symtom wie ich? 
Wo kann man diese angaben auch Statisch in den Code einpflegen? 

VG Martin

dewenni, 2024-11-03:
du schreibst, dass du im Serial Monitor siehst, wenn du im Webinterface Werte änderst.
Siehst du auch, dass die Werte dann in der config gespeichert werden?
Das müsste man nämlich auch im Log erkennen.
Dann wäre interessant ob du die Änderungen auch in der config.json siehst. Die kannst du dir unter "Tools" im WebUI anzeigen lassen.
Wenn die Werte da drin stehen und vor allem auch nach dem Neustart, dann ist das erstmal nicht das Problem.
Dann könnte das Problem eher daran liegen, dass die Werte nach einem Neustart im WebUI nicht aktualisiert werden.
Wenn dem so ist, dann teste doch bitte auch mal das neue Release v4.1.4

maddinmm, 2024-11-03:
Also ich habe dir das mal überprüft: 

Ändere ich die Werte der GPIO dann ändert sich in der config.json leider nix. Zum versuch habe ich auch schon den ESP32 komplett formatiert um wirklich einen "leeren" Datenstand zu bekommen, jedoch ändert dies nichts an dem Problem. 

Ich will da auch nicht ausschließen, dass das Problem auf meinem ESP entsteht und dein Code einwandfrei ist.  

Version 4.1.4 werde ich dir testen und dazu berichten. 

Zwischenzeitlich habe ich evtl. einen Verbessrungsvorschlag für dich. 
Aktuell ist es so, dass du nur geänderte Werte in MQTT schreibst, dies ist jedoch mit der Anzeige in Grafana und Aufzeichnung in InfluxDB suboptimal. Eventuell könntest du dies in einer Version einpflegen, dass alle Nachrichten zyklisch alle x Sekunden gesendet werden sollen. 

Workaround meinerseits: In NodeRed einfach ein tigger Element dazwischen und Steuerung neu starten, so werden die Werte so lange wiederholt bis eine Änderung entsteht.

dewenni, 2024-11-04:
Danke für die Rückmeldung.
Wir müssen aber mal noch herausfinden wo das Problem auftritt.

Beim Start vom ESP solltest du auf jeden Fall folgendes im Monitor lesen können:
`LittleFS successfully started`

und wenn ich einen Wert bei z.B. den GPIOs ändere sollte folgendes zu lesen sein:
`Received - Element ID: p12_gpio_trig_oil, Value: 17`

und kurz darauf dann die Meldung das es auch in die Datei gespeichert wurde:
`config successfully saved to file: /config.json`

wenn ich mir das dann noch vor einem Neustart unter "Tools -> öffne config.json" anzeigen lasse, dass sehe ich auch die Änderung:
[image]

Und nach einem Neustart sollte es natürlich auch noch so ein, da dann die Datei wieder gelesen wird.
Kannst du das so bestätigen?

Ansonsten, welchen ESP nutzt du? LittleFS müssten sie aber denke doch alle unterstützten?!

maddinmm, 2024-11-07:
Also ich habe diesbezüglich mal alles soweit durchgetestet und habe mich mit PlatfomIO genauer auseinandergesetzt. 

LittleFS successfully started -> Kommt bei mir und es ist auch Grundsätzlich wird die config im ESP gespeichert. 

Ich habe mir die Logausgabe bei 2 verschiedenen Modellen von ESP angeschaut. (ESP-WROOM-32 und ESP-WROOM-32UE)

Bei beiden wird die Speicherung einiger Werte nicht durchgeführt. (jedoch bei beiden unterschiedlich)
 
ESP-WROOM-32:
`Received - Element ID: p12_gpio_km271_rx, Value: 2
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_km271_tx, Value: 4
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_led_heartbeat, Value: 17
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_led_logmode, Value: 21
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_led_wifi, Value: 22
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_led_oil, Value: 25
`
Hier wird z.B die OIL LED nicht gespeichert. Der Rest schon. 

ESP-WROOM-32UE:
`Received - Element ID: p12_gpio_km271_rx, Value: 2
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_km271_tx, Value: 4
config successfully saved to file: /config.json
Received - Element ID: p12_gpio_led_heartbeat, Value: 17
Received - Element ID: p12_gpio_led_logmode, Value: 21
Received - Element ID: p12_gpio_led_wifi, Value: 22
Received - Element ID: p12_gpio_led_oil, Value: 25
`
Hier werden sogar noch mehr Werte nicht in die Config übernommen. 

Sobald es im Log steht (`config successfully saved to file: /config.json`) ist es auch gespeichert und bleibt bestehen.

Ich denke mal, dass es sich hierbei um ein Problem des ESP Handelt, da bei beiden Fällen die selbe Binary hochgeladen habe. Falls es bei allen anderen funktioniert, brauchst du an deinem Source Code nichts verändern, da es nur ein Problem von meinem ESP kommt. 

Zwei Sachen sind mir noch Aufgefallen.
1. im Logbuch sollten eig. Werte kommen. Tut es bei mir nicht, hab sämtliche Browser erfolglos Probiert. 
2. Ab und zu kann es vorkommen, dass sich der ESP aufhängt, es wäre gut wenn man wie bei jedem Microcontroller den Watchdog aktiviert (wenn MQTT nicht erreichbar ist, machst ja auch schon einen Reset vom ESP).

janihani, 2024-11-19:
Auch in Version 4.1.4 scheint das Problem (noch) zu existieren, dass die Daten nicht dem retain flag gesendet werden, obwohl in der config.json config_retain auf true gesetzt ist.

 "mqtt": {
    "config_retain": true,
  },

Beim Import der config.json Datei wird retain nicht aktiviert, obwohl es in der Datei auf true gesetzt ist.

Auch bei den GPIO Einstellungen ist mir aufgefallen, dass die Pins nicht (korrekt/dauerhaft) übernommen werden.

maddinmm, 2024-11-19:
> Auch in Version 4.1.4 scheint das Problem (noch) zu existieren, dass die Daten nicht dem retain flag gesendet werden, obwohl in der config.json config_retain auf true gesetzt ist.
>
> "mqtt": { [...]

Wissen wir schon ;-) ist aber aktuell belanglos, da eine ganz neue Version gebaut wird und wenn man vorm Kompilieren die Pins und Einstellungen statisch setzt funktioniert es auch wunderbar

dewenni, 2024-11-19:
Die überarbeitete Version läuft gerade im Test. Ich denke nächstes Wochenende kann ich ein Release machen.

maddinmm, 2024-11-19:
> Die überarbeitete Version läuft gerade im Test. Ich denke nächstes Wochenende kann ich ein Release machen.

Mach dir keinen Stress und teste sauber durch bevor du ein Release machst.

janihani, 2024-11-19:
> wenn man vorm Kompilieren die Pins und Einstellungen statisch setzt funktioniert es auch wunderbar

Ich denke, dass folgende Stellen im Quellcode geändert werden müssen um mit statischen Werten zu arbeiten:

    config.mqtt.config_retain = true; //doc["mqtt"]["config_retain"];

    config.gpio.led_wifi = 21; // doc["gpio"]["led_wifi"];
    config.gpio.led_heartbeat = 18; // doc["gpio"]["led_heartbeat"];
    config.gpio.led_logmode = 27; // doc["gpio"]["led_logmode"];

Leider bekomme ich beim kompilieren am Ende einen Fehler:

Successfully created esp32 image.
merge_bin([".pio\build\esp32\firmware.bin"], [".pio\build\esp32\firmware.elf"])

[Nachfolgend die Pfade zwecks Lesbarkeit gekürzt]

esptool.py --chip esp32 merge_bin --flash_size 4MB -o ... .pio\build\esp32/firmware_merged.bin 0x1000 ... .pio\build\esp32\bootloader.bin 0x8000 .pio\build\esp32\partitions.bin 0xe000 ... boot_app0.bin 0x10000 ... .pio\build\esp32/firmware.bin 

esptool merge_bin: error: argument <address> <filename>: Address ".. .pio\build\esp32/firmware_merged.bin" must be a number       
*** Error 2
*** [.pio\build\esp32\firmware.bin] D:\Janaczeck\Dokumente\Coding Projekte\logamatic\ESP_Buderus_KM271-4.1.4\ESP_Buderus_KM271-4.1.4\.pio\build\esp32/firmware_merged.bin: No such file or directory

Schade, bin leider noch nicht so firm

maddinmm, 2024-11-20:
```
//config.gpio.led_wifi = doc["gpio"]["led_wifi"];
//config.gpio.led_heartbeat = doc["gpio"]["led_heartbeat"];
//config.gpio.led_logmode = doc["gpio"]["led_logmode"];
//config.gpio.led_oilcounter = doc["gpio"]["led_oilcounter"];
//config.gpio.trigger_oilcounter = doc["gpio"]["trigger_oilcounter"];

config.gpio.led_wifi = 22;
config.gpio.led_heartbeat = 21;
config.gpio.led_logmode = 17;
config.gpio.led_oilcounter = 25;
config.gpio.trigger_oilcounter = 33;
```

So habs ich gemacht und läuft auch so.

dewenni, 2024-11-23:
Bitte das neue Release testen  [v5.0.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v5.0.0)

maddinmm, 2024-12-02:
Also habe nun die Version 5.0.0 getestet. Die von oben beschriebenen Fehler sind mit dieser Version behoben.

dewenni, 2024-12-02:
@maddinmm 
Bitte geh noch auf die 5.1.0.
In der 5.0.0 gab es noch einen Fehler im MQTT Handling

maddinmm, 2024-12-08:
> @maddinmm Bitte geh noch auf die 5.1.0. In der 5.0.0 gab es noch einen Fehler im MQTT Handling

Sieht soweit gut aus. hab bisher keinen fehler Feststellen können. 

Habe aber noch einen anderen Fehler gefunden, der wurde dir sicher noch nicht gemeldet und ist ein reiner Anzeigefehler. 

Wenn man die Betriebsart beim WW auf Manuell/Nacht stellt, dann wird im Dashboard nur immer Tag angezeigt, über MQTT kommt aber der richtige Wert. 

[image]
[image]

dewenni, 2024-12-08:
@maddinmm kann es sein, dass es nur nicht aktualisiert wurde?
Bleibt das auch so wenn du einen Browser refresh machst?

maddinmm, 2024-12-08:
Japp, mit Refresh im browser wars nicht abgetan, sobald ich einen neuen Inkognito Tab auf mache schaltet es korrekt um, hat sich also erledigt. 

Hab vor kurzem bei mir einen Abgastemperaturfühler installiert, falls jemand nicht die "überteuerte" Variante von Buderus kaufen will, der braucht sich nur einen NTC 100k besorgen. Gibts dann für ein paar Euro in der Bucht. 

Man muss auf der Platine D5 entfernen (für Temperaturen >80°C) und R35 + R39 (0R) bestücken. Das könntest du evtl. in deine Doku mit aufnehmen. Kann ja sein, dass es den einen oder anderen Interessiert.

Danke noch mal für deine Hilfe und ich denke dass diese Issue nun endgültig abgeschlossen ist.

dewenni, 2024-12-08:
ok, danke für die Rückmeldung. Ich muss mal sehen das ich den Refresh der Daten nach dem Absetzten eines Befehls nochmal optimiere. Ich habe generell das Aktualisieren bei der V5 deutlich reduziert, weil dies eines der Hauptprobleme für die Instabilität war. Jetzt ist es ggf. in der ein oder anderen Situation noch etwas zu restriktiv. Schau ich mir nochmal an.

Bezüglich dem Sensor. Ja ich habe mir auch einen im freien Handel gekauft. Das ist ja auch in der Doku enthalten:
https://github.com/dewenni/ESP_Buderus_KM271?tab=readme-ov-file#optional-exhaust-sensor

janihani, 2024-12-09:
Zunächst erstmal ein herzliches Dankeschön, für die neue Version, die seit mehreren Tagen problemlos läuft.

Bezüglich der Konfiguration der GPIO-Pins habe ich nichts mehr unternommen, da ich den ESP in einer Verteilerdose untergebracht habe und somit die LEDs nicht mehr wirklich sichtbar sind. 

Bezüglich MQTT ist mir aufgefallen, dass nur die Config-Nachrichten als retained übermittelt werden, was die Einstellung Config Nachrichten als retain auch Nahe legt.

Alle anderen Nachrichten werden ohne retained übermittelt, was bei mir dazu führt, dass wenn ich Home Assistant neu starte und damit auch den MQTT Broker, es vereinzelt sehr lange dauert, bis wieder Werte (in meinen Heizungs-Dashboard) angezeigt werden.

Von meinem Python-Script [(Link)](https://homematic-forum.de/forum/viewtopic.php?f=18&t=26955&sid=1e00e12bc302c2ff77779245eb7ee5ba&start=250#p796627) war ich etwas verwöhnt, weil ich alle Nachrichten mit retain übermittelt habe, so dass die Werte auch nach einem HA Neustart sofort wieder zur Verfügung standen.

Bei der Durchsicht des Quellcodes ist mir in message.cpp aufgefallen, dass in 

case KM_TYP_CONFIG:
   mqttPublish(newCfgTopic, value, config.mqtt.config_retain);

retain in Abhängigkeit der Konfiguration gesetzt wird.

In allen anderen Fällen ist retain (hardcoded) auf false gesetzt. Ebenso in basic.cpp und mqttDiscovery.cpp.

In oilmeter.cpp ist hingegen retain auf true gesetzt.

void sendOilmeter() {
  mqttPublish(addTopic("/oilcounter"), uint64ToString(data.oilcounter), true);

Bezüglich des virtuellen Oilmeters habe ich noch eine Frage.
Welche Werte habt ihr für den Verbrauch und die Dichte eingetragen?

Auf meiner Öldüse steht 0,5 US Gal/h, so dass ich hier 1,89271 eintragen würde.
Welchen Wert für die Dichte verwendet ihr, da im Netz ein Bereich 0,820–0,860 kg/l angegeben ist.