dewenni/ESP_Buderus_KM271 #122: WLAN-Connection nicht möglich

bsbs-net, 2024-11-05:
Hallo lieber "dewenni",
heute wende ich mich mit meinem Problem mal hier ans Forum. Ich habe im vor ca. 1/2 Jahr von "The Mole" die KM271 mit WLAN in aktueller und deiner Firmware bereits installiert bekommen. Mit "The Mole" hatte ich vor längerem auch schon Kontakt und einiges getestet bzw. nun in den letzten Monaten vertieft getestet. 

Nun aber zum Problem:
Ich habe die Platine sowohl in einer Buderus HS2105 also auch mit externer USB-Stromversorgung ohne Einbau im Simulationsmodus betreiben. Nach dem Boot läuft es jeweils gut. Schaue mit dem Webinterface drauf, ist alles OK und ich kann Einstellungen vornehmen.
Lasse ich das System dann einfach laufen, läuft es tage- / wochenlang vor sich hin. Die Eine LED ist grün die andere blinkt grün.
Nutze ich das Webinterface jedoch öfters bzw. lassen den Browser offen, verliert er immer wieder die Connection. Manchmal kommt die Connection nochmal aber dann ist relativ schnell Schluß und die LED blinkt auch nicht mehr. Es hilft nur ein Roboot.
Ich habe MQTT mit HA konfiguriert, aber dieses auch ausgeschaltet und sonst auch nichts konfiguriert. Gleiches Bild.
Grundsätzlich finde ich eure Arbeits ja echt Spitzenklassen, Platine und Interface, aber Ziel ist ja ein zuverlässiger Betrieb.
Ich habe den Eindruck, dass der ESP damit aber überlastet ist. Speicherüberlauf oder Fehler oder was auch immer. Wie gesagt, es kommt immer nur dazu, wenn ich ihn mit meinem Webzugriff beschäftige.
Gibt es etwas was ich noch testen oder tun kann?

Meine Wunschvorstellung:
Das Ganze läuft bei mir stabil und grundsätzlich würde mir sogar ein Ausgang als Signalisierung eines Fehlers ausreichen. Also einfach einen Ausgang, welcher mir bei Fehler ohne weitere Details ein Signal gibt. Diesen könnte ich dann in meinem Smarthome (HomematicIP) direkt anbinden und verknüpfen, sowie mich informieren. Dann wäre das Webinterface ausreichend für weitere Schritte oder eben einfach der Gang in den Keller.

Letzte Woche habe ich auch ein OTA-Update auf v4.1.3 gemacht, jedoch der selbe Effekt. Lief ne Woche im Standby in der HS2105 und heute via Web weniger als 10 mal zugegriffen bzw. den View gewechselt -> Reboot.

Schon mal Danke vorab

dewenni, 2024-11-05:
Bitte teste in dem Fall auch noch die v4.1.4
In Bezug auf das WebUI habe ich hier noch ein mögliches Problem gefixt. Speziell beim verbinden eines Web Clients konnte es hier zu Problemen kommen.

Ansonsten gebe ich dir natürlich recht. Das wichtigste ist ein stabiler Betrieb. Interessanterweise läuft es bei mir auch sehr stabil, was es dementsprechend schwer macht einen potenziellen Fehler zu finden.

bsbs-net, 2024-11-05:
Danke für das schnelle Feedback und sorry, ich hatte nach dem Update letzter Woche nicht geschaut, dass es schon eine neue Version gibt. Schwubs drauf gespielt und es schaut glaub ich besser aus. Nachdem ich mit 2 Geräten via Web drauf war lief es einige Zeit und dann gleicher Effekt. Aber mit 2 Geräten macht ja keinen wirklichen Sinn. Nun hab ich nach dem Neustart nur Eines laufen und ich bin mir schon recht sicher, dass es stabiler läuft. Danke. Ich werde es die Tage weiter überwachen und berichten. Nochmals Danke.

Falls noch jemand eine Idee zu meiner Zusatzfrage bezgl. eines Ausgangs auf dem Board hat, welchen man bei einem Fehler nutzen könnte, wäre ich ebenfalls dankbar.

maddinmm, 2024-11-07:
> Nun aber zum Problem: Ich habe die Platine sowohl in einer Buderus HS2105 also auch mit externer USB-Stromversorgung ohne Einbau im Simulationsmodus betreiben. Nach dem Boot läuft es jeweils gut. [...]

Von dem Problem habe ich in meinem Post schon geschrieben, Es hängt sich dabei der ESP komplett auf, dies passiert in der Version 4.1.3 sowie 4.1.4. Die Aktivierung des Watchdog würde hier einfache Abhilfe schaffen. Programmiert man in jedem Microcontroller so ein um ein "Hängenbleiben" zu vermeiden.

dewenni, 2024-11-09:
Wie an anderer Stelle schon geschrieben arbeite ich im Hintergrund an einer neuen Version.
Dort werde ich auch den Vorschlag mit dem Watchdog mit aufnehmen.

dewenni, 2024-11-09:
@bsbs-net 
> Meine Wunschvorstellung:
> Das Ganze läuft bei mir stabil und grundsätzlich würde mir sogar ein Ausgang als Signalisierung eines Fehlers ausreichen. Also einfach einen Ausgang, welcher mir bei Fehler [...]

das kann ich grundsätzlich auch machen. 
Im einfachsten Fall würde man dazu nur einen entsprechenden GPIO definieren. Ergänzend zu den bisherigen.
Wenn der ungleich "-1" ist, dann ist die Funktion aktiv.
Der Ausgang müsste dann aber auch irgendwann oder irgendwie wieder zurückgesetzt werden.
Um das möglichst unkompliziert zu machen, könnte der Ausgang ggf. auch nur einen Impuls von z.B. 10sek ausgeben.
Das genügt ja um etwas angeschlossenes zu triggern.
Ansonsten bräuchte man wieder einen Befehl zum zurücksetzten oder einen Button um webUI.

maddinmm, 2024-11-10:
> Wie an anderer Stelle schon geschrieben arbeite ich im Hintergrund an einer neuen Version. Dort werde ich auch den Vorschlag mit dem Watchdog mit aufnehmen.

Perfekto. So umgehst du einfach relativ unkompliziert das Problem mit dem "hängen" bleiben. Dies hattte ich noch mal geprüft und kann dir einen Tipp geben um dies zu reproduzieren. 

Wenn du einige male die Seite mit F5 oder aktualisieren Button betätigst, wärend er daten aus der Logamatik holt, hängt er irgend wann. Heartbeat LED bleib dann auch auf deren Status.  

So hab ich grundsätzlich nichts an der Firmware zu bemängeln. Sogar dein Design ist wirklich gelungen (Einheitlich, übersichtlich, Dunkle Farben und Automaische Aktualisierung der Daten mit Javascript)

bsbs-net, 2024-11-10:
@ dewenni: Vielen Dank für deine Rückmeldung oben zu dem Thema GPIO.

Das wäre eine prima Option, falls das Möglich wäre. Nach meinem Ermessen würde wie von dir vorgeschlagen ein Impuls von 10sek ausreichen um etwas im eigenen Smarthome zu triggern. Dann könnte der Ausgang wieder zurück gesetzt werden.
Wobei wenn es darum geht, dass die Heizung einen Fehler hat, wäre es wahrscheinlich eh egal, denn man muss dann ja eh Vorort und kann es zurücksetzen. Aber mit 10 Sekunden wäre definitiv "sauberer".

Wenn ich dich also richtig versthe musst du das in die Firmware einbauen und könntest dann mitteilen wo bzw. wie das auf der Platine abzugreifen ist. Sorry, wenn die Frage etwas unbedarft kommt, aber ich bin hier komplett neu und finde eure Arbeit total spannend und echt toll - aber ich habe da ganz schön Nachholbedarf.  Meine Z80-Programmierung ist über 40 Jahre her!   :-O

bsbs-net, 2024-11-10:
@ dewenni: Zur Stabilität hatte ich ja Feedback versprochen. Also das Board läuft bei mir momentan in einer Steuerung ohne Heizkessel, somit nur zum Test und das Ganze sogar am Schreibtisch, ca. 2m von der Fritzbox entfernt. Also bestes WLAN.
Dennoch gibt es immer wieder Abbrüche. Entweder wenn ich das Webinterface aufrufe dann direkt oder auch ohne dass irgendjemand darauf zugreift. Ich sehe dass nun an den PushOver-Meldungen. Hier habe ich dir mal 2 Screenshots von den letzten Tagen beigelegt. Ca. 2/3 der "Software panic resets" komen einfach so, ohne dass jemand aktiv war. Die anderen beim Zugriff. Heute Morgen auch wieder. War dann direkt noch einige male auf dem Webinterface und habe einiges nachgeschaut. Ergebnis jetzt hing es wieder mal komplett und ich musste manuell rebooten. Ich hoffe dieses Feedback hilft dir ein wenig für die weitere Fehleranalyse. Für meinen Einsatz habe ich schon überlegt, den Powerjumber mit einem Schalter nach außen zu legen, sodass man den ESP einfach neu booten kann ohne immer die Heizung abschalten zu müssen. Mal sehen wann ich es in den produktiven Betrieb bringe.

bsbs-net, 2024-11-10:
@ dewenni: Last but noch least - noch eine Anregung. Du schreibst ja, dass du bereits an einer neuen Version arbeitest. 
Also dies fällt aber klar in die Kategorie: Darf ich mir noch was wünschen?
Die Section mit PushOver ist prima und ich habe mir hierdurch dort auch mal einen Testaccount angelegt und dies eingerichtet. Prima. Ich kenne dies aus anderen Anwendungen aber "offener" und nicht fixiert auf einen Lösungsanbieter. Bei meinen Instar-CAMs ist dies mit einem Dialog Alarmserver gelöst. Dort kann man dann die Parameter des eigenen Servers eintragen, sogar bis zu 3 Stück. Das Ganze etwas abgespeckt und vielleicht einfach mit 3 oder 5 festen Parametereingaben, würde es "noch" flexibler machen. Aber nur so als Idee mal über den Zaun geworfen. Anbei auch ein Screenshot. 
[image]

dewenni, 2024-11-23:
Bitte das neue Release testen [v5.0.0](https://github.com/dewenni/ESP_Buderus_KM271/releases/tag/v5.0.0)

bsbs-net, 2024-12-08:
Hi Dewenni, vor gut ner Woche hab ich die V5.1.0 aufgespielt. Lief in meiner Testumgebung (HS2105 Steuerung ohne Brenner) schon mal sehr stabil. Daher gab ich sie Dienstag dann in meine Heizung eingebaut und auch da läuft es stabil. Ich hatte mehrere Browsersessions und die WLAN-Anbindung ist stabil. In der Heizung hatte ich 2 mal über die Woche den Fehler "Brownout reset (voltage dip)", was ja wohl an der Spannungsversorgung der Steuerung liegt, aber es hat durchgestartet uns lief weiterhin stabil.
Mit the78mole (Daniel) hatte ich auch Kontakt bezgl. Der Netzanbindung wired, wenn er ein Board verfügbar hat, werde ich beim Testen unterstützen. 
Danke dir für das tolle Update und ich freue mich schon auf weitere Fratures.
Grüße Bernd

dewenni, 2024-12-08:
Das freut mich zu hören.
Danke für das Feedback!

closed 2024-12-08 by dewenni