dewenni/ESP_Buderus_KM271 #128: hostname mit V5.0.0 nicht als MQTT

kalleick, 2024-11-25:
Seit der neuen Version v5.0.0 wird nicht mehr der WiFi Hostname bei MQTT gesendet, sondern ESP Name. 
Das verursacht bei FHEM einige Änderungen, die ich noch nicht wieder so umbauen konnte, dass ich konsistente Daten erhalte.
Daher die Bitte in der Config eine Möglichkeit einzubauen den hostname für mqtt selbst bestimmen zu können.

2024-11-23_15:26:46 MQTT2_ESP_Buderus_KM271 Aussentemperatur: 4
2024-11-23_17:53:38 MQTT2_ESP_Buderus_KM271 HK1_Pumpe: 100
2024-11-23_19:35:33 MQTT2_ESP_Buderus_KM271 Oelverbrauch_Gesamt_berechnet: 42467.1
2024-11-23_19:04:07 MQTT2_ESP_Buderus_KM271 WW_Isttemperatur: 57
2024-11-23_19:35:16 MQTT2_ESP_Buderus_KM271 boiler_temp: 58
2024-11-23_19:31:16 MQTT2_ESP_Buderus_KM271 burner: 1
2024-11-23_19:36:14 MQTT2_ESP_Buderus_KM271 message: OTA Update finished!
2024-11-23_19:35:36 MQTT2_ESP_Buderus_KM271 oelverbrauch: 0
2024-11-23_19:36:06 MQTT2_ESP_Buderus_KM271 pump: 100
2024-11-23_19:34:46 MQTT2_ESP_Buderus_KM271 ww_temp: 57
2024-11-25_05:47:50 MQTT2_esp32_c020cdef49c0 Aussentemperatur: 12
2024-11-25_05:45:33 MQTT2_esp32_c020cdef49c0 Oelverbrauch_Gesamt_berechnet: 42479.7
2024-11-25_05:47:44 MQTT2_esp32_c020cdef49c0 WW_Isttemperatur: 52
2024-11-25_05:47:47 MQTT2_esp32_c020cdef49c0 boiler_temp: 77
2024-11-25_05:47:47 MQTT2_esp32_c020cdef49c0 burner: 0
2024-11-25_05:47:57 MQTT2_esp32_c020cdef49c0 date_time: 25.11.2024 - 05:47:57
2024-11-25_05:45:38 MQTT2_esp32_c020cdef49c0 oelverbrauch: 0
2024-11-25_05:43:17 MQTT2_esp32_c020cdef49c0 pump: 0
2024-11-25_05:47:47 MQTT2_esp32_c020cdef49c0 ww_temp: 52

dewenni, 2024-11-25:
ah ok, jetzt habe ich es verstanden. Dein Log hat mich etwas verwirrt.

Der Topic Name für MQTT ist unverändert und kann in den MQTT Einstellungen vorgegeben werden.
In deinem Fall ist das vermutlich `MQTT2`.
Zur eindeutigen Identifizierung des MQTT-Client am MQTT-Broker gibt es dann noch die `Clinet-ID`.
Hier wurde bisher der "WiFi-Hostname" verwendet und seit 5.0.0 wird die Client-ID aus der MAC-Adresse gebildet.
Normalerweise taucht diese Client-ID aber auch nirgendwo auf. Zumindest hat das normalerweise keinen Einfluss auf die abonnierten Nachrichten. Die Nachricht selbst besteht ja nur aus `Topic` und `Payload`. Bei FHEM scheint er diese Client-ID aber irgendwie da mit einzubauen. Keine Ahnung.

Ich kann das aber gerne wieder ändern. Ist auch zugegeben eher unbewusst von mir geändert worden.

kalleick, 2024-11-25:
Ja, korrekt, Vermutlich ist der neue Ansatz MAC Adresse auch besser. 
Nur hat scheinbar FHEM die Eigenheit genau das voran zu stellen im FHEM Log und damit habe ich einen Bruch in den Daten.
Vielleicht kannst Du es konfigurierbar machen (nur WiFi-Hostname oder Client-ID plus MAC)

dewenni, 2024-11-25:
ich ändere es zurück auf den WiFi-Hostname. Der sollte ja auch eindeutig sein.

Kenaschon, 2024-12-02:
Vielen Dank für das "zurück" auf Wifi-Hostname. Bei der 5.0.0 Version funktioniert das "Umschalten" der Betriebsarten per mqtt nicht.

Jetzt geht es wieder.

closed 2024-12-02 by dewenni