dewenni/ESP_Buderus_KM271 #129: Update not sucessfull!

gtsy7, 2024-11-25:
[image]

dewenni, 2024-11-25:
The memory on a 4MB chip is now with Arduino V3 tight, but it should actually still fit (approx. 83% occupied)

- Have you tried it again?
- what kind of ESP do you use?
- if the OTA update is not possible, you can try to flash it with the new version

gtsy7, 2024-11-25:
Buderus-KM217-WIFI | Rev: 0.0.5
the78mole | 2022-10-13
you can send me the version 0.7 of the printed circuit board I want to make a new one.

gtsy7, 2024-11-25:
Will I have to buy it again?

dewenni, 2024-11-25:
> you can send me the version 0.7 of the printed circuit board I want to make a new one.

I thought you asked about that. Then I didn't understand the question.

gtsy7, 2024-11-25:
I already bought it, but I have an old version, I think that's why there are problems.

closed 2024-12-02 by dewenni

gtsy7, 2025-03-12:
???