dewenni/ESP_Buderus_KM271 #131: MQTT subscription für WW_BETRIEBSART eingeschränkte Reaktion seit Version 5.0

dremeier, 2024-11-27:
möchte man das subscripted topic "Buderus-KM271/setvalue/WW_BETRIEBSART" steuern, wird nur noch der Wert "1" akzeptiert.

es gilt:
## Warmwasser (WW)
Betriebsart (WW_OPMODE)
Thema: Buderus-KM271/setvalue/WW_BETRIEBSART
Werte: Ganzzahlen (z.B. 0, 1, 2 = Nacht, Tag, AUTO)

In Version 4.1 wurden alle Werte akzeptiert.

Egal ob ich nun mit MQTTX ein "Plaintext" sende oder per java-script z.B. mit  "script.js.Buderus_KM271.SendToMqtt: SendToMQTT: {"topic":"Buderus-KM271/setvalue/WW_BETRIEBSART","message":0,"retain":false}" funktioniert nur der Wert "1" alle anderen Werte werden ignoriert.

dewenni, 2024-11-27:
Danke für den Hinweis - schaue ich mir an.

dewenni, 2024-11-27:
Mir ist hier leider bei der Entwicklung der V5 ein Fehler unterlaufen bei der Verarbeitung des MQTT Payload.
Ich hatte hier zwischenzeitlich mal eine andere MQTT Library getestet bei der das Handling etwas anders war und dann beim Wechsel zurück etwas übersehen. Sorry!

Ich habe es mit der soeben veröffentlichen 5.1.0 behoben - bitte updaten!

closed 2024-11-27 by dewenni