dewenni/ESP_Buderus_KM271 #133: Solar Kollektor Temperatur falsch

solutionfound, 2024-12-01:
Hallo @dewenni 

seit gestern spinnt die Temperatur beim Solar-Kollektor plötzlich. Davor ging es noch ohne Probleme und dann springt der Wert plötzlich auf unmögliche Höhen - besonders wo es dann dunkel ist.

Ich habe ein paar Screenshots des Graphen angefügt und der WebUI. Beim ersten Screenshot ist noch alles in Ordnung und dann ab 17 Uhr kommen die falschen Werte. Auffällig ist, dass die Werte falsch sind wenn die Temperatur unter 0°C fällt. Vielleicht ein Vorzeichenfehler?

[image]
[image]
[image]

dewenni, 2024-12-01:
Ja ich denke das ist klar.
Die Werte können negativ sein und müssen daher ich anders ausgewertet werden.
Da es bisher über 0 Grad war, ist das bisher nicht aufgefallen.
Ich werde es anpassen und ein neues Release machen.
Das wird aber der einzige Wert sein, wer negativ seien kann, oder was denkst du?

solutionfound, 2024-12-01:
Im Bereich Solar denke ich nicht, dass noch ein anderer Wert negative Werte annehmen sollte - wenn das WW-Solar negativ wird, dann ist die Auswertung das kleinere Problem ;-) 

Und der unbekannte Wert 9147 hat jetzt noch nicht solch ein Verhalten gezeigt - der bleibt bei 0 und wird erst positiv wenn eine Ladung stattfindet, daher würde ich sagen, der geht nicht ins Negative. Ansonsten würde ich mich nochmals melden.

Danke Dir für die schnelle Reaktion!

Anderer Bereich, aber weil dort auch die Werte so springen: Beim Mischer springen die Werte auch so komisch. Könnte es dort auch daran liegen, dass dort negative Werte die Ursache sind?

dewenni, 2024-12-24:
Hi @solutionfound 
ich habe mal vorab eine Testversion erstellt.

#### changelog

- fix [[BUG] ws.closeAll() causes a crash on Safari](https://github.com/mathieucarbou/ESPAsyncWebServer/issues/162)
- fix "Solar Kollektor Temperatur falsch" #133
- update ArduinoJSON @ 7.2.0
- update AsyncTCP @ 3.3.1
- update ESPAsyncWebServer @ 3.4.5
- update to Arduino core 3.1.0 based on IDF 5.3.2.241210

kannst du das die Tage mal testen - vorausgesetzt es wird etwas kälter :-)

[Archive.zip](https://github.com/user-attachments/files/18238124/Archive.zip)

solutionfound, 2024-12-24:
Hi @dewenni 

danke dafür. Ich habe gerade die Firmware aus dem Archive installiert - zufällig bin ich auch bei der Heizung. 
Es ist aktuell zu warm am Solarkollektor, aber ich werde die Werte beobachten, da sie ja auch über Openhab protokolliert werden.

Es scheint soweit alles wie zuvor zu funktionieren.
Allerdings ist habe ich gerade die Werte für Ferien von 0 auf 2 geändert und habe keine Veränderung bei der Steuerung gesehen.
Ich bin jetzt kurz zur Heizung und dort war im Urlaubsmenü noch 0 bei allen Heizkreisen. Dann habe ich dort den Wert auf 1 gesetzt und der wird in der Weboberfläche angezeigt.
Es liegt daran: Wenn man im Browser nur die Pfeiltasten verwendet um den Wert einzustellen, wird er nicht übertragen. Man muss ins Feld gehen und 1x Return drücken. Ich konnte das über das Logbuch so testen. Das wäre noch eine kleine Verbesserung oder Du nimmst die Pfeile weg oder Du schreibst einen Hinweistext, dass ins Feld gegangen werden muss und Return gedrückt werden.
Wenn man es mit dem Handy einstellt, wo keine Pfeile vorhanden sind funktioniert es. 

Eine weitere "Kleinigkeit" - das war in V5.1 schon so: wenn man das Handy im Querformat verwendet und das Menü aufruft, kann man das Menu nicht scrollen, sondern es scrollt der rechte Bereich. Damit kommt man nicht an die unteren Menüs.

Und noch etwas - Im HK 1 werden unter Programme folgendes angezeigt, was nicht die richtigen Werte sind. Getestet mit Single und Senior Programm, wo immer das Folgende angezeigt wird.
HK1 Timer 01 | SP01: Mo 00:00  (An)  \| SP02: Mo 06:00  (An)  \| SP03: Di 00:00  (An)
HK1 Timer 02 | SP04: Di 06:00  (An)  \| SP05: Mi 00:00  (An)  \| SP05: Mi 06:00  (An)
HK1 Timer 03 | SP07: Do 00:00  (An)  \| SP08: Do 06:00  (An)  \| SP09: Fr 00:00  (An)
HK1 Timer 04 | SP10: Fr 06:00  (An)  \| SP11: Sa 00:00  (An)  \| SP12: Sa 06:00  (An)
HK1 Timer 05 | SP13: So 00:00  (An)  \| SP14: So 06:00  (An)  \| SP15: -

Bei Heizkreis 2 dachte ich, dass die Werte richtig sind, weil sie dem Senior Programm entsprechen, das eingestellt war:
HK2 Timer 01 | SP01: Mo 05:30  (An)  \| SP02: Mo 22:00  (Aus)  \| SP03: Di 05:30  (An)
HK2 Timer 02 | SP04: Di 22:00  (Aus)  \| SP05: Mi 05:30  (An)  \| SP05: Mi 22:00  (Aus)
HK2 Timer 03 | SP07: Do 05:30  (An)  \| SP08: Do 22:00  (Aus)  \| SP09: Fr 05:30  (An)
HK2 Timer 04 | SP10: Fr 22:00  (Aus)  \| SP11: Sa 05:30  (An)  \| SP12: Sa 22:00  (Aus)
HK2 Timer 05 | SP13: So 05:30  (An)  \| SP14: So 22:00  (Aus)  \| SP15: --

Wenn man das Programm wechselt, ändern sich aber die Werte nicht. Auch ein Neustart oder kompletter Power Cycle der Heizung hat nichts geändert.

Auch beim Wechsel eines Programms über das Menü Bedienung muss man nach dem Wechsel des Wertes im Browser erst irgendwo außerhalb des Feldes klicken, damit der Wert übertragen wird. Das finde ich nicht intuitiv.

Jetzt hoffe ich, dass das nicht zu viele Themen sind und ich weiß auch nicht, ob die mit dem neuen Release zusammenhängen.

"Trotzdem" noch schöne Weihnachten ;-)

dewenni, 2024-12-24:
Vielen dank für das ausführliche Feedback. Ich schaue mir das die Tage mal an.

solutionfound, 2024-12-24:
Kurzes Feedback zu der Kollektortemperatur.
Unter 0°C springen diese immer noch auf ca. 255.
Aktuel sind es 252°C.

solutionfound, 2024-12-24:
Herde noch enteckt:
Bei der Bedienung ist für HK1 
- HK1-Frost ab bei -19°C
- HK1-Aussenhalt ab bei -20°C
Ich denke die Werte machen keinen Sinn.

HK2-Frost ab ist bei 7°C das erscheint mir auch hoch.

Verstellt habe ich nichts und ich glaube davor war es korrekt.

dewenni, 2024-12-25:
### Kollektor Temperatur:
Den Fehler von der Kollektortemperatur habe ich vermutlich gefunden.
Ich hatte die "Decodierung" von dem Wert aus dem Telegramm angepasst, aber der Datentyp der Variablen war noch als "vorzeichenloser Wert" definiert.

### Bestätigung Eingabefelder
Das schaue ich mir noch in Ruhe an. Das kann man sicher optimieren.

### Programme
Das ist bei mir tatsächlich auch so und ist mir bisher nicht aufgefallen, da ich es auf "Eigen" stehen habe und da passen die eingestellten Werte auch:
[image]
Wenn man auf ein anderes Programm umschaltet, werden aber weiterhin die gleichen Werte gesendet.
Vermutlich ist das aber auch so richtig und es werden hier nur die "eigenen" Timer gesendet und nicht die Timer der jeweils anderen Programme. 

### HK1-Frost ab bei -19°C / HK1-Aussenhalt ab bei -20°C
Das kann ich jetzt spontan nicht nachvollziehen.
Wird das bei dir nur im WebUI so angezeigt oder auch per MQTT?
Bei mir stimmt beides:
[image]

[image]

Hier nochmal ein Archiv mit nur der Änderung für die Kollektortemperatur:
[Archive.zip](https://github.com/user-attachments/files/18245362/Archive.zip)

solutionfound, 2024-12-26:
[image]

Der Fix für die negative Kollektortemperatur funktioniert. 
Auch in der Übersicht werden die Solarwerte richtig angezeigt.
Danke!

Bezüglich der Programme hast Du wahrscheinlich recht, dass nur die Werte des individuellen Programmes gesendet werden und das müssten auch die Werte sein, die in der Steuerung hinterlegt sind, da ich früher mal individuelle Programme verwendet hatte.
Ich bin nur darauf gekommen, weil ich die Bedienung getestet habe und um zu sehen, ob der Wert übernommen wurde, wollte ich bei den Heizkreisen nachschauen. Dort ist aber in der Übersicht (Konfiguration / Status)  nicht das gewählte Programm aufgeführt, sondern untern Programme nur die Einzelwerte, von denen ich fälschlicherweise angenommen hatte, dass diese sich ändern.
Vielleicht kannst Du das gewählte Programm noch unter Konfiiguration aufnehmen, dann wäre alles zusammen. Oder unten bei Programme (auch) aufführen mit dem Hinweis, dass das die Werte für das Programm "Eigen" sind.

Bezüglich der "komischen" Werte bei Frost, etc.
Gestern habe ich das Update gemacht und die Urlaubstemperatur über das Handy im Webfrontend geändert. Beim Überprüfen ist mir aufgefallen, dass die Auslegung einen viel zu hohen Wert hat. Den habe ich sicher nicht eingestellt. Ich glaube auch nicht, dass ich aus versehen, den Wert geändert habe, da die Werte im Frontend weit auseinander liegen.
Hier das Protokoll dazu:
[26.12.2024 - 00:13:44] Message : Befehl: HK1_Urlaubtemperatur - Wert: 17.0
[26.12.2024 - 00:13:42] Message : Befehl: HK1_Urlaubtemperatur - Wert: 17.5
[26.12.2024 - 00:12:55] Message : Befehl: HK1_Auslegung - Wert: 84
[26.12.2024 - 00:00:54] Message : ESP hat neu gestartet! (ota update)

Vielleicht sind die verstellten Weerte bei Frost "ähnlich" zustande gekommen. Mir scheint als gibt es irgendeine Konstellation nach dem Update vielleicht mit offener Webpage.
Das wird wahrscheinlich schwer zu testen sein, aber an einen Zufall glaube ich jetzt nicht mehr.

Dann nochmals kurz der Hinweis, dass ich das "Gefühl" habe, dass die komschen Mischerwerte auch wegen negativer Werte so komisch ausschauen, da sie springen. Vielleicht implementierst Du die Werte wie die negativen Solarwerte - ich glaube dann ergeben die mehr Sinn, siehe Werte bei mir.
[image]

dewenni, 2024-12-29:
@solutionfound 
Ich habe die Punkte größtenteils umgesetzt im neuen Release v5.2.0

solutionfound, 2025-01-01:
@dewenni 👍 funktioniert alles wunderbar - ich habe die SW jetzt ein paar Tage laufen lassen.
Auch beim Update wurde nichts "verstellt".
Danke für die Umsetzung.

closed 2025-01-02 by dewenni