dewenni/ESP_Buderus_KM271 #135: nicht steuerbar über MQTT

DonCanDo, 2024-12-04:
Moin Leute,

leider habe ich seid ein paar Tagen das Problem, dass Änderung im Homeassistant nicht an das Modul weitergeleitet werden.

Was ich feststellen kann, dass sich der Logmode alle paar Stunden für wenige Sekunden von true auf false und dann wieder true ändert.

In den MQQT Informationen vom Modul werden keine Fehler angezeigt. Die WLAN Verbindung ist auch stabil bei einem rssi von nie unter 50dbm.

dewenni, 2024-12-04:
Bist du bereits auf der 5.1.0?

Das die Verbindung zum Broker immer wieder abbricht, ist aber schon komisch

DonCanDo, 2024-12-04:
Moin,

ach habe ich nicht mitbekommen, dass es ein neues Update gab. Musste das Board neu flashen weil es wieder offline war. Und jetzt läuft es mit der Version 5.1.0 wieder und lässt sich über MQTT steuern.

Danköö

DonCanDo, 2024-12-10:
Hatte ich geupdatet dann war die Verbindung weg.

Habs mit 5.1.0 noch mal geflasht soweit läuft es wieder =)