dewenni/ESP_Buderus_KM271 #137: Langzeitstatistiken

stefangrosse, 2024-12-30:
Hi.

Ich habe ein Problem mit den Statistiken, was mir heute erst aufgefallen ist:

[image]

Oben ist mein Ölverbrauch den ich aus der Brennerlaufzeit berechne. Die Brennerlaufzeit geht aber nur bis zum 20.12. zurück. Es wurde aber schon vorher der Verbrauch errechnet. Das ist bei allen Entitäten die von der Schnittstelle kommen so. 
Woran kann das liegen? Bin für jeden Tipp dankbar!

stefangrosse, 2024-12-30:
In der Datenbank sind die States vorhanden:

[image]
[image]

Es sind aber keine Statistiken vorhanden:
[image]

Meine alten Sensordaten sind noch da (Vor der Umstellung auf Karte) :
[image]

paxrsi, 2024-12-30:
Ich hatte das gleiche Problem. Die Sensordaten waren nur die letzten x Tage im Homeassistant vorhanden. So wie ich das verstanden habe ist die Ursache, dass Homeassistant zwei Datenbanken hat. Eine mit den vollständigen Daten für ein paar Tage. Danach werden die Daten in längere Zeiträume zusammengefasst in eine andere (Langzeit)-Datenbank übertragen. Dieser Übertrag fand bei mir bei den Logamatic-Daten nicht statt.

Ich habe das Problem bei mir wie folgt gelöst:

1.) Es gibt eine Einstellung, mit der man Homeassistant sagt, wie lange die Daten in der Kurzzeitdatenbank sein sollen. Die Einstellung kann man nur für alle Datenquellen universell angeben. D.h. die Größe der Datenbank wächst damit stark an. In der configuration.yaml habe ich dazu folgendes für 31 Tage aufgenommen:

```
recorder:
  purge_keep_days: 31
```

2.) Damit die Daten in die Langzeitdatenbank übertragen werden, habe ich eine weitere Anpassung in der configuration.yaml vorgenommen. Dazu habe ich mich an dem orientiert, was ich für meinen Stromverbrauchsmesser am Zählerkasten mal aufgenommen hatte. Was das genau tut und warum das funktioniert, kann ich dir leider auch nicht sagen. So tief stecke ich in Homeassistant nicht drin.

```
homeassistant:
  customize_glob:
   sensor.logamatic_oelverbrauch_gesamt_berechnet:
      state_class: measurement
      device_class: volume
      unit_of_measurement: "l"

    sensor.logamatic_aussentemperatur_gedaempft:
      state_class: measurement
      device_class: temperature 
      unit_of_measurement: "°C"
```
Ich hoffe, das hilft!

Keine Ahnung, ob man das dem Homeassistant schon bei der Erkennung der MQTT-Daten beibringen kann. Aber ehrlich gesagt, brauche ich die Langzeitspeicherung auch nicht für alles.

Grüße
Rainer

stefangrosse, 2024-12-30:
Danke! Das hat mich auf den richtigen Weg gebracht. Es muss die richtige state_class gesetzt werden, damit eine Langzeitstatistik erstellt wird. Die Recordereinstellungen sind aus meiner Sicht nicht nötig, die sind im Standard 10 Tage (https://www.home-assistant.io/integrations/recorder/#purge_keep_days).

Hier wird es in Doku beschrieben:
https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

Für den Sensor Brenner Laufzeit Summe ist keine state_class angegeben: 

[image]

@dewenni Wäre es nicht sinnvoll das zu ergänzen? 

Ich probiere diese jetzt über:
```
homeassistant:
  customize:
    sensor.logamatic_brenner_laufzeit_summe:
      state_class: measurement
```

dewenni, 2024-12-30:
ich kann `state_class: measurement` gerne in der Konfiguration ergänzen.
Wenn Ihr sonst noch Änderungsvorschläge bei der Konfiguration habt, dann stellt mir das gerne mal zusammen.
Ich bin zugegeben nicht der größte Home Assistant Experte und habe das nach bestem Wissen und Gewissen gemacht.
Also macht mir gerne eine Übersicht bei welchen Datenpunkten ich noch was ändern oder ergänzen sollte.