dewenni/ESP_Buderus_KM271 #149: [BUG] Fehler der Buderus Steuerung wird nicht gemeldet

bsbs-net, 2025-03-28:
### Wurde der Fehler in der neuesten Version getestet?

- [x] Ich nutze die **neueste Version**. (bitte zuerst aktualisieren und erneut testen, bevor der Fehler gemeldet wird)
- [x] Ich habe die verfügbare Dokumentation gelesen und keine Hinweise auf mein Problem gefunden
- [ ] Ich habe mögliche Hinweise in den aktuellen Release Notes gelesen und keine Hinweise auf mein Problem gefunden

### Hardware-Informationen

- **KM271-WiFi Board** [the78mole 0.0.8 2023-09-27] 

### Fehlerbeschreibung

Heute ging mein Buderus BE1 mit einem Error auf Störung. Nach
einiger Analyse konnte ich feststellen,  habe keinen Zündfunken. Im Rahmen der Fehlersuche hatte ich mehr als 10 Fehler produziert, jedoch wurde keiner gemeldet, nicht im Webfrontend, nicht via MQTT und auch nicht im HA. Zu diesem Zeitpunkt hatte ich über das Webinterface zum Test vom Modus "Auto" auf "Day" und zurück geschaltet, funktionierte problemlos. Die Komminikation zwischen Modul und Steuerung scheint OK. Aber eben leider keinen Fehlerm was mein eigentlicher Grund für diese Erweiterung war. Auch beim PowerOn oder Änderungen über das Webfrontend, kommt die MQTT-Notification.

Ich hatte letztes Jahr vor dem Einbau der Platine auch einen Fehler in der Buderus und dieser steht noch immer als letzter Alarm in Meldung1 "unknown error (>>00:03 - 191 days | <<128:01 -0 days)". Meldung 2 bis 4 waren zuvor leer und sind es noch immer.

Ich wüsste jetzt nicht, was in meinen Einstellungen falsch sein könnte. Hast Tipps zum Testen? Was kann ich tun oder benötigst du weitere Infos.

Übrigens wie vor Monaten geschrieben, deine neue Version v5.3.1 läuft wesentlich stabiler 😀, aber seit heute morgen bin jetzt gefrustet, da die gewünschte Alarmierung nicht klappt. 😒

Hoffe du hast eine Idee und schon mal Danke
Bernd

dewenni, 2025-03-28:
Meine Buderus läuft glücklicherweise absolut störungsfrei, daher habe ich das lange nicht mehr getestet.
Damals hatte ich mal einen der Sensoren abgeklemmt im Fehler zu provozieren. Das könnte man nochmal testen.

bsbs-net, 2025-03-28:
Hab gerade nochmal weiter gesucht und im HA gesucht. Den hatte ich aufgesetzt aber (noch) nicht viel damit gemacht. Hier finde unter "Heating Boiler failure burner" die Fehler. Sorry. Dann liegt es wohl doch an meiner Config. Ich habe PushOver aktiviert mit "Alarme und Info" und bekomme darüber auch die PowerUp bzw. Statusinfos, aber der Alarm blieb aus. Via MQTT ging es wohl doch an den HA. Sorry.

[image]

dewenni, 2025-03-29:
Im WebUI sollte der Fehler aber auch unter Alarme auftauche.

Du könntest zum Test mal einen der Temperaturfühler oben an der Steuerung abziehen. So hatte ich es damals getestet. Z.B den Stecker "FK"

bsbs-net, 2025-03-29:
Hi, schon mal Danke für den Tipp.

Hab den Stecker FK abgezogen und im Display der Steuerung erscheint direkt der Fehler.
Im HA kann ich diesen auch sehen, siehe Screenshot. 

Im WebUI sehe ich keinen Fehler, auch nicht im Logfile, es bleibt bei Abruf leer.
Im Logfile erhalte ich von diversen Abfragen nur bei der letzten Möglichkeit überhaupt ein Ergebnis. Siehe Screenshot. 

Die einzige Einstellmöglichkeit bietet nach meinem Ermessen der Bereich Logamatic, auch hier mal einen Screenshot. 

Liege ich richtig, dass der Fehler von der Logamatic-Steuerung schon im Board ankommt und via MQTT an den HA richtig weiter gereicht wird? Aber 

[image]

weder im WebUI sichtbar noch via Pushover gesendet wird.

[image]

[image]

[image]

dewenni, 2025-03-30:
Dane fürs Testen. Ich schaue mir das bei nächster Gelegenheit bei mir an.

closed 2026-11-15 by dewenni