dewenni/ESP_Buderus_KM271 #152: [BUG] Ethernet Informationen nocht vollständig

the78mole, 2025-04-14:
### Wurde der Fehler in der neuesten Version getestet?

- [X] Ich nutze die **neueste Version**. (bitte zuerst aktualisieren und erneut testen, bevor der Fehler gemeldet wird)
- [X] Ich habe die verfügbare Dokumentation gelesen und keine Hinweise auf mein Problem gefunden
- [X] Ich habe mögliche Hinweise in den aktuellen Release Notes gelesen und keine Hinweise auf mein Problem gefunden

### Hardware-Informationen

- **ESP32-Typ**: ESP32-WROOM-32DN8
- **KM271-WiFi Board** Meine (the78moles) neueste HW 0.1.0
- **ETH-Extension**: W5500

### Fehlerbeschreibung

Im Reiter **System** wird die IP-Adresse und der Status korrekt angezeigt, aber Geschwindigkeit und Duplex noch nicht.

[image]

Da die ETH-Extension ja noch nicht für alle verfügbar ist, eilt es nicht 😛

dewenni, 2025-04-14:
wenn das bei dir nicht die Ursache ist, dann muss ich das nochmal mit dem Extension Board testen.
Vielleicht klappt da auch die Abfrage über der Ethernet Library noch nicht.
Schaue ich mir dann an.

the78mole, 2025-04-14:
Bei mir hängt gar keine Heizung dran... Ich teste ja gerade nur die ETH-Extension. In meinem Heizungskeller habe ich auch gerade keine Netzwerkanbindung, weswegen ich das auch nicht ohne größeren Aufwand testen kann...

closed 2025-04-19 by dewenni